open62541/open62541 #251: Server Timeout when using subscriptions

ichrispa, 2015-05-15:
Since I've been implementing subscriptions I have been getting the following error: I can start both server and client; the server will listen for as long as necessary for the client to connect. After connecting, I can interact for ~7 seconds, then the client locks up. 

UAExperts error message is:

    Connection Status of server ... changed to 'ConnectionWarningWatchdogTimeout'
    Connection Status of server ... changed to 'ConnectionErrorApiReconnect'

There is no reconnect occuring after that, even though the client tries to send HELLO messages. Wireshark shows that the client periodically sends requests, but the server doesn't react. When the server is stopped, I always catch it running in `static void UA_Server_cleanup(UA_Server *server, void *nothing)`, in particular:

    while(sentry) {
            if(sentry->session.validTill < now) {
            ...
            }
    }

When I try to step through the code with gdb, I get a program termination. 

At first I thought the timeout was due to the errating behavior of my subscriptions (as in not returning data, not deleting acknowledged notifications, etc.), but now that they do the problem persists.

ichrispa, 2015-05-20:
I had some time to find out what happens here and the answer is the following:

The server will periodically enter this part of server_cleanup() in ua_server.c:

```
while(sentry) {  
      if(sentry->session.validTill < now) {
            session_list_entry *next = LIST_NEXT(sentry, pointers);
           ...
            sentry = next;
        }
    }
```
When this loop is entered and a session is still valid:
  * `now` will never be updated.
  * `sentry` will not advance.
One of the two has to be fixed. Either sentry=next; must be updated outside the if()-block, or we need to update `now` inside the loop. Otherwise the server gets stuck here.

I fixed this in the dev branch by updating `now` inside the loop because I am not quite clear on what the sentry is trying to accomplish. Also the session timeout will simply overflow but is still processed if the loop is escaped... don't know if that's intentional.

ichrispa, 2015-05-20:
Quote from the Spec on Timeout Value in SessionCreate:
```
Requested maximum number of milliseconds that a Session should remain open
without activity. If the Client fails to issue a Service request within this interval, then
the Server shall automatically terminate the Client Session.
```
So my understanding is that each time a message is received by the server, this the validTill value should be updated to now+TimeOutValule? Is this something each service should do on its own (unlikely) or should this be handeled in processMSG() ?

FlorianPalm, 2015-05-20:
Like you say within processMsg, as new valid-time should be set for every service which relies on a session.

jpfr, 2015-05-20:
Hey @ichrispa ,

master and 0.1 do this already.
I will soon make a merge of master into dev to keep the branches in sync.
Sorry if I delayed this for too long. Things were a bit hectic prior to 0.1 (and still are outside of open62541...).

Server cleanup: https://github.com/acplt/open62541/blob/master/src/server/ua_server.c#L119
Calls session cleanup: https://github.com/acplt/open62541/blob/master/src/server/ua_session_manager.c#L24
Every service call updates the session timeout: https://github.com/acplt/open62541/blob/master/src/server/ua_server_binary.c#L154

ichrispa, 2015-05-21:
Hi @jpfr,

I am both happy to have identified this problem correctly and that you already solved it. I'll remove my hack in the dev branch and we can consider this issue closed :)

closed 2015-05-21 by ichrispa