open62541/open62541 #253: State of Subscription implementation

ichrispa, 2015-05-22:
Greetings,

the development branch now has support for subscriptions (Create, Delete) and monitoring (Add, Delete) as well as publish. All three are in a very rudimentary form, as there is currently no Event system in place. As far as necessary changes go, I advise reading the FIXME tags in `Service_Publish()`.

For reference, the heavy lifting of the current implementation is handled by two functions:
`void MonitoredItem_QueuePushDataValue(UA_MonitoredItem *monitoredItem)` is called to copy a monitored Attribute into the queue. The function will not copy data if the monitoringIntervall is not reached or the value has not changed. This method should be called by an event system.
`void Subscription_updateNotifications(UA_Subscription *subscription)` will scan all monitored Item queues, compile them into a notificationMessage list and clear the queues. The top message of these messages is send on a publish request. This method should be delegated to a timed work thread.

Currently as a workaround all monitoredItems are being updated when a publish request is received, then all subscriptions are asked to compile their messages. The top message is then sent. I tested this using both 'servertime' (dynamic value) and 'the answer' (static value).

I would leave active this ticket as a work item to remind me that subscriptions are not ready yet. However I urgently need to switch the development focus to methodcalls for now.

Stasik0, 2015-05-22:
@ichrispa wow, great news! I will try it asap. The multi-threaded version, does however fail https://travis-ci.org/acplt/open62541/builds/63651540#L731 are these major issues or can you fix them easily?

Stasik0, 2015-05-22:
okay, we badly need to reintegrate master into dev, there is a couple of bugs in the dev version...

however, I can not create a subscription with my UaExpert v 1.3.1. When I pull "the answer" to the "Data Access View", I become a ```Create Subscription failed [ret=BadUnknownResponse]```.
@ichrispa how do you test the subscriptions?

ichrispa, 2015-05-22:
Multithreading and tests: I will look into it.

I compile with -DENABLESUBSCRIPTIONS and then drag the item in UA_Expert... weird. There is obviously still a lot of compat testing to be done here.

I will look into it asap. However 1.3.1 just crashed on my linux (no libpng14). I'm running 1.2.2.

Stasik0, 2015-05-22:
My bad, I did not activate subscription flag... Will test asap

ichrispa, 2015-05-22:
@Stasik0: Neither builds failing the tests are using subscriptions or my headers. These errors must be remnants of pre-fork bugs you have already fixed in master or 0.1. I suggest we merge the master and dev, then check if these errors persist.

I can't reproduce the error on Win7/UAExpert 1.3.1. Subscriptions apear to be working as they do under Linux/UAExpert 1.2.2.

I have not patched the cleanup() extensively, since it's fixed in the master branch. It's just barely patched enough to keep the server from hanging itself during cleanup. So you will run into continuous reconnects every 7 seconds with lags between reconnects. Sry about that.

jpfr, 2015-05-22:
ok. master just got merged into dev.
it compiles. but i didn't test it thoroughly.

should be could. the new subscription code is mostly self-contained.

jpfr, 2015-05-22:
Here's the current diff against master. The three _subscription files have ~1k lines of code. Thats good and small!
```
 CMakeLists.txt                                 |   18 +
 src/server/ua_securechannel_manager.c          |    1 -
 src/server/ua_server.c                         |    1 -
 src/server/ua_server_binary.c                  |   20 +-
 src/server/ua_server_internal.h                |    4 +
 src/server/ua_server_worker.c                  |   10 +-
 src/server/ua_services.h                       |   45 +-
 src/server/ua_services_subscription.c          |  282 +++++
 src/server/ua_subscription_manager.c           |  546 +++++++++
 src/server/ua_subscription_manager.h           |  115 ++
 src/ua_session.c                               |    3 +
 src/ua_session.h                               |    7 +
 tools/generate_datatypes.py                    |   11 +-
 tools/pyUANamespace/generate_open62541CCode.py |   99 ++
 tools/pyUANamespace/logger.py                  |   45 +
 tools/pyUANamespace/open62541_MacroHelper.py   |  133 +++
 tools/pyUANamespace/ua_builtin_types.py        | 1082 +++++++++++++++++
 tools/pyUANamespace/ua_constants.py            |   57 +
 tools/pyUANamespace/ua_namespace.py            |  629 ++++++++++
 tools/pyUANamespace/ua_node_types.py           | 1488 ++++++++++++++++++++++++
 20 files changed, 4568 insertions(+), 28 deletions(-)
```

ichrispa, 2015-05-22:
Compiles and runs perfectly ^^ This is the first time I ever monitored ServerTime for more than 7 seconds at a time - glorious :)

@Stasik0 can you confirm that subscriptions are working now?

Stasik0, 2015-05-22:
wooow i confirm it working!!

just being a nagger: there are memleaks and memory problems indicated in valgrind

```valgrind --leak-check=full ./server_static```

ichrispa, 2015-05-22:
Your're not nagging, but perfectly correct. I know where the leaks are, but I've been putting of fixing them for getting the subscriptions to run at all.

Subscriptions use a pretty deeply nested structure of Notifications containting datasources containing variants that might contains pointers or bytestring or... That nested structure is created when notifications create their queues; then they are packed into a bytestring. They should be deallocated after that, but only part them are (right at UA_Free(changeNotification) in updateNotifications).

I'll get around to improving that along with more attribute types and minor improvements i've been putting off.

jpfr, 2015-05-22:
> All three are in a very rudimentary form, as there is currently no Event system in place.
We can just use the event system in ua_server_worker.c

- If a notification needs to be sent with no delay, either do it directly (before answering the service that triggered the notification) or add a "timed-work" with the current datetime to the server. That will be executed in the next main-loop iteration of the server.
- For interval notifications, just add a "repeated timed-work" with an interval to the server. If the node is changed, we could set a flag in the "listening" subscriptions. The subscriptions then react "by themselves" when their callback is invoked at the end of their interval.

What is afaik missing is removing repeated timed work from the server. But, we generate a random Guid during insertion. So we can at least find the entry. :-)

jpfr, 2015-05-23:
Hey @ichrispa ,

1. I just split `ua_subscription.h` out from `ua_subscription_manager`. Was that ok?

You extensively use the lists in queue.h. That's good.
Normally, it should not even be necessary to use the lh_first, etc. pointers.
That makes the code a bit easier to read.

For traversing lists, there is this shortcut:
```c
MonitoredItem_queuedValue *queueItem;
LIST_FOREACH(queueItem, &monitoredItem->queue, listEntry) { 
    // do stuff with queueItem
}
```
For traversing lists and removing its elements, we just use this pattern:
```c
void UA_SessionManager_deleteMembers(UA_SessionManager *sessionManager) {
    session_list_entry *current;
    while((current = LIST_FIRST(&sessionManager->sessions))) {
        LIST_REMOVE(current, pointers);
        UA_Session_deleteMembersCleanup(&current->session);
        UA_free(current);
    }
}
```

2. How would I point from a node to a subscription? Do we just add a list fo subscriptions to every node?

ichrispa, 2015-05-23:
@jpfr: Splitting the header is perfectly ok by me. What I submitted is a working state - that doesn't meen its orderly (though I tried to keep it that way) ^^

FOREACH() has several pitfalls: Most notably it ends with a NULL-Pointer in the iterated item. That is not always desirable, for example when you want to have a fall-through as in 'if none match, use the last one'. That's also the case when you want to remove the last item of a list (ie "DiscardLast" option in monitored items). Because I ended up mixing FOREACH() with other iteration strategies, I chose the more coherent approach of just using the none-macro version. Of course, FOREACH makes the code a lot more readable.

I would not point to a subscription, but to a list of monitoredItems. A subscriptions just organizes them for the client. It's the monitoredItem that points to the nodes, so I would in accordance have the node point back at the monitoredItem. Subscriptions keep a list of `MonitoredItems` in their struct, you could just reuse that list entry definition.

So I would propose adding something like this to a node's base attributes:
`LIST_HEAD(UA_ListOfUAMonitoredItems, UA_MonitoredItem_s) MonitoredByMonitoredItems;`

Note that there are cases when a node can get deleted and the MonitoredItem persists, especially in StatusChangeNotification types.

ichrispa, 2015-05-23:
@Stasik0: I fixed several of the memleak issues, but I'm kind of stumped by one.

Publish copies the toplevel NotificationsMessage into the response (which is a NS0 type). I expected this to be deallocated by UA_deleteMembers() when the reply is send, but apparently its not. Safe for stepping through the message deletion process, is there any quick answer as to why NotificationsMessages are not deleted?

Stasik0, 2015-05-23:
@ichrispa: our _copy functions just allocate the memory by themselfs (probably, for historical reasons)... I fixed it...

ichrispa, 2015-05-24:
@Stasik0: That segfault was mine. I pushed a fix. I can't reproduce it on my 32 Bit laptop, but my work computer segfaults sure enough. I hope that was the only place where I did that...

Stasik0, 2015-05-24:
okay, the segfault is gone

1 more question: in UaExpert i have 2 subscriptions: the.answer and current time. Then in the second thread i call "./client". The time updates nicely, while "the.answer" seems to stay unchanged in UaExpert. What is the deal?

ichrispa, 2015-05-24:
I did the following: I started the server, then subscribed to 'the.answer'. I then ran the example client - the answer stayed 42. When I created a second new subscription the 'the.answer', the second item showed the value 43. I halted the server in the debugger and took a look at the monitored item struct. Not suprisingly, there where two monitored items pointing to two different node addresses.

The answer to your question is very simple: "the.answer" does not show an updated value because the variable nodes answer never gets updated. Instead you apparently copy the the "the.answer" node and substitute the copy with the new value in the nodestore. The old monitoredItem keeps tracking the old node address, so the value is never changed.

We can solve this in two ways: (1) I search the nodestore for the monitoredNodes NodeID and only store that (increases execution time for the lookup and the size of the monitoredItem structs) or (2) we modify the write service to just change the variant, not the entire node.

Stasik0, 2015-05-24:
For (2) we need to ask @jpfr about immutable operations.

jpfr, 2015-05-25:
Yes, it was a deliberate decision to replace the entire nodes upon edits, since that can be done with an atomic pointer compare-and-switch.

Otherwise, we would have to protect the node currently being edited with mutexes/semaphores. Writers and readers would both need to lock always. Looking 5 years ahead (with 32-core processors already in servers today), we could easily spend 98% of the cores' capacity waiting for mutex releases on a congested node (see also Amdahl's law).

We can use normal POSIX mutexes for all the sporadic stuff, like adding a session or adding a subscription. Yet, I would be very grateful if the **hot code path** (reading, writing, browsing, notifications) was entirely **lock-free**, so that each core can run as fast as possible. The worker code already distributes incoming service requests and timed work to the available worker threads (when multithreading is baked in).

Let's get the single-core stuff working first! And node-lookups with the nodeid aren't so problematic. I easily achieved a couple million lookups/sec on my laptop when I last checked...

Stasik0, 2015-05-26:
@jpfr, @ichrispa cant we go a pragmatic way and have best of two worlds?

for single-threaded - just replacing the variant and have subscriptions work with pointers
for multi-threaded - replace the node and have subscriptions work with nodeids/hashtable lookups?

or will it introduce too much untestable code?

I have just committed a version where write just updates the variant and not the node itself in the single-threaded case. @jpfr feel free to rollback it. @ichrispa the subscription is, unfortunately, not working even in this case

jpfr, 2015-05-26:
It's a good idea to just replace the variant in single-threaded mode. We do that already in places where it was easy enough to separate mt / non-mt code paths.

But to just point into a node, we would have to remove the function `ua_nodestore_replace` (and `ua_nodestore_delete`) entirely for singlethreaded. Otherwise, it will be very difficult to track if a node still exists or has been garbage collected.

I'm on a tight schedule and can't invest much time before next week...

ichrispa, 2015-05-26:
@Stasik0 your version did not change the new-object-address problem. Here's a screenshot of how I see it manifest itself. As you can see "the.answer" is monitored twice, and both monitoredItems still have different node addresses - the first one actually returns the correct value.

[image]

I pushed a fix where the monitoredItem  looks up the node address on each QueuePushDataValue. This now shows the behavior that you would expect when running the client, though it has a performance penalty.

jpfr, 2015-05-26:
The performance penalty of the lookup is on the order of 200-300 cycles.
That's similar to getting data from RAM, i.e. accessing data outside of the processor cache.
On a 3Ghz machine that gives 10mio lookups per second and core.

Stasik0, 2015-05-26:
@ichrispa okay, it turns out that i mixed up "ifdef" and "ifndef" *fail*. Now the problem is that nodestore_get returns const pointer that i cannot change. I do not want to update too much of the code, therefore i will just revert my commit.

jpfr, 2015-06-04:
Hey everybody,

I just pulled master into dev. Before the pull, subscriptions did not update in UAExpert on my box. And stopping the server with ctrl-c segfaulted.
Now, at least I can ctrl-c exit without a segfault. Still no updates in UAExpert.

The big news is that we have `UA_Server_addRepeatedJob` and  `UA_Server_removeRepeatedJob`. Both can be called asynchronously from within the worker threads.

Currently, we update the subscriptions and send a publish response whenever a publish requests comes in. Now, we can decouple those
- For MonitoredItems, add a "sampling" job with the samplinginterval of the monitoreditem
- For Subscriptions, add a "publish" job that dequeues a request and sends the relevant updates

When the MonitoredItem or Subscription is removed, we can then remove the repeated jobs with a Guid that was returned from the addRepeatedJob function.

Stasik0, 2015-06-04:
Subscriptions work here!

ichrispa, 2015-06-05:
@jpfr: Do you have steps that can reproduce the segfaulting behavior? My subscriptions run in the precise way they always did in UA Expert (i.e. working as expected).

Concerning the timeWorkItem: I will eventually come around to equipping subscriptions with that mechanism. Please note that MonitoredItems do not have a sampling interval, only a trigger condition (currently: change).

The current server is not able to not return a response message; so how do we queue publish requests? Or, if we have somehow managed to queue 4 of them and suddenly get 4 notificationmessages from 4 subscriptions... how would the server send these without a request by the client? My take is that OPC UA is basically query-response driven and if a publish request always returns some data (if just a keepalive) we do not strictly speaking need a queue at all.

jpfr, 2015-06-05:
I misconfigured something. Subscriptions work for me now. The segfault is already fixed on dev.
So all is well!

> The current server is not able to not return a response message;
Then we will have to change the server a bit. I believe this affects only the publish service. Maybe we only need to add a return statement to the publish-case in the big switch in `ua_server_binary.c`.

What is currently happening is that the client tries to pipeline requests. But we answer all of them immediately. That effectively floods the connection with a much higher response frequency than the subscription normally would have.
Section 5.13 of part 4 of the standard is very detailed on how subscriptions ought to work.

> NotificationMessages are sent to the Client in response to Publish requests. Publish requests are
normally queued to the Session as they are received, and one is de-queued and processed by a
subscription related to this Session for each publishing cycle, if there are Notifications to report.
When there are not, the Publish request is not de-queued from the Session, and the Server waits
until the next cycle and checks again for Notifications.

> NotificationMessages are sent to the Client in response to Publish requests. Publish requests are
normally queued to the Session as they are received, and one is de-queued and processed by a
subscription related to this Session for each publishing cycle, if there are Notifications to report.
When there are not, the Publish request is not de-queued from the Session, and the Server waits
until the next cycle and checks again for Notifications.

> In other cases, especially in high latency networks, the Client may wish to pipeline Publish requests
to ensure cyclic reporting from the Server. Pipelining involves sending more than one Publish
request for each Subscription before receiving a response. For example, if the network introduces a
delay between the Client and the Server of 5 seconds and the publishing interval for a Subscription
is one second, then the Client will have to issue Publish requests every second instead of waiting for
a response to be received before sending the next request.

jpfr, 2015-06-05:
> Or, if we have somehow managed to queue 4 of them and suddenly get 4 notificationmessages from 4 subscriptions. how would the server send these without a request by the client?

Those **are** the requests! Just think of the queued requests as delayed responses.

Stasik0, 2015-06-05:
@jpfr this can be a valid idea, in the swtich case - add message to queue in the session and return. Then on some event or time-trigger dequeue the message from the queue and pass it to whatever is now in the switch case i.e. service invocation etc. How do we prevent server from being overflown by the requests?

jpfr, 2015-06-05:
The time-triggered subscription does not need to get "back into the switch statement".
Since it knows the users session, it can just take the connection from there and do `connection->send(bytestring)` all by itself. Like a real grownup! :-)

There is a mechanism where the server refuses publish requests and returns an error code that the queue is full. That should stop the client from being too queueing-happy.

jpfr, 2015-06-07:
Ok. I just pushed the `UA_SecureChannel_sendBinaryMessage` to master and dev.
It unifies quite a bit of duplicated functionality in the server and the client

And now, we can send messages outside of the immediate request/response path. 
On only needs to know the requestId of the message (we are responding to).
However, the client does not yet demultiplex responses that come out-of-order.

So we have
- repeated callbacks with a time interval
- removing the callbacks when no longer needed
- out-of-order sending of responses

@ichrispa Do you need anything else for the subscriptions?

ichrispa, 2015-06-08:
Currently no. Though ultimately it would of course be cool if wrting access to the nodestore (change, statuschange) could be tracked and reported.

Stasik0, 2015-06-08:
@ichrispa, this can only be done for nodes (like node deletion). Content updates in variables need own "notification" methods we need to build in (which is roughly simple)

jpfr, 2015-06-08:
@ichrispa Ok. How do you want to have it? Should the node call "its" monitoreditems when it gets changed?

ichrispa, 2015-06-12:
@jpfr: On a change, `MonitoredItem_QueuePushDataValue(server, monitoredItem)` should be called. The monitoredItem will decide on it's own if a value needs to be queued, which attribute is targeted, if it has changed, etc. The only information necessary here is that the monitoredItems `monitoredNodeId` attribute corresponds to the node where the change occured.

I am happy to report the following progress on subscriptions in the dev branch:
* Subscriptions now use timedWork items on the serverside when a publishInterval of >5ms is specified. 
* Subscriptions can be modified and will use or destroy their timedWork items as appropriate.
* Updating a subscription/monitored Item when a publish request arrives is still the fallback if no timeWork item is responsible for this procedure.
* Clients have a rudimentary, interrupt-like mechanism for attaching methods to changed items. If a publishResponse is received and it contains changes to that item, the method will be called. The userspace is thereby spared any complexities of the creating/removing the subscriptions and monitored items.
* All that needs to be done to inform the client is calling `UA_Client_processPublishRx(client, response)`; this is conceptionally also possible for asynchronous received responses, though that matter is still unimplemented. The client can however fork a thread that calls publish in regular intervals (also untested).
* There is a client example that demonstrates subscriptions.

I would suggest we also merge the createNodes into the dev branch and extending the example to creating "the.answer" on the server using the client.

jpfr, 2015-06-12:
> I am happy to report the following progress on subscriptions in the dev branch:

Just great!
What features do we want to add before pulling subscriptions into master?

> I would suggest we also merge the createNodes into the dev branch and extending the example to creating "the.answer" on the server using the client.

Probably we can get remote node creation into master faster if we pull it as a separate branch. If we just want to add nodes from remote, the existing `UA_Client_addNodes` might do the job as well.

ichrispa, 2015-06-13:
As far as I can see, the changes in the client_createNodes branch are pretty much compatible with either master or dev or both. 

Subscriptions still lack several things, most notably:
- Not all NodeAttributes are monitorable yet.
- We have DataChangeItem-Type Notifications, but there is still no support for EventList and StatusChange (the 'other two' monitoring types).

If these two are not considered major bugs, I principally do not object to merging the current subscription state into master.

Stasik0, 2015-06-13:
This are great news!

Did you already have a look on subscriptions with valgrind? Last time i looked into it, there were some minor leaks.

jpfr, 2015-10-16:
The branch has landed long time ago.
Individual regression merit their own issue now.

closed 2015-10-16 by jpfr