open62541/open62541 #254: client.c cannot link to UA_calcSizeBinary?

ichrispa, 2015-05-26:
While I somehow got this to work on Friday, I broke it somehow and don't know whats wrong.

I am testing the AddNodes service in the client. I need to pack the NodeAttributes into an ExtensionObject. But calling either `UA_calcSizeBinary()` or `UA_encodeBinary()` (or their respective type-macros) always results in a link error:

     Linking C executable client
     CMakeFiles/client.dir/examples/client.c.o: In function `main':
     /home/ichrispa/work/svn/working_copies/open62541/examples/client.c:121: undefined reference  to `UA_calcSizeBinary'
     /home/ichrispa/work/svn/working_copies/open62541/examples/client.c:123: undefined reference to `UA_encodeBinary'

The symbol is defined in libopen62541.so as it should be:

     objdump -tT libopen62541.so  | grep calcSize
     ...
     0000000000009e60 l     F .text  00000000000002db              UA_calcSizeBinary
     ...

I also tried to define UA_calcSizeBinary as extern in the client.c, which changed nothing.

Any ideas?

ichrispa, 2015-05-26:
Fixable by adding `__attribute__ ((visibility ("default")))` (`UA_EXPORT`) to calcSizeBinary/encodeBinary, but I'm not sure this is ideal.

jpfr, 2015-05-26:
Objdump tells you it is a local symbol (the `l` in the second column).
Are you trying to link against a precompiled lib? Only the functions in the /include directory are exported. Try to add `UA_EXPORT` to the function signature in the header.

jpfr, 2015-05-26:
> Fixable by adding __attribute__ ((visibility ("default"))) (UA_EXPORT) to calcSizeBinary/encodeBinary, but I'm not sure this is ideal.

Jup. That's the way. We currently hide everything that is not in the /include directory.

ichrispa, 2015-05-26:
OK, will use this as a workaround and maybe add a "nicer" Node-Create proxy function to UA_Client.c later on.

closed 2015-05-26 by ichrispa