open62541/open62541 #255: Creating new nodes

ichrispa, 2015-05-27:
Hi everyone,

for our current project we need to add nodes using the client (variables, types, objects, etc.). I have fumbled around with it (fixing some minor issues in 03aeed8bd446e59fbdce9734844c9f19b7bb0183) could create new nodes with relative ease.

Adding nodes is principally possible thanks to @jpfr's work and I don't see a problem with it on the serverside. But the clientside is kind of unwieldly. I would like to write some sort of proxy-function for creating the [Object|Variable...]Attribute ExtensionObject along with the other attributes in ua_client.c, which also eliminated the current need to UA_EXPORT calcSizeBinary and encodeBinary. 

I imagine something like "copy that node" would proof usefull; the user simply creates a node and passes it to a function like `copyNodeToServer(NodeId* revisedId, NodeId* parent, NodeId* referenceType)`.

Is that a viable concept? How do other frameworks handle this?

jpfr, 2015-05-27:
Good idea to have some helpers in the client for that.
Afaik, we currently don't give entire nodes to the users. Only the nodeids and the services to pull the different attributes...

If I understood you correctly, the main difficulty is that the Attribute-structs for the nodes are "hidden" in an extensionobject in the server signature.
What about differentiating by the node type, `createVariableNode(UA_Server, UA_VariableAttributes, ..)`?

Can we develop this on a feature-branch an merge it into master soon? I guess this will become mature before the subscriptions.

ichrispa, 2015-05-27:
Is something like this ok? https://github.com/acplt/open62541/blob/dev/src/client/ua_client.c#L609

If you're ok with it, I'm going to add some proxies for Variable[Type]Nodes and Object[Type]Nodes and wrap the addNodes service into a new ENABLE_NODEMANAGEMENT option. 

My current exampleClient creates 'TheGreatNode' successfully, so I would slightly expand the example to create a Folder/Object and include a new answer, then write a new value to that answer.

Stasik0, 2015-05-27:
Please, bring it to a dedicated branch, otherwise we will never manage to merge clearly.

ichrispa, 2015-05-27:
@Stasik0: I branched master to client_createNodes and added my recent work from dev there; I left some bugfixes behind in the current dev branch, but the client is back to its pre-me state.

I can currently add objects and variables. I'd like to keep some options unexposed to the client - if the user really wants to change these he can still issue write requests for the attributes in question. If the design pattern is ok for you, I'll get the rest of the node types in place.

Stasik0, 2015-05-27:
Thank you for branching, yes its fine for me, @jpfr, r u ok with that too?

jpfr, 2015-05-27:
Great! Imho you don't even need to `#ifdef` the createnode functions. Those are standard services after all..

Concering the signatures in `ua_client.h`. Normally the userWriteMask should be different for every user. But we can't know about the users at that point.
Normally, there would be some per-node role-based access policy on the server. Like an active directory. But that is left to the different implementations.
That's a bit puzzling. Especially since the mask comes from the (standard-defined) struct...

ichrispa, 2015-05-27:
Feel free to remove the ifdefs ^^

Here's a weird one: When creating OBJECTTYPES (nodeclass 4), they always read as objects. The server adds an OjectType to its nodestore (traced from Wireshark all the way done to UA_NodeStore_insert); but when being read back, the server sends an OBJECT (nodeclass 1).  There's no problem with any of the predefined types though...

Does that sound like a familiar problem to anyone?

Stasik0, 2015-05-28:
@ichrispa everything is good, you just mixed functions up in client.c, node called "TheNewObjectType" is created using "UA_Client_createObjectNode" while node called "TheNewGreatNode" is created using "UA_Client_createObjectTypeNode". Therefore, "TheNewGreatNode" is a type node even in UaExpert

ichrispa, 2015-05-28:
Thank @Stasik0. That one earns me the fail of the week award.

jpfr, 2015-05-28:
We all have earned those at one point. :-)

jpfr, 2015-05-29:
> Feel free to remove the ifdefs ^^

Just did. ^^

Stasik0, 2015-05-29:
In client or server? Or both?

jpfr, 2015-05-29:
Currently on both.
- Removing the service has nearly no savings. Since we use the same functions to create the address space at startup
- Access rights (who is allowed to add nodes) is a whole different topic that needs a solid solution. Say for 0.3 and onward.

Stasik0, 2015-05-29:
Ack

ichrispa, 2015-06-08:
Is this feature closable/merge worthy? Or are there any improvements pending?

Stasik0, 2015-06-08:
is it working? if yes, i'd merge it in

ichrispa, 2015-06-18:
no, not working. I need to incorporate the API changes from master.

Stasik0, 2015-06-18:
okay, and then we can merge it asap :) before it divergates more

ichrispa, 2015-06-19:
@Stasik0: Fixed it :) It actually was a typo of mine I introduced during the merge causing the problem.

ichrispa, 2015-06-24:
Feature has been merged into master, considering the issue closed.

closed 2015-06-24 by ichrispa