open62541/open62541 #256: Errors when building with MS Visual Studio Enterprise 2015 RC

stephanhensel, 2015-06-03:
Hello everybody,

I tried to build open62541 (dev branch) with MS Visual Studio Enterprise 2015 RC. In CMake I enabled BUILD_DOCUMENTATION, BUILD_EXAMPLECLIENT and BUILD_EXAMPLESERVER. When I try to build in Visual Studio I get the errors shown in the block beneath.
I think there are two separate errors. At first in ua_types_encoding_binary.h the restrict keyword is not known by Visual Studio. Replacing restrict by __restrict in the whole project will fix the syntax errors (see also https://msdn.microsoft.com/de-de/library/5ft82fed.aspx). Secondly the unistd.h cannot be found (it is not available at Windows).
Can you please fix the occurring errors when building with Visual Studio or provide a workaround?
Thanking you in anticipation.

Best regards,
Stephan

```
Schweregrad	Code	Beschreibung	Projekt	Datei	Zeile
Warnung	C4244	"=": Konvertierung von "UA_DateTime" in "UA_UInt32", möglicher Datenverlust	open62541-object	H:\open62541\open62541\src\server\ua_securechannel_manager.c	75
Fehler	LNK1104	Datei "Debug\open62541.lib" kann nicht geöffnet werden.	client	H:\open62541\open62541\build\LINK	1
Fehler	LNK1104	Datei "H:\open62541\open62541\build\open62541-object.dir\Debug\ua_types_encoding_binary.obj" kann nicht geöffnet werden.	client_static	H:\open62541\open62541\build\LINK	1
Fehler	LNK1104	Datei "H:\open62541\open62541\build\open62541-object.dir\Debug\ua_types_encoding_binary.obj" kann nicht geöffnet werden.	open62541	H:\open62541\open62541\build\LINK	1
Warnung	C4996	'gethostbyname': Use getaddrinfo() or GetAddrInfoW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings	open62541-object	H:\open62541\open62541\examples\networklayer_tcp.c	529
Warnung	C4244	"=": Konvertierung von "UA_DateTime" in "UA_UInt32", möglicher Datenverlust	open62541-object	H:\open62541\open62541\src\server\ua_securechannel_manager.c	72
Warnung	C4244	"=": Konvertierung von "UA_DateTime" in "UA_UInt32", möglicher Datenverlust	open62541-object	H:\open62541\open62541\src\server\ua_securechannel_manager.c	105
Warnung	C4244	"=": Konvertierung von "UA_DateTime" in "UA_UInt32", möglicher Datenverlust	open62541-object	H:\open62541\open62541\src\server\ua_securechannel_manager.c	108
Warnung	C4244	"=": Konvertierung von "const UA_Double" in "UA_Int64", möglicher Datenverlust	open62541-object	H:\open62541\open62541\src\server\ua_session_manager.c	68
Warnung	C4244	"=": Konvertierung von "ULONGLONG" in "long", möglicher Datenverlust	open62541-object	H:\open62541\open62541\src\ua_types.c	192
Warnung	C4996	'gmtime': This function or variable may be unsafe. Consider using gmtime_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	open62541-object	H:\open62541\open62541\src\ua_types.c	216
Warnung	C4996	'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	open62541-object	H:\open62541\open62541\src\ua_types.c	234
Warnung	C4028	Formaler Parameter "3" unterscheidet sich von der Deklaration	open62541-object	H:\open62541\open62541\src\ua_types.c	715
Fehler	C2146	Syntaxfehler: Fehlendes ")" vor Bezeichner "offset"	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2061	Syntaxfehler: Bezeichner "offset"	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ";"	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ")"	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ","	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2146	Syntaxfehler: Fehlendes ")" vor Bezeichner "offset"	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2061	Syntaxfehler: Bezeichner "offset"	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ";"	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ")"	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ","	open62541-object	h:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2146	Syntaxfehler: Fehlendes ")" vor Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2061	Syntaxfehler: Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ";"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ")"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ","	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2146	Syntaxfehler: Fehlendes ")" vor Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2061	Syntaxfehler: Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ";"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ")"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ","	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2146	Syntaxfehler: Fehlendes ")" vor Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2061	Syntaxfehler: Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ";"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ")"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ","	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2146	Syntaxfehler: Fehlendes ")" vor Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2061	Syntaxfehler: Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ";"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ")"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ","	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2146	Syntaxfehler: Fehlendes ")" vor Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2061	Syntaxfehler: Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ";"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ")"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ","	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2146	Syntaxfehler: Fehlendes ")" vor Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2061	Syntaxfehler: Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ";"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ")"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ","	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2146	Syntaxfehler: Fehlendes ")" vor Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2061	Syntaxfehler: Bezeichner "offset"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ";"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ")"	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
Fehler	C2059	Syntaxfehler: ","	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	38
[... 1466 lines cut ...]
Fehler	C1003	Mehr als 100 Fehler gefunden; Kompilierung wird abgebrochen.	open62541-object	H:\open62541\open62541\src\ua_types_encoding_binary.h	58
[... cut ...]
Fehler	LNK1104	Datei "H:\open62541\open62541\build\open62541-object.dir\Debug\ua_types_encoding_binary.obj" kann nicht geöffnet werden.	server_simple	H:\open62541\open62541\build\LINK	1
Warnung	C4996	'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	server_simple	H:\open62541\open62541\examples\server_simple.c	35
Fehler	C1083	Datei (Include) kann nicht geöffnet werden: "unistd.h": No such file or directory	server_static	H:\open62541\open62541\examples\server.c	11

```

Stasik0, 2015-06-03:
@stephanhensel Hi, the problem is that, at least i, have to testing environment for MSVS. Could you try replacing unistd.h by io.h? Could you also try to apply your fix? You can do it both in a fork, and then apply for a pull-request (https://help.github.com/articles/using-pull-requests/)

Stasik0, 2015-06-03:
we continue to discuss in PR

closed 2015-06-03 by Stasik0