open62541/open62541 #262: Message buffer not reused and instead allocated for each transaction

pfpmeijers, 2015-06-09:
For each transaction the message buffer looks to be allocated (and  freed afterwards) instead of being reused. Because the message buffer size is fixed (fixed maximum), it sounds more efficient to reuse the same memory block.

See e.g. valgrind output of a test doing multiple read transactions:
==36267== 
==36267== HEAP SUMMARY:
==36267==     in use at exit: 0 bytes in 0 blocks
==36267==   total heap usage: 78,782 allocs, 78,782 frees, 1,114,503,121 bytes allocated
==36267== 
==36267== All heap blocks were freed -- no leaks are possible
==36267== 
==36267== For counts of detected and suppressed errors, rerun with: -v
==36267== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
More than 1 GB is allocated in total.

jpfr, 2015-06-09:
Hey @pfpmeijers 

that's a good point. Currently, the network layer provides a callback to get buffer memory and to release it. So we can change that without even touching the library itself.

Anybody wants to take this up? I'm blocked until next week...

jpfr, 2015-06-09:
By the way, those allocations should be very cheap. The libc gets a block of memory (a few MB) from the operating system and handles that all in userspace.
In a sense, we already reuse memory again and again...

pfpmeijers, 2015-06-10:
Typically doing constant reallocations (free + malloc) could lead to memory
fragmentation, in particular when other memory requests can come in between
 Though I don't know whether that is the case here.
Grt, Patrick

jpfr, 2015-06-13:
Implemented in e0f7d135b9288da0c6b9c4468aa67189bcdcfb22.
This actually increases the memory requirements at first, since we allocate the full 65kb of the maximum message size.

Thx for raising the topic @pfpmeijers.

pfpmeijers, 2015-06-15:
I cloned the latest version and rebuild, but unfortunately I now get
UA_STATUSCODE_BADINTERNALERRO for not that huge small transaction sizes
(array size > 2000 ints).
Grt, Patrick

Stasik0, 2015-06-15:
@pfpmeijers could you make your code accessible at github or somewhere else?

pfpmeijers, 2015-06-15:
I am not that familiar with github mechanisms. I will send you a zip file.

Stasik0, 2015-06-15:
@pfpmeijers okay, or just upload it to some file-hoster and link it here

pfpmeijers, 2015-06-15:
See attached zip file (renamed to .zi_)

The open62541.c/h files are build from todays snapshot.
The .org variants are those that were present on the website main page
(currently these links are dead).

The .org variant is ok. The new ones give following error on windows:

>start server.exe
>client.exe -size 512 -repeat 100000
ERROR: Read failed. Service result is not good: 0x80020000

A small number of measurent repetitions (indicated by the -repeat) is ok.
When increasing the repeat count it fails at certain point.
But only on Windows. On linux I cannot reproduce the error.
Feels like memory corruption.

Grt, Patrick

pfpmeijers, 2015-06-15:
All blocked by company policy.

Stasik0, 2015-06-15:
send us the code as plaintext by using a gist (https://gist.github.com/)... how do you compile on windows? VS or mingw

Stasik0, 2015-06-15:
btw @pfpmeijers there was a memleak in client yesterday, are you sure to take todays snapshot? Like this one? http://open62541.org/releases/2410369656.zip

pfpmeijers, 2015-06-15:
I will test again with this zip file tomorrow.

pfpmeijers, 2015-06-15:
https://gist.github.com/anonymous/56559890cd076ad0d886

jpfr, 2015-06-15:
That's some nice code there!

I just pushed some changes to master. The benchmark works fine on my box.
The latest single-file release from master is in this folder: http://open62541.org/releases/

Once the messages are too big, we return UA_STATUSCODE_REQUESTTOOLARGE.
Once we have message chunking, the maximum message size will increase a lot.
But there will always remain an upper limit for message sizes.

jpfr, 2015-06-15:
As a side note. With open62541, you will mostly measure a mixture of roundtrip-time and throughput of the network interface. The server is much faster than the network interface (that require a context switch into the kernel).

On my windows box I get 25,000 requests per sec on a single core.
With a more lightweight (non-POSIX) network API, we could probably increase this 5-10 times.

http://blogs.technet.com/b/wincat/archive/2012/12/05/fast-tcp-loopback-performance-and-low-latency-with-windows-server-2012-tcp-loopback-fast-path.aspx

pfpmeijers, 2015-06-16:
The latest version (http://open62541.org/releases/b44a64a8fc.zip) still
gives me 0x80020000 errors.
Added two python scripts to the gist that run various use cases with the
client app: https://gist.github.com/pfpmeijers/e35e1bd74fbef2d165c9
If you run "run_measurements.py" I expect you see these errors passing by
rather quickly.

pfpmeijers, 2015-06-16:
"But there will always remain an upper limit for message sizes."
You mean a hardcoded upper limit, or something I can configure using the
config structs with message size/buffer size fields?

pfpmeijers, 2015-06-16:
I would be very interested to see what level of performance could be
reached with this non-POSIX API.

Stasik0, 2015-06-16:
@pfpmeijers on linux the script gives 0x80020000 only on buffer sizes > 65k which is expected.. are you having problems on the windows side? can you show us a cvs of such a file?

@pfpmeijers cool scripts, which license are you having on them? i.e. may we use it as a benchmark?

pfpmeijers, 2015-06-16:
No license, so free to use. The client.c file actually needs some
refactoring to remove the #ifdef spagetti.

I select a buffer size > 65K in the client and server config. Doesn't this
help? These value are ignored?

Stasik0, 2015-06-16:
nono, I just used the vanila open62541.c/.h files, so 65k are set there... so you say it work for windows, but not for linux?

pfpmeijers, 2015-06-16:
Did not try for windows yet. Today did a test run on linux system,
intra-host, so no network traffic. Will show you some results tomorrow.

jpfr, 2015-06-17:
> I would be very interested to see what level of performance could be reached with this non-POSIX API.

That depends on the effort you are willing to make.
On Linux (with increasing difficulty) there are
- netmap (http://info.iet.unipi.it/~luigi/netmap/). This gives you access to a ringbuffer in kernel-memory and thus zero-copy messaging. Code changes are straightforward. The improvement over vanilla TCP sockets is 10-20x for small packets.
- http://dpdk.org/ They reimplement much of the network stack and drivers in userland to save syscalls. Requires more work to integrate. The website claims up to 160 million frames/sec on a beefy server.

I have zero time until next week. Then, I'd try to get netmap running on a branch of our tcp networking layer.

Stasik0, 2015-07-20:
@pfpmeijers is the issue solved?

pfpmeijers, 2015-07-23:
The original issue (memory buffer not reused) is indeed solved, so can be closed from my point of view.
In the discussion some other items are mentioned, but I think these are addresses by different issue numbers.

closed 2015-07-23 by pfpmeijers