open62541/open62541 #263: check python install

HolgerJeromin, 2015-06-09:
open62541 needs a python install (always?) so it would be nicer to check the availability somewhere. Otherwise you get a ugly abort from make.
This is especially important for windows...

Stasik0, 2015-06-09:
You need em to generate code from xml files. Checking for py in cmake is easy, does anyone have a snipplet for checking for lxml?

HolgerJeromin, 2015-06-10:
Thank you @bilach and Stasik0!

Stasik0, 2015-06-10:
@ichrispa is your srcipt runnable with Pyhton 3? or is it 2.7 only?

HolgerJeromin, 2015-06-10:
@Stasik0: did you meant chris or bilach?
My windows machine has this output:

```
Found PythonInterp: C:/Program Files (x86)/python/python.exe (found version "3.4.3") 
  File "<string>", line 1
    from distutils.sysconfig import get_python_lib; print get_python_lib()
                                                                       ^
SyntaxError: invalid syntax

CMake Error at CMakeLists.txt:11 (message):
  Python-lxml is not installed.
```
So whole thing is not optimal right now :-)
We probaby should use `PYTHON_EXECUTABLE`.

Stasik0, 2015-06-10:
i meant @ichrispa and his ns0 generation script, they may not run with py3 at all. 
@HolgerJeromin could you tweak a bit on the script in CMakeLists?

PR #266 (2015-06-10): using ${PYTHON_EXECUTABLE} and a better check for lxml.
this could fix #263 with python 3 and python 2.x
closed 2015-06-10 by HolgerJeromin