open62541/open62541 #270: BADDECODING error when large message is not complete

pfpmeijers, 2015-06-10:
In case of sending large messages (e.g. large arrays) then BADDECODING errors occur.

This is a result from the function UA_Connection_completeMessages() which at open62541.c, line 5386, resets the returned buffer length to -1, in case the message is not complete yet:
    if(end_pos == 0) {
        /* no complete message in current */
        connection->incompleteMessage = current;
        UA_ByteString_init(&current);
    }

The following change looks to fix the problem, but don't know if this is the preferred solution, or working in all cases.
In open62541.c, line 10485, the loop around the client->connection.recv() call, part of the sendReceiveRequest() function, I changed:
   while(retval != UA_STATUSCODE_GOOD);
into 
  while(retval != UA_STATUSCODE_GOOD  || reply.length == -1);

pfpmeijers, 2015-06-11:
The fix as mentioned in previous comment is not ok in general. 
If you request a non-existing end point, then the loop will block forever instead of the client returning "No suitable endpoint found". 
Some timeout mechanism to be added?

Stasik0, 2015-06-11:
Isnt the -1 coming from the timeout, already? Do we set SO_RCVTIMEO?

pfpmeijers, 2015-06-11:
SO_RCVTIMEO doesn't ring a bell to me.
The -1 is set in the code as I listed in the issue description - because
the received data is not complete yet.
Grt, Patrick

ichrispa, 2015-06-11:
@pfpmeijers As far as I know the server does not support chunking. If you message is too large, the tcp transport will segment the data without propperly chunking the message. The result is something akin to the error you mentioned.

Use wireshark or another packet analyzer to check if this occurs.

Stasik0, 2015-06-18:
@pfpmeijers is it considered solved now?

pfpmeijers, 2015-06-18:
I still have the problem/error code.
Using the a4cd3c575b.zip release now.
Problem also occurs for smaller array sizes, but not fully
deterministically, it looks like.

On one PC we get:

client_executable: ./opcClient, client_count: 1, array_size: 512,
transaction_size: 1, transaction_bytes: 2048
ERROR: Client 0 transactions failed: ERROR: Read failed. Service result is
not good: 0x80020000

On other PC (virtual) we see this case going ok. Failures start at larger
array sizes:

ERROR: Client 0 transactions failed: ERROR: Read failed. Service result is
not good: 0x80020000
Retrying ...
client_executable: ./opcClient, client_count: 1, array_size: 32768,
transaction_size: 1, transaction_bytes: 131072,

--> Retry with same parameters goes ok.

I know the 131072 bytes is more than the 65536 you mentioned as supported
buffer size.
But I specify a sufficiently large buffer size in the client and server.
If internally there is still a 65536 limitation, then I would expect an
error code indicating this, deterministically.

One thing that pops up my mind is that I select buffer size to be equal to
payload data per transaction plus a fixed number of 1024 bytes for
overhead. Maybe this overhead is not sufficient?

Anyhow, it does not explain the error for array size 512, which requires
2048 bytes data and thus should fit easily in the 65536 buffer.

Grt, Patrick

Stasik0, 2015-06-18:
which array size do you have? how can we reproduce it?

pfpmeijers, 2015-06-18:
Run this script (with my client/server apps and multi_client.py script, as
I uploaded into the gist):

import os

for client in ["./opcClient"]:
  for client_count in [1 << i for i in range(1)]:
    for transaction_size in [1 << i for i in range(1)]:
      for array_size in [1 << i for i in range(20)]:

        os.system("python multi_client.py"
                  " -client "  + client +
                  " -clients " + str(client_count) +
                  " -size "    + str(array_size) +
                  " -vars "    + str(transaction_size) +
                  "")

Grt, Patrick

Stasik0, 2015-06-18:
@jpfr do you have a clue, Julius? I cannot reproduce it here (32-bit linux) all array until 32768 perform good, 65536 fails as expected

pfpmeijers, 2015-06-18:
@sten, you can also not reproduce with my script/test app? (I use linux 64
bits.)

pfpmeijers, 2015-06-18:
Just tried again on windows (10, x64):

Z:\OPC_UA\open62541\perf_test>client -size 524288 -repeat 10
10 transactions, 0.469109 seconds, 2097152 B/transaction, 44.705 MB/s,
latency min/avg/max: 0.042808/0.0516574/0.0762038 seconds

Z:\OPC_UA\open62541\perf_test>client -size 1048576 -repeat 10
ERROR: Read failed. Service result is not good: 0x80020000

So I can go up to 2MB per transaction (single array of int32).
Maybe by accident?

Grt, Patrick

ichrispa, 2015-06-18:
I'm gonna give this another try: We don't support chunking as far as I know. If you are testing this over a loopback adapter, then ip/tcp will split the packets at several kilobyte intervals (not necessarily always the same).The second piece of the packet will be missing its transport and security headers, which results in a failed decoding.

The MTU is defined by the device driver, not the tcp/ip parameters.  You can increase the loopback MTU using 'ifconfig lo mtu <x>' on linux/bsd (check windows documentation for ipconfig).

Could you please check and/or provide a wireshark/tcpdump trace of these transactions?

Stasik0, 2015-06-18:
@ichrispa still we should be able to coupe with the split, we should be safe under 65k! Since it may occur anytime - TCP is stream oriented

Stasik0, 2015-06-18:
@pfpmeijers i think you were right. This is exactly the case when a "partial" message was received, but "hold back" by the ```UA_Connection_completeMessages``` function
@pfpmeijers would you test this one http://open62541.org/releases/1ca92d05db.zip which is a build from the "network_fix" branch https://github.com/acplt/open62541/tree/network_fix

ichrispa, 2015-06-18:
@Stasik0: I agree with you completely on the TCP specs. It should even be capable of handling arbitrary length transactions even though buffers are limited. But to debug this I want to see what the client and server are exchanging and whether this is a server/client fault or the packages grow larger then the buffers negotiated during HEL/ACK. I would also be interested to see if TSO is used, which is driver specific and may explain why we are having problems reproducing this.

For example:
> 10 transactions, 0.469109 seconds, 2097152 B/transaction, 44.705 MB/s

That's a 2MB transaction - the second one would thus be an estimated 4MB/transaction. Was this a single package operation? What were the negotiated parameters? What did the server actually send?

pfpmeijers, 2015-06-19:
@Sten, I will try the network_fix branch version and let you know the resul.

pfpmeijers, 2015-06-19:
@ichrisp <https://github.com/ichrispa>: unfortunately i am not familiar
enough with TCP/IP communication technology to understand what you mean
with MTU and other abbreviations. I also have no experience with
wireshark/tcpdump. So if i need to do a test, then you need to give me
detailed instructions like what to install, which commands to run etc.
On the other hand, since I posted my test apps and scripts, it might be
more efficient to build that locally and try to reproduce yourself.

pfpmeijers, 2015-06-19:
@Sten, is this release 1ca92d05db ? (then I do not have to generate the
single file versions myself)

Stasik0, 2015-06-19:
yep: http://open62541.org/releases/1ca92d05db.zip

pfpmeijers, 2015-06-19:
With release a4cd3c575b: from array size of 512 int32 items onward,
*non-*deterministically
BADINTERNALERROR:

patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100
1000 transactions, 0.983501 seconds, 4 B/transaction, 0.0040671 MB/s,
latency min/avg/max: 0.000593597/0.000981233/0.0219491 seconds
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 512
1000 transactions, 1.20215 seconds, 2048 B/transaction, 1.70362 MB/s,
latency min/avg/max: 0.000744148/0.00119903/0.00756372 seconds
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 512
ERROR: Read failed. Service result is not good: 0x80020000
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 512
1000 transactions, 1.04549 seconds, 2048 B/transaction, 1.95889 MB/s,
latency min/avg/max: 0.000743433/0.00104485/0.00431626 seconds
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 512
1000 transactions, 1.31141 seconds, 2048 B/transaction, 1.56168 MB/s,
latency min/avg/max: 0.00085758/0.0013073/0.0136729 seconds
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 512
1000 transactions, 1.15964 seconds, 2048 B/transaction, 1.76606 MB/s,
latency min/avg/max: 0.000740842/0.00115705/0.0230341 seconds
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 512
ERROR: Read failed. Service result is not good: 0x80020000
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./make.sh
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 512
1000 transactions, 1.10164 seconds, 2048 B/transaction, 1.85904 MB/s,
latency min/avg/max: 0.000707011/0.00109872/0.00479155 seconds
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 512
1000 transactions, 1.09615 seconds, 2048 B/transaction, 1.86836 MB/s,
latency min/avg/max: 0.000656951/0.00110105/0.0223733 seconds
patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 512

With release  1ca92d05db files the BADINTERNALERROR looks to be gone
(good!) and now deterministic errors from transactions above 64kB.

If I use a single array above 64 kB, then BADDECODINGERROR:

patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 32768
ERROR: Read failed. Service result is not good: 0x80070000

Note that size is in int32 items, not in bytes.
Occasionally, when doing a next read transaction after this 0x80070000
error, then the client hangs.

If I use multiple variables per transaction, then BADOUTOFMEMORY:

patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 8192 -vars 4
ERROR: Read failed. Service result is not good: 0x80030000

Grt, Patrick

pfpmeijers, 2015-06-19:
With 64 arrays per transaction my client crashes.

patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 1 -vars 64
ERROR: Read failed. Service result is not good: 0x80AE0000
*** Error in `./opcClient': double free or corruption (!prev):
0x0000000001b40310 ***
Aborted (core dumped)

The -g version just gives an error:

patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 1 -vars 64
ERROR: Read failed. Service result is not good: 0x80AE0000

but with valgrind:

patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ valgrind
--leak-check=yes ./opcClient -server 192.168.0.100 -size 1 -vars 64
==5307== Memcheck, a memory error detector
==5307== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==5307== Using Valgrind-3.10.0.SVN and LibVEX; rerun with -h for copyright
info
==5307== Command: ./opcClient -server 192.168.0.100 -size 1 -vars 64
==5307==
ERROR: Read failed. Service result is not good: 0x80AE0000
==5307== Invalid read of size 8
==5307==    at 0x41D30D: ClientNetworkLayerGetBuffer (open62541.c:11679)
==5307==    by 0x40A07E: UA_SecureChannel_sendBinaryMessage
(open62541.c:5632)
==5307==    by 0x41AA99: synchronousRequest (open62541.c:10848)
==5307==    by 0x41B4C8: CloseSession (open62541.c:11012)
==5307==    by 0x41BAD8: UA_Client_disconnect (open62541.c:11098)
==5307==    by 0x40238C: main (client.c:574)
==5307==  Address 0x51fc3b8 is 8 bytes inside a block of size 16 free'd
==5307==    at 0x4C2BDEC: free (in
/usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==5307==    by 0x402CFB: UA_String_delete (open62541.c:2004)
==5307==    by 0x41D357: ClientNetworkLayerClose (open62541.c:11695)
==5307==    by 0x41C2A3: socket_recv (open62541.c:11247)
==5307==    by 0x41AB15: synchronousRequest (open62541.c:10864)
==5307==    by 0x41BB85: UA_Client_read (open62541.c:11110)
==5307==    by 0x401A19: run_transactions (client.c:306)
==5307==    by 0x402317: main (client.c:563)
==5307==
==5307== Invalid read of size 8
==5307==    at 0x41D311: ClientNetworkLayerGetBuffer (open62541.c:11679)
==5307==    by 0x40A07E: UA_SecureChannel_sendBinaryMessage
(open62541.c:5632)
==5307==    by 0x41AA99: synchronousRequest (open62541.c:10848)
==5307==    by 0x41B4C8: CloseSession (open62541.c:11012)
==5307==    by 0x41BAD8: UA_Client_disconnect (open62541.c:11098)
==5307==    by 0x40238C: main (client.c:574)
==5307==  Address 0x51fc3b0 is 0 bytes inside a block of size 16 free'd
==5307==    at 0x4C2BDEC: free (in
/usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==5307==    by 0x402CFB: UA_String_delete (open62541.c:2004)
==5307==    by 0x41D357: ClientNetworkLayerClose (open62541.c:11695)
==5307==    by 0x41C2A3: socket_recv (open62541.c:11247)
==5307==    by 0x41AB15: synchronousRequest (open62541.c:10864)
==5307==    by 0x41BB85: UA_Client_read (open62541.c:11110)
==5307==    by 0x401A19: run_transactions (client.c:306)
==5307==    by 0x402317: main (client.c:563)
==5307==
==5307== Invalid write of size 1
==5307==    at 0x4064C7: UA_Byte_encodeBinary (open62541.c:3328)
==5307==    by 0x4070FD: UA_NodeId_encodeBinary_nodeid_offset
(open62541.c:3611)
==5307==    by 0x4072FB: UA_NodeId_encodeBinary (open62541.c:3648)
==5307==    by 0x40A0AE: UA_SecureChannel_sendBinaryMessage
(open62541.c:5637)
==5307==    by 0x41AA99: synchronousRequest (open62541.c:10848)
==5307==    by 0x41B4C8: CloseSession (open62541.c:11012)
==5307==    by 0x41BAD8: UA_Client_disconnect (open62541.c:11098)
==5307==    by 0x40238C: main (client.c:574)
==5307==  Address 0x18 is not stack'd, malloc'd or (recently) free'd
==5307==
==5307==
==5307== Process terminating with default action of signal 11 (SIGSEGV)
==5307==  Access not within mapped region at address 0x18
==5307==    at 0x4064C7: UA_Byte_encodeBinary (open62541.c:3328)
==5307==    by 0x4070FD: UA_NodeId_encodeBinary_nodeid_offset
(open62541.c:3611)
==5307==    by 0x4072FB: UA_NodeId_encodeBinary (open62541.c:3648)
==5307==    by 0x40A0AE: UA_SecureChannel_sendBinaryMessage
(open62541.c:5637)
==5307==    by 0x41AA99: synchronousRequest (open62541.c:10848)
==5307==    by 0x41B4C8: CloseSession (open62541.c:11012)
==5307==    by 0x41BAD8: UA_Client_disconnect (open62541.c:11098)
==5307==    by 0x40238C: main (client.c:574)
==5307==  If you believe this happened as a result of a stack
==5307==  overflow in your program's main thread (unlikely but
==5307==  possible), you can try to increase the size of the
==5307==  main thread stack using the --main-stacksize= flag.
==5307==  The main thread stack size used in this run was 8388608.
==5307==
==5307== HEAP SUMMARY:
==5307==     in use at exit: 743 bytes in 7 blocks
==5307==   total heap usage: 113 allocs, 106 frees, 467,495 bytes allocated
==5307==
==5307== LEAK SUMMARY:
==5307==    definitely lost: 0 bytes in 0 blocks
==5307==    indirectly lost: 0 bytes in 0 blocks
==5307==      possibly lost: 0 bytes in 0 blocks
==5307==    still reachable: 743 bytes in 7 blocks
==5307==         suppressed: 0 bytes in 0 blocks
==5307== Reachable blocks (those to which a pointer was found) are not
shown.
==5307== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==5307==
==5307== For counts of detected and suppressed errors, rerun with: -v
==5307== ERROR SUMMARY: 3 errors from 3 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)

On windows10 the client runs ok, up to arrays of 256. With 64 arrays of 256
int32 items (64KB) it hangs without message.

pfpmeijers, 2015-06-19:
Btw, with older release (a4cd3c575b), on windows10, the 64 arrays of 256
items gives:

Z:\OPC_UA\open62541\perf_test>client -size 256 -vars 64 -repeat 10
ERROR: Read failed. Service result is not good: 0x80020000

instead of hanging.

On linux with older release no memory error.

patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ valgrind
--leak-check=yes ./opcClient -server 192.168.0.100 -size 256 -vars 64
-repeat 10
==8210== Memcheck, a memory error detector
==8210== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==8210== Using Valgrind-3.10.0.SVN and LibVEX; rerun with -h for copyright
info
==8210== Command: ./opcClient -server 192.168.0.100 -size 256 -vars 64
-repeat 10
==8210==
ERROR: Read failed. Service result is not good: 0x80020000
==8210==
==8210== HEAP SUMMARY:
==8210==     in use at exit: 0 bytes in 0 blocks
==8210==   total heap usage: 118 allocs, 118 frees, 806,543 bytes allocated
==8210==
==8210== All heap blocks were freed -- no leaks are possible
==8210==
==8210== For counts of detected and suppressed errors, rerun with: -v
==8210== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

Conclusion: network_fix release solves the BADINTERNALERROR.
But it has a memory issue for multiple nodes (>= 64) in a single read.

Stasik0, 2015-06-19:
i can not reproduce it.... 

sten@gemini:~/Desktop/gist/e35e1bd74fbef2d165c9$ valgrind ./client.bin -size 1 -vars 64
==6275== Memcheck, a memory error detector
==6275== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==6275== Using Valgrind-3.10.0.SVN and LibVEX; rerun with -h for copyright info
==6275== Command: ./client.bin -size 1 -vars 64
==6275== 
1000 transactions, 0.707279 seconds, 256 B/transaction, 0.361951 MB/s, latency min/avg/max: 0.000527962/0.000630388/0.0197745 seconds
==6275== 
==6275== HEAP SUMMARY:
==6275==     in use at exit: 0 bytes in 0 blocks
==6275==   total heap usage: 66,155 allocs, 66,155 frees, 69,844,564 bytes allocated
==6275== 
==6275== All heap blocks were freed -- no leaks are possible

but the non-deterministic error is gone, right? so it has got better, should i merge this fix into master? are we running the same code? what has changed?

pfpmeijers, 2015-06-19:
Did you use my app? Latest code is in
https://gist.github.com/pfpmeijers/e35e1bd74fbef2d165c9 (latest client code
changes are EPICS only).
Build with make.sh or make.bat
Run server on different system - the error only pops up for remote server
(tested linux64 to linux64):

patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
localhost -size 1 -vars 64 -repeat 10
10 transactions, 0.00245885 seconds, 256 B/transaction, 1.04114 MB/s,
latency min/avg/max: 0.000149957/0.000687806/0.00447691 seconds

patrick@ubuntu:~/Shares/OPC_UA/open62541/perf_test$ ./opcClient -server
192.168.0.100 -size 1 -vars 64 -repeat 10
ERROR: Read failed. Service result is not good: 0x80AE0000
*** Error in `./opcClient': double free or corruption (!prev):
0x000000000089b310 ***
Aborted (core dumped)

From our perspective you can merge it in. Deterministic behavior is better.
What is left is the mem error and the fact that for transaction sizes >
64kB the error messsages are not what they should be (BADDECODING,
OUTOFMEMORY) instead of something like NOTSUPPORTED.

Thanks, Patrick

Stasik0, 2015-06-19:
the question is why does the memory message occur? I mean, thy does 0x80AE0000 appear. I need to test it on different machines... but first i will merge the commit in

Stasik0, 2015-06-19:
0x80AE0000 can appear if a timeout occurs, you have currently the timeout set to 5ms... are you sure it is enough? I pushed some memory fixes for the client, the double free should not occur

Stasik0, 2016-02-09:
issue expired

closed 2016-02-09 by Stasik0