open62541/open62541 #271: dev branch client does not set requestHeader infos

ichrispa, 2015-06-11:
Hi all,

I was wondering why the client in the dev branch can't create a subscription... and some further looking into the problem showed that the client can't really do anything.

Everyting after the activateSessionRequest has stopped filling in the request header, resulting in a statuscode BADSESSIONINVALID being returned by the server. Also, the client now SEGFAULTS on linux x64 when the connection fails. I can't find the reason for these looking through the code, but I know these errors where not there some weeks ago.

If anyone can reproduce these errors, I would kindly request @jpfr to look into the changes introduced with 3220a651db3793e02f46307ac7dc2a6a854ac399 and the subsequent merge.

Best Regards,

Chris

Stasik0, 2015-06-11:
i think i fixed them in b2f2dd9adce3c150202a1852a7bc71f4e78570fa and 5e665ec563c3cf4b344349ee77e107d5773d1ce9, you can just cherry pick them to dev

ichrispa, 2015-06-11:
Fixed by merging master back into dev.

closed 2015-06-11 by ichrispa