open62541/open62541 #272: A GUI?

ichrispa, 2015-06-12:
Dear all,

we had a little discussion today with a colleague of mine which revolved around the issue of not having a propper, free and inter-plattform UA modelling tool. So I hacked away for the past couple of hours to see if I can harness GUIs, in particular the drawing and boundary-detection of objects on a surface (which is kind of a new thing to me). Here's a bit of a mockup:

[image]

I think you might suspect where I am going with this... I want to chain my pyUANamespace behind a GUI for drawing and editing namespaces in an UML like fashion. The mockup shows this is possible and a somewhat presentable GUI should be codable within a few weeks (at least to demonstrate the concept). Here are some considerations:
* No additional software dependancy should be introduced by the GUI (java, perl, ...) just because it "can do graphics". I used Python for the classes, now I'll have to stick with it.
* No addtional transpiling steps (e.g. Visio->XML-(transpile)->UA XML->(transpile)->open62541 Header). I would like a tool that can export to open62541 header format directly. 
* No modification to pyUANamespace. The classes there are set and are fullfilling all UA related tasks. The GUI is strictly bound to the model by proxies, cleanly separated from the UA model.
* Ability to import XML files and generate a viewable graph, hide nodes (generate ignore lists), etc.

Before I delve any further into this topic, I'd like to ask your opinion: Is this a standalone project or would you like to assist/improve/reuse such an undertaking. There are currently two UA Modelers, one by UA (which not interactively draw nodes) and one by the fraunhofer iosb by the same name, which costs several thousand dollars to licence and is nothing but undeveloped beta software. So I would speculate that a free, opensource, redistributable, cross-plattform UA modelling GUI application has a market. But on the other hand, I will need help with this...

Thoughts?

jpfr, 2015-06-13:
As you mentioned, IOSB has developed such a modeller some years ago.
I'd like to avoid a conflict of interests.
Can we keep such an initiative outside of the main repository?

ichrispa, 2015-06-13:
The iosb has developed a specialized graphical modelling tool for generating c/c++ targeted at the unified automation sdk.

From my point of view, there is no conflict of interest. Their tool targets the UA SDK, this one would target open62541. There is about as much conflict potential here as in creating an open source opc ua stack inspite of a commercial one being available from ua.

Stasik0, 2015-06-13:
@jpfr works at ISOB ;)
From my point of view, I like the idea, and it seems to be a logical continuation of your script (all backend is already there). Unfortunately, I do not have so much Pyhton competence....  We are also looking into UnifiedAutomation modelling tool ATM which seems to fulfill our current needs.

ichrispa, 2015-06-13:
Ok, with @jpfr working for the IOSB I can definetely understand the conflict of intereset argument. 

I really want to avoid multiple transfomation steps when generating a model for our server; we also talked over schemes to modify YEd, Dia or even use MS Visio, then transfrom their save files to the UA XML scheme. 

I would like a tool that integrates with the open62541 server/client properties, can run DRC checks on the model to point out problems or generate specific hooks for subscription handling or method calls.

The main overhead in this tool would be the graphics (I still have problems with zooming for example). But once these are solved, I would mostly need feedback concerning the UI design, desired features and such. So the support effort I am looking for is less on the coding side, but more on the testing and usage feedback. As @Stasik0 pointed out, the main backend is already there.

uleon, 2015-06-13:
Is there really a need for just another modeller? I see a lot of tools, some of them quite elaborated and almost all of them export to xml-files, e.g. 

http://www.commsvr.com/DownloadCenter/ProductSoftware/UAModelDesigner.aspx
http://www.hb-softsolution.com/opc_ua/?site=products3&lang=de&title=Comet%20UA%20Model%20Designer
https://www.prosysopc.com/products/opc-ua-modeler/

Wouldn't it be much more efficient to provide migration pathes to open62541, that is, a set of python scripts or XSLT scripts that generate the open62541-xml-files?

Stasik0, 2015-06-13:
@uleon since opc foundation has a standardized xml schema, no transformations are needed - Chris' script should parse it right away

uleon, 2015-06-13:
true enough :-)  I simply don't know if the above mentioned modellers export to OPC F's xml schema...

ichrispa, 2015-06-13:
One of the problems I see is that none of them are available as open source; you need to register and/or buy them. If you don't buy them, there will propably be restrictions on commercial use. This one would maybe not be as elaborate, but it would be freely available.

Additionally, we could leverage from the community to improve them. If you ask several thousand dollars of licencing costs, the amount of feedback is limited to those select few who can afford the product.

ichrispa, 2015-06-19:
OK, I got all the feedback I needed from this discussion (noone wants a gui for the python classes but me); I'll be working on this on my free time. Considering this issue closed.

closed 2015-06-19 by ichrispa