open62541/open62541 #273: Malformed/Unexpected Response to CreateSubscription

kevinherron, 2015-06-13:
I know subscriptions aren't supported, but in response to a CreateSubscription service call the server sends back a bad response - apparently just a ResponseHeader. I'm assuming you mean to be returning a ServiceFault containing that ResponseHeader.

Further, after the client reconnects and sends an ActivateSessionRequest, the server segfaults.

Stasik0, 2015-06-13:
Which branch/Version are you using? Which client? How can we reproduce it?

kevinherron, 2015-06-13:
I am using the 32-bit linux binary available on the front page of http://open62541.org/ - so I guess that makes it v0.1?

Running [this example](https://github.com/digitalpetri/opc-ua-sdk-examples/blob/master/src/main/java/com/digitalpetri/opcua/sdk/examples/client/SubscriptionExample.java) (after modifying the endpoint URL to point at open62541) is enough to induce the crash.

Stasik0, 2015-06-13:
Have you eventually got a jar for me so i can test quickly?

kevinherron, 2015-06-13:
I'm running this out of the IDE right now, I don't have a jar that you can run.

You can replicate the incorrect response using UaExpert, but it doesn't crash after that for some reason.

Stasik0, 2015-06-13:
@kevinherron okay, it seems to be correct. How can I build your example as quickly as possible to reproduce the segfault?

kevinherron, 2015-06-13:
Clone [this repo](https://github.com/digitalpetri/opc-ua-sdk-examples), modify the endpoint URL in SubscriptionExample.java so it points to an open62541 server, and then run the following command from the base of the repo dir:

```
mvn clean compile exec:java -Dexec.mainClass="com.digitalpetri.opcua.sdk.examples.client.SubscriptionExample"
```

You'll need Maven and JDK8 installed.

jpfr, 2015-06-13:
@Stasik0 Can we turn this into a unit test example for the simulated network interface?

Stasik0, 2015-06-13:
@kevinherron ```[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:compile (default-compile) on project sdk-examples: Fatal error compiling: invalid target release: 1.8 -> [Help 1]```
java -version says ```java version "1.8.0_45"```

kevinherron, 2015-06-13:
Where does your $JAVA_HOME env variable point to and what does `javac -version` return? What about `$JAVA_HOME/bin/javac -version`?

Stasik0, 2015-06-13:
okay... now i get
```
[ERROR] /home/sten/Desktop/petri/opc-ua-sdk-examples/src/main/java/com/digitalpetri/opcua/sdk/examples/client/SubscriptionExample.java:[8,41] cannot find symbol
[ERROR] symbol:   class AttributeId
[ERROR] location: package com.digitalpetri.opcua.stack.core
[ERROR] /home/sten/Desktop/petri/opc-ua-sdk-examples/src/main/java/com/digitalpetri/opcua/sdk/examples/client/SubscriptionExample.java:[46,28] package AttributeId does not exist
```

kevinherron, 2015-06-13:
Try again in ~30 seconds, I just deployed the updated stack artifacts to the Sonatype nexus. I just shuffled a bunch of things around this morning, unfortunately.

Stasik0, 2015-06-13:
@kevinherron no change, same error

kevinherron, 2015-06-13:
*sigh*

I think Maven has cached the older snapshot. Try deleting `~/.m2/repository/com/digitalpetri/opcua/stack-core` and try again...

Stasik0, 2015-06-13:
@kevinherron okay, this helps 
```
21:41:58.424 [com.digitalpetri.opcua.sdk.examples.client.SubscriptionExample.main()] INFO  c.d.o.s.e.client.SubscriptionExample - Using endpoint: opc.tcp://gemini:16664 [None]
21:41:58.852 [ua-serialization-pool-0] WARN  c.d.o.stack.core.util.ExecutionQueue - Uncaught Throwable during execution.
java.lang.ClassCastException: com.digitalpetri.opcua.stack.core.types.structured.ResponseHeader cannot be cast to com.digitalpetri.opcua.stack.core.serialization.UaResponseMessage
	at com.digitalpetri.opcua.stack.client.handlers.UaTcpClientSymmetricHandler.lambda$onSecureMessage$29(UaTcpClientSymmetricHandler.java:189) ~[stack-client-1.0.0-SNAPSHOT.jar:1.0.0-SNAPSHOT]
	at com.digitalpetri.opcua.stack.client.handlers.UaTcpClientSymmetricHandler$$Lambda$535/7938212.accept(Unknown Source) ~[na:na]
	at com.digitalpetri.opcua.stack.core.channel.SerializationQueue.lambda$decode$15(SerializationQueue.java:59) ~[stack-core-1.0.0-SNAPSHOT.jar:1.0.0-SNAPSHOT]
	at com.digitalpetri.opcua.stack.core.channel.SerializationQueue$$Lambda$522/26938938.run(Unknown Source) ~[na:na]
	at com.digitalpetri.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:113) ~[stack-core-1.0.0-SNAPSHOT.jar:1.0.0-SNAPSHOT]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [na:1.8.0_45]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) [na:1.8.0_45]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142) [na:1.8.0_45]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617) [na:1.8.0_45]
	at java.lang.Thread.run(Thread.java:745) [na:1.8.0_45]
```
there is no reconnect, everything stops after getting a create subscriptionrequest, i need to trace the segfault....

kevinherron, 2015-06-13:
Hmm. Yeah, when I run it I see the same thing, but then I see a subsequent reconnect as well. Either way it looks like you've got the crash reproducing now, so good luck!

Stasik0, 2015-06-13:
@kevinherron yes, but my server does not crash

kevinherron, 2015-06-13:
Oh, from the stack trace I can see that you've got the slightly older version of stack-client; do the same delete-the-cache dance you did for stack-core and then you should see the reconnect happen.

Stasik0, 2015-06-13:
@kevinherron now the server does not stop, i see a 60sec timeout... but no reconnect...

kevinherron, 2015-06-13:
Hmm. I'm kinda stumped on the reconnect thing. The example doesn't shut down cleanly right now, I expect that, but when I run it I get the following before it eventually shows the 60sec timeout:

```
12:56:34.787 [com.digitalpetri.opcua.sdk.examples.client.SubscriptionExample.main()] INFO  c.d.o.s.e.client.SubscriptionExample - Using endpoint: opc.tcp://ubuntu:16664 [None]
12:56:35.316 [ua-serialization-pool-0] ERROR c.d.o.s.c.h.UaTcpClientSymmetricHandler - Error decoding symmetric message: com.digitalpetri.opcua.stack.core.types.structured.ResponseHeader cannot be cast to com.digitalpetri.opcua.stack.core.serialization.UaResponseMessage
java.lang.ClassCastException: com.digitalpetri.opcua.stack.core.types.structured.ResponseHeader cannot be cast to com.digitalpetri.opcua.stack.core.serialization.UaResponseMessage
        at com.digitalpetri.opcua.stack.client.handlers.UaTcpClientSymmetricHandler.lambda$onSecureMessage$16(UaTcpClientSymmetricHandler.java:189) ~[stack-client-1.0.0-SNAPSHOT.jar:1.0.0-SNAPSHOT]
        at com.digitalpetri.opcua.stack.client.handlers.UaTcpClientSymmetricHandler$$Lambda$528/1201576332.accept(Unknown Source) [stack-client-1.0.0-SNAPSHOT.jar:1.0.0-SNAPSHOT]
        at com.digitalpetri.opcua.stack.core.channel.SerializationQueue.lambda$decode$15(SerializationQueue.java:59) [stack-core-1.0.0-SNAPSHOT.jar:1.0.0-SNAPSHOT]
        at com.digitalpetri.opcua.stack.core.channel.SerializationQueue$$Lambda$515/139696830.run(Unknown Source) [stack-core-1.0.0-SNAPSHOT.jar:1.0.0-SNAPSHOT]
        at com.digitalpetri.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:113) [stack-core-1.0.0-SNAPSHOT.jar:1.0.0-SNAPSHOT]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [na:1.8.0_25]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266) [na:1.8.0_25]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142) [na:1.8.0_25]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617) [na:1.8.0_25]
        at java.lang.Thread.run(Thread.java:745) [na:1.8.0_25]
12:56:35.354 [ua-serialization-pool-0] WARN  c.d.o.s.c.h.UaTcpClientAsymmetricHandler - Server revised secure channel lifetime to 0; renewal will not occur.

```

[Here](https://www.dropbox.com/s/4dv4cuezuwbx3rr/open62541.pcapng?dl=0) is the wireshark capture for the exchange.

Off the top of my head, the only reason I can think we're seeing different results is we're not actually running the same code. You could try deleting everything under `~/.m2/repository/com/digitalpetri/opcua` as a last effort.

I'm going to spend some time cleaning up the client examples so that they execute and exit cleanly. They were thrown together in a hurry a few weeks ago for someone who needed somewhere to start from.

Stasik0, 2015-06-13:
okay @kevinherron , i can not reproduce it unfortunately... i got a clean 60 sec response

may I ask you to send me a backtrace of the segfault (it is really important)... Can we hook up in IRC or in skype so i can explain you what to do?

kevinherron, 2015-06-13:
I just updated the examples so they exit cleanly, but yeah, I'll join you in IRC.

Stasik0, 2015-06-13:
okay, the reply is fixed, the segfault is also fixed in the master. @kevinherron, thank you

closed 2015-06-13 by Stasik0