open62541/open62541 #274: Please review building wiki articles

HolgerJeromin, 2015-06-13:
I fighted a bit with my windows machine to build the library. 
To document this i removed the autotools section in the wiki and added cmake stuff.

I would be happy if some windows and/or eclipse users could review the documents:

https://github.com/acplt/open62541/wiki/Building-open62541
https://github.com/acplt/open62541/wiki/Eclipse%20CDT

ichrispa, 2015-06-13:
I can also confirm build problems with eclipse and visual studio.

Stasik0, 2015-06-15:
i have fixed some typos... but it looks good

HolgerJeromin, 2015-06-15:
@ichrispa my problems were strict setup specific.

pspeybro, 2015-06-16:
Allow me to add a small suggestion for the build documentation: an overview of the cmake options that can be enabled and what the effect of those options are.

For example the effect of enabling something like:
-DENABLE_MULTITHREADING
-DENABLE_EXTERNAL_NAMESPACES
-DEXTENSION_UDP
...

What is the benefit/downside of enabling multithreading?

When not yet too familiar with the code, it's not obvious what all those options do exactly.

HolgerJeromin, 2015-06-16:
@pspeybro very good suggestion. These should get IMO documented separate (and linked from the build documentation).

Stasik0, 2015-06-29:
config is there, will probably go into the sphinx documentation https://github.com/acplt/open62541/wiki/Configuration-options

closed 2015-06-29 by Stasik0

HolgerJeromin, 2015-06-29:
Thanks stasik. I linked this options page from the build doc.