open62541/open62541 #275: Reduce overhead of service calls

jpfr, 2015-06-13:
Hey there,

We are good. But we can improve performance. :-)
Think **30% of the burned clock cycles** per service call...

**We don't need UA_calcSizeBinary**
We can just allocate the maximum message size (this is no more costly than allocating a very small buffer) and return an error if the message does not fit. That reduces encoding times by half.

**We spend a lot of time for small mallocs**
Most requests and responses contain many small arrays (counting strings as well). Thats 600+ cycles per malloc+free. We can improve this by reusing a small (thread-local) memory-buffer-stack that is reset after each interaction.
Also, we can finally switch to in-situ-decoding where strings (and generally arrays of zero-copy dataTypes, which we already detect) point into the received buffer.
With in-situ decoding, a fixed 20kb working-memory-buffer-stack should be enough for embedded devices. And we can always gracefully exit with UA_BADOUTOFMEMORY.

Any more things we should consider?

Stasik0, 2015-06-18:
https://github.com/acplt/open62541/blob/master/src/server/ua_server_binary.c#L160

here, the session list will be searched every time for the token. There a session is already allocated to the channel. Should we store the token in the session object, so we do not have a O(n)?

jpfr, 2015-06-29:
@Stasik0 
1) There are potentially multiple sessions per channel. That's why we call UA_SecureChannel_getSession first.
2) Yes. That lets us choose a custom timeout on the connection.

- UA_calcSizeBinary is removed from the main code path.
- The in-situ decoding approach has been implemented in the membuf branch.
But its results are a bit disappointing. We need to benchmark it once we have a client that can do async service calls.

Closing the issue for now

closed 2015-06-29 by jpfr

Stasik0, 2015-06-29:
Wy not merging in-situ in? Does it make stuf complicated? I mean it is not slower and has probably a better ram-consumption...