open62541/open62541 #276: Back-porting 0.1

Stasik0, 2015-06-13:
Hey,

we published 0.1 and were aiming of porting some fixes back from the master to it. The question: is it really needed? Do we need such *enterprise* feature at a such an early stage? We need to think about fixed APIs, etc.

ichrispa, 2015-06-14:
I think we do not need to backport features to 0.1. But we should backport any changes to the API. Someone building an application based on 0.1 should get more, new capabilities when upgrading to say 0.2 - but his application should be able to work just as it did before.

In the past few weeks, things like adding nodes and references has changed; these changes need to be backported imho.

Stasik0, 2015-06-14:
Nodes, references and networklayer api....

I'd vote for breaking the api and releasing the current master as a new number. The changed are more or less cosmetic and can be fixed in a couple of minutes

jpfr, 2015-06-14:
> But we should backport any changes to the API.

You volunteering @ichrispa ? :-)
I guess nobody is expecting more than bugfixes on 0.1
There are already some improvements in the 0.1 branch. We could tag 0.1.1 any time.

I'm with @Stasik0 that API changes in 0.2 are okay when they provide a substantial improvement or when new abstractions are required to consistently integrate new features.
Overall, the API changes are *really* straightforward so far. Especially if people use the logger and networklayer from the example code.

Stasik0, 2015-06-16:
okay, i conclude that there is a 2:1 vote

closed 2015-06-16 by Stasik0