open62541/open62541 #279: requirements and call logic for the client

HolgerJeromin, 2015-06-15:
Right now the client code and example is synchronous and does not take care of the session keep alive by itself (if it has no work to do).

I want to change this. My Idea was to reuse the pattern of the server code. 

But perhaps it is better to decide what are our requirements for the client code before.

* like now: synchronous fetching of one data from a server (and closing the connection).
* fetching data from a server and leave (and keeping) the connection+session open for a long time.

IMO the call logic should be the same as the server code. 
For integration into ACPLT/OV it would be best to have a client_run_mainloop which also has different code for support of pthreads or cyclic call of client_run_mainloop().

Do you have more wishes for the client integration into an application?

Stasik0, 2015-06-15:
+1 for the same as for the server code, did you see that the server loop is divided into a 1-iteration callable function? [1]

We need two or maybe three cases:
* 1 (a) client application takes own care of calling the client_worker() function aka OV style
* 1 (b) the simplest client does not care and my not call _worker at all since it needs only one value
* 2 client application forks a thread that runs client_worker()+some sleep and is joined before the client leaves

[1] https://github.com/acplt/open62541/blob/master/src/server/ua_server_worker.c#L544

jpfr, 2015-06-15:
Good idea.
We need this also for subscriptions in the client.

We can move `src/server/ua_server_worker.c` to `/src/ua_async_events.c`.
Then, the server and client can instantiate their very own mainloop.

ichrispa, 2015-06-15:
As discussed on IRC with @HolgerJeromin, I am really looking forward to using the timed work with the clients subscriptions.

But I do want the current functionality (userspace having single-threaded control over all message transmissions) maintained - i.e. multithreading in the client remaining optional - for backwards compatibility.

Stasik0, 2015-06-15:
@ichrispa this is covered by 1(a) and 2

Pro, 2018-02-01:
Related to #552 and implemented in https://github.com/open62541/open62541/commit/bffefa4cd0331d75a8ed46698ab1a959217c14f9

closed 2018-02-01 by Pro