open62541/open62541 #280: Get a Windows build going

Stasik0, 2015-06-16:
There is a Travis-Ci clone for windows-based VMs:

http://www.appveyor.com/

At least we can get builds running there.

ichrispa, 2015-06-18:
Interesting... all our attempts to use MSVS2015 failed due to limited c99 support when expanding macros.

Stasik0, 2015-06-18:
yea, but cannot you workaround it somehow?

ichrispa, 2015-06-18:
I need to switch my development focus from linux/stack dev to windows application dev soon. Windows is unfortunately a requirement due to third party dependencies. I'll see what I can do in the coming days.

Stasik0, 2015-06-18:
@ichrispa we need to workaround it anyway. Here: http://blogs.msdn.com/b/vcblog/archive/2015/04/29/c-11-14-17-features-in-vs-2015-rc.aspx it is stated as "[3] C99 preprocessor support is unchanged.  It's Partial because while the compiler supports variadic macros, our preprocessor's behavior doesn't conform to C99/C++11 in many ways. ". But getting an msvc is very imporant!

ichrispa, 2015-06-24:
I can't really find out where the fuss is. Maybe my VS2015 is afraid of me and just cooperates.

Turning intellisense off reduces the MSVS build of the master branch to the following problems:
  *  F_OK definitions need to be added to ua_util.h (as in dev several weeks ago)
  * When amalgamation is enabled, MSVS murks up the build order and skips building the header. Some manual fiddeling with the build order is required (place one check mark to make server depend on amalgamation)
  * When amalgamation is enabled, the ua_util.h contents for F_OK are not available to the server. Adding them to server.c fixes that problem.

After that, I have good compiles in the master branch.

Stasik0, 2015-06-24:
Can't the build order be changed in CMakeList? or will it break the other build?

ichrispa, 2015-06-24:
@Stasik0: In MSVS: Project->Project Build Order; Select dependencies and "server" from the dropdown list. Add amalgamation as a dependency.

The Development branch had two added pains; there was a stray `COMMAND python` in the CMakeFile and a cast in subscriptions.c was a problem for MSVS ("Cannot cast UA_Job to UA_Job"). Both have been fixed.

commit fa9cb05f (2015-06-26): appveyor ci support, fixes #280
commit 2c869f59 (2015-06-26): appveyor ci support, fixes #280
commit 01a07cd0 (2015-06-27): appveyor ci support, fixes #280
closed 2015-06-26 by Stasik0