open62541/open62541 #283: Feature merging

ichrispa, 2015-06-22:
Both client_addnodes and branches service_call_implementation branches are currently inactive, as they appear to be working as expected. Unless there are any issues to report, I'd suggest merging them into master and/or dev?

Stasik0, 2015-06-22:
adding nodes:
* last build failed
* it is cool, since they almost need no binary space on the server side

service call:
* example method is attached to a magic number id 11489, I think we do not have this node if no generated ns is used. May i propose to add another method node like "the answer" and attach an "echo" method to it? Maybe a method that calls the logger to print the method input?
* there is a similar problem with 
```#ifdef ENABLE_GENERATE_NAMESPACE0
#include "ua_methodcall_manager.h"
#endif``` 
in ua_nodes.c

jpfr, 2015-06-22:
**client_createNodes**
+1 for merging

**service_call**
Good job! Ofc some things can always be improved. :-)

- Just compile it in every time. The 2kb binary size are not worth the added complexity of the compiler flags.
- We have `ua_method_manager` files. But no manager. Maybe we can just fold the functionality into the `ua_nodes.c/h` and `ua_services_call.c`.
- `examples/server.c` includes files that the user is not supposed to see. Only the headers in `/include` are the user-space.

Stasik0, 2015-06-22:
+1 to remove ua_method_manager
-1 regarding always compile-in... this is again not only 2kb of code but some features

jpfr, 2015-06-22:
@Stasik0 
I fear the discussion. But do you suggest that we ifdef every feature that is not in the nano profile?

Stasik0, 2015-06-22:
@jpfr yea... and have some kind of (multiple) standard configurations. Do not want to discuss too much, aswell. But currently this fine modularity it is possible without any further effort. Later we will have to strep things out in a manual way if some server reduction is needed and small footprint is one of our USPs.

jpfr, 2015-06-23:
@ichrispa I found a new type in the xml: UA_Argument
Currently trying to use that for the method signature. I'll call you tomorrow if that works out.

ichrispa, 2015-07-13:
Both branches have been merged into dev (I think). Closing the issue...

closed 2015-07-13 by ichrispa