open62541/open62541 #284: Method Call Service improvements

ichrispa, 2015-06-24:
Commit b1a2edcbdf6821e58e99031d8b96c7ccd45c01a2 by @jpfr effectively eradicated two weeks of my work by substituting my method call implementation for his own. As far as my experience goes this is considered rude to say the least. In an equally rude fashion I have reverted that commit.

@jpfr's premise is that the spec explictly equipes MethodNodes with UA_Arguments that serve to check the conformance of any arguments passed or returned by the method. This is completely correct and I will incorporate that notion into method calls. To do so, I must first get the namespace compiler to actually produce valid functions with argument definitions (as exported by UA Modeller). Thereafter the checking of said arguments will be incorporated into methodcalls.

The headers for ua_methodcall_manager.c/.h are mislabeled, since the manager does not exist anymore. This defect will also be dealt with.

Are there any other suggestions for improving on this service that should be taken into account?

jpfr, 2015-06-24:
@ichrispa,

I'm sorry if the commit came across as rude. I was under the impression of *helping out* on this important feature.
Especially since you announced to no longer work on the branch, the changes were discussed in #283, and we had a call this morning about the commit in question and the changes therein.

The branch went from "methods not usable with UAExpert" to "methods usable with UAExpert".
But the commit was no substituation of your work at all! I couldn't have done it as fast/well if there wasn't your good precedent.

I'm sure your ongoing work on methods will soon be ready.

ichrispa, 2015-06-24:
@jpfr: From my point of view, we're good :)  I must have miscommunicated something; it was never my intention to abandon any of the branches. I'll try to get the best out of both our implementations into the branch; after all your improvements are valid, but undo some of mine.

My UA Expert did not complain - in fact it was my primary mode of testing, and I always got "Call Suceeded". How can I reproduce the failing call? I tested by calling GetMonitoredItems under ObjectTypes->Base->ServerType->GetMonitoredItems using 1.2.2 and 1.3.x.

Your input about UA_Arguments was invaluable - we should definetely get that in. But here's what always confuses me: Spec 4 says that Arguments are parts of the MethodNode attributes. However I have not found a single XML File that defines them as node attributes. Instead XML files define methods with two UA_Variables (hasProperty) with dataType Argument, containing an array of Arguments encoded as ExtensionObjects.

So which convention do we implement? Your solution is clean and simple (and documented in Spec 4), but apparently noone bothers to use it. We can scan hasProperty relations of a method and pull the Arguments out of variables... but apparently not even the foundation does it that way.

jpfr, 2015-06-25:
> Spec 4 says that Arguments are parts of the MethodNode attributes.

The arguments are part of the method node. But not as attributes. Think of the method node as an object and the arguments as the object's variables. They are added as separate variable nodes with a `hasProperty` hierarchical reference.

> My UA Expert did not complain - in fact it was my primary mode of testing, and I always got "Call Suceeded".

But you could not set the input arguments / see the output arguments in UA Expert GUI.

For that, add the arguments as two variable nodes (holding a `UA_Argument` array) and the browsenames `QualifiedName(0, "InputArguments")` and `QualifiedName(0, "OutputArguments")` respectively.

There is currently a bug in the nodestore. It replaces the browsename-namespace-id with the id of the added node. So you end up with `QualifiedName(1, "InputArguments")`. So the arguments are not recognized by UA Expert.

Stasik0, 2015-06-25:
@jpfr the bug is actually mine :( should i remove this feature of forcing BrowseName namespace id?

ichrispa, 2015-06-25:
I'm currently trying to get my compiler to encode extensionObjects present in the namespace; that way we should be able to generate the contents of any hasProperty relations. Since so far this was not the case, there was no way UAExpert could be informed that any arguments exist.

jpfr, 2015-06-27:
> should i remove this feature of forcing BrowseName namespace id?

Yes. UAExpert won't detect the argument lists otherwise.

ichrispa, 2015-06-29:
OK, the namespace compiler can now build the Arguments in the variable nodes. They show up in UA_Expert when Call is executed. The compiler is currently limited to flat extensionObjects (those not containing other extensionObjects) and might fail to encode multidimensional arrays.

jpfr created a UA_Server_addMethodNode(); the current implementation does not store UA_Arguments in the node anymore. So I would suggest I rewrite that userspace function as a macro that creates the InputArgument and OutputArgument variables/extensionObjects?

jpfr, 2015-07-10:
methods were merged into master.

closed 2015-07-10 by jpfr