open62541/open62541 #285: Documentation Reset

jpfr, 2015-06-29:
Dear all,

currently, open62541 is nearly impossible to use for somebody who does not want to read the actual sources. Besides the raw Doxygen definitions, we should have more prose in the documentation.

Please have a look at http://open62541.org/doc/sphinx/
The source of the pages is
- https://raw.githubusercontent.com/acplt/open62541/master/doc/index.rst
- https://raw.githubusercontent.com/acplt/open62541/master/doc/datatypes.rst

The structure could look llike this.

#. Introduction
  - Information Model
  - 
#. DataTypes
#. Services
#. Server
  - Startup
  - Adding Variables
  - Adding Methods
#. Client

Anything else?

Stasik0, 2015-06-29:
is there some tutorial on this magic markup?

jpfr, 2015-06-29:
http://sphinx-doc.org/contents.html
https://breathe.readthedocs.org/en/latest/directives.html

FlorianPalm, 2015-06-29:
Interface to external nodestores?

Stasik0, 2015-06-29:
Configuration options: https://github.com/acplt/open62541/wiki/Configuration-options, probably it is better to remove them from the wiki and transfer to sphinx

jpfr, 2015-07-09:
http://open62541.org/doc/sphinx/building.html

jpfr, 2015-07-15:
@Stasik0 told me about this Sphinx documentation: http://thrift-tutorial.readthedocs.org/en/latest/intro.html

That is really well made! Should we simply copy their structure?

Stasik0, 2015-07-15:
I'd also like to have some "articles" section that are more or less blog posts on various topics

jpfr, 2015-07-15:
Yeah!
So we have a series of articles to build up basic understanding and a separate "API documentation" with inline code examples.

@Stasik0 you have some themes for articles in mind?

Stasik0, 2015-07-15:
You had a couple of artcles un your blog :)))

ichrispa, 2015-07-28:
Is there room in the documentation for a tutorial on open62541? Something to talk new users through creating server/client applications, datasources on variable nodes, callbacks in methods and monitoreditems an such?

jpfr, 2015-08-02:
> I went on a limb and wrote a firstSteps.rst

Just perfect. :smile:

jpfr, 2015-10-16:
The new infrastructure is in place.
Now we need to write the content. :-)

closed 2015-10-16 by jpfr