open62541/open62541 #288: check not found and found

btrepp, 2015-07-06:
Using v0.1 release. In debian jessie.
```
 apt-get install check
Reading package lists... Done
Building dependency tree
Reading state information... Done
check is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
```
```
 cmake -D BUILD_UNIT_TESTS=ON .
-- Could NOT find Git (missing:  GIT_EXECUTABLE)
CMake Warning at CMakeLists.txt:17 (message):
  Git not found.  Build will not contain git revision info.

-- Git version: unknown (git not found!)
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE)
-- checking for one of the modules 'check'
CMake Error at /usr/share/cmake-3.0/Modules/FindPkgConfig.cmake:425 (message):
  None of the required 'check' found
Call Stack (most recent call first):
  cmake/FindCheck.cmake:27 (PKG_SEARCH_MODULE)
  tests/CMakeLists.txt:6 (find_package)

-- Found CHECK: /usr/lib/x86_64-linux-gnu/libcheck.a
-- Configuring incomplete, errors occurred!
See also "/open62541/open62541-0.1/CMakeFiles/CMakeOutput.log".
See also "/open62541/open62541-0.1/CMakeFiles/CMakeError.log".
```

I'm not to great with CMake, but You can see lower down it finds a check static binary. Is there something else I need to install for check to work?

btrepp, 2015-07-06:
Apparently you need to install pkg-config to build the tests too.

This isn't in debian after running the ubuntu instructions. Is it in ubuntu?. Might need to be added to the build instructions.

closed 2015-07-06 by btrepp

Stasik0, 2015-07-06:
yes it seems that cmake relies on pkg-config to locate check. I have added a line on the wiki, feel free to edit.