open62541/open62541 #289: Amalgamation fails for v0.1

btrepp, 2015-07-06:


Stasik0, 2015-07-06:
these is a race condition that we fixed in master already. To fix this try to replace "make -j" by "make. v0.1 is kinda abandoned ATM, so try master instead

btrepp, 2015-07-06:
Still occurs for me in the master branch (gotten the tarball of of github).

```
 RUN cmake -D BUILD_UNIT_TESTS=ON         -DENABLE_AMALGAMATION=ON      .
 ---> Running in a94816331cc5
-- The C compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.9")
-- Could NOT find Git (missing:  GIT_EXECUTABLE)
CMake Warning at CMakeLists.txt:28 (message):
  Git not found.  Build will not contain git revision info.

-- Git version: unknown (git not found!)
-- CMAKE_BUILD_TYPE not given; setting to 'RelWithDebInfo'.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28")
-- checking for one of the modules 'check'
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: /open62541/open62541-master
 ---> 2a8dc5be2a1a
Removing intermediate container a94816331cc5
Step 8 : RUN make
 ---> Running in 190da7a95cf0
Scanning dependencies of target amalgamation
[  6%] Generating src_generated/ua_nodeids.h
[ 12%] Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h
[ 18%] Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h
[ 25%] Generating open62541.h
/bin/sh: 1: Syntax error: "(" unexpected
CMakeFiles/amalgamation.dir/build.make:75: recipe for target 'open62541.h' failed
make[2]: *** [open62541.h] Error 2
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/amalgamation.dir/all' failed
make[1]: *** [CMakeFiles/amalgamation.dir/all] Error 2
Makefile:86: recipe for target 'all' failed
make: *** [all] Error 2
```

Stasik0, 2015-07-06:
okay this seems to be something new... :/ it fails on the same line.... please run
VERBOSE=1 make -j1

btrepp, 2015-07-06:
Okay so I've done some testing. Amalgamate fails without the presence of git.

With git installed the error shifts. Note: I'm getting the version from a tarball, so it doesn't have .git info

```
Running in da0a3f6a458f
Scanning dependencies of target amalgamation
[  6%] Generating src_generated/ua_nodeids.h
[ 12%] Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h
[ 18%] Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h
[ 25%] Generating open62541.h
Starting amalgamating file /open62541/open62541-master/open62541.h
Integrating file '/open62541/open62541-master/src_generated/ua_config.h'...done.
Integrating file '/open62541/open62541-master/include/ua_statuscodes.h'...done.
Integrating file '/open62541/open62541-master/include/ua_types.h'...done.
Integrating file '/open62541/open62541-master/src_generated/ua_nodeids.h'...done.
Integrating file '/open62541/open62541-master/src_generated/ua_types_generated.h'...done.
Integrating file '/open62541/open62541-master/include/ua_connection.h'...done.
Integrating file '/open62541/open62541-master/include/ua_log.h'...done.
Integrating file '/open62541/open62541-master/include/ua_server.h'...done.
Integrating file '/open62541/open62541-master/include/ua_client.h'...done.
Integrating file '/open62541/open62541-master/examples/networklayer_tcp.h'...done.
Integrating file '/open62541/open62541-master/examples/logger_stdout.h'...done.
The size of /open62541/open62541-master/open62541.h is 227069 Bytes.
[ 31%] Generating open62541.c
Starting amalgamating file /open62541/open62541-master/open62541.c
Integrating file '/open62541/open62541-master/src/ua_util.h'...done.
Integrating file '/open62541/open62541-master/deps/queue.h'...done.
Integrating file '/open62541/open62541-master/src_generated/ua_transport_generated.h'...done.
Integrating file '/open62541/open62541-master/src/ua_types_encoding_binary.h'...done.
Integrating file '/open62541/open62541-master/src/ua_securechannel.h'...done.
Integrating file '/open62541/open62541-master/src/ua_session.h'...done.
Integrating file '/open62541/open62541-master/src/server/ua_nodes.h'...done.
Integrating file '/open62541/open62541-master/src/server/ua_nodestore.h'...done.
Integrating file '/open62541/open62541-master/src/server/ua_session_manager.h'...done.
Integrating file '/open62541/open62541-master/src/server/ua_securechannel_manager.h'...done.
Integrating file '/open62541/open62541-master/src/server/ua_server_internal.h'...done.
Integrating file '/open62541/open62541-master/src/server/ua_services.h'...Traceback (most recent call last):
  File "/open62541/open62541-master/tools/amalgamate.py", line 72, in <module>
    file.write(line)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2019' in position 76: ordinal not in range(128)
CMakeFiles/amalgamation.dir/build.make:117: recipe for target 'open62541.c' failed
make[2]: *** [open62541.c] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/amalgamation.dir/all' failed
make[1]: *** [CMakeFiles/amalgamation.dir/all] Error 2
Makefile:86: recipe for target 'all' failed
make: *** [all] Error 2
```

Stasik0, 2015-07-06:
okay, both issues should be fixed now

Stasik0, 2015-07-07:
@btrepp did it work out?

btrepp, 2015-07-13:
Yep that worked perfectly

https://registry.hub.docker.com/u/btrepp/open62541-docker/build_id/72951/code/b6d4aykwnovjdfy2yp4cs2l/

https://registry.hub.docker.com/u/btrepp/open62541-docker/build_id/72951/code/bcuovxlvrmeryndvfhs2gei/

closed 2015-07-13 by btrepp

Stasik0, 2015-07-13:
@btrepp whats the best way of us to cross-link it?

btrepp, 2015-07-14:
Atm it will auto build from new debian releases, taking the v1 release and whatever is current on master (so there is options).

As is it won't require any changes, it will just build (though new releases might just need a new branch to get them building). 

If you want I'm happy for you to link to it just saying docker images are available, alternatively I could send you a pull request with a docker file if you would like to fold it into the project.

Stasik0, 2015-07-14:
@btrepp: I'd preffer to have a PR, so it can be stored here in the project

Stasik0, 2015-08-20:
@btrepp any progress?

btrepp, 2016-03-15:
Hi there,

I've created a dockerfile for the latest version from source and sent a pull request. See #557