open62541/open62541 #292: Adding a UA_Server pointer in UA_Session struct?

Stasik0, 2015-07-07:
I'd like to implement SubscriptionManager_deleteMembers. Unfortunately, Session_deleteMembers needs a pointer to Server in order to remove the job? What's the plan? A pragmatic solution would be to add a server pointer to a session, since a session may not exist without the server or in any other context. However this is somehow dirty... Anyone a better solution?

ichrispa, 2015-07-08:
A session will propably be deleted either by the server or the network layer. Both usually have a valid server pointer... so why not just pass the server as a parameter to deleteMembers?

We could also "tag" subscriptions as to be deleted, then add a safeguard inside the job function. It can then unregister itself if it is called:

```
static void Subscription_timedUpdateNotificationsJob(UA_Server *server, void *data) {
	    // Timed-Worker/Job Version of updateNotifications
	    UA_Subscription *sub = (UA_Subscription *) data;
	    UA_MonitoredItem *mon;
	    
	    if (!data || !server)
	        return;
```
When deleting a subscription that has a job, it is quite simply not deleted. When the Job is called, 
instead of returning if the subscription is NULL, the work item removes itself from the server?

Stasik0, 2015-07-08:
i have passed everything up to the function, there seem no subscription-related leaks anymore... just a bunch of uninitialised value warnings... :)

Stasik0, 2015-07-08:
fixed in d5002592c86687a9c3351dd4e08801cc4ff8ec45

closed 2015-07-08 by Stasik0