open62541/open62541 #293: Compiling dev fails

Stasik0, 2015-07-08:
```
/home/sten/Desktop/open62541/src/server/ua_subscription_manager.c: In function ‘SubscriptionManager_init’:
/home/sten/Desktop/open62541/src/server/ua_subscription_manager.c:23:5: error: right shift count >= width of type [-Werror]
/home/sten/Desktop/open62541/src/server/ua_subscription_manager.c:31:5: error: right shift count >= width of type [-Werror]
/home/sten/Desktop/open62541/src/server/ua_subscription_manager.c:32:5: error: right shift count >= width of type [-Werror]
/home/sten/Desktop/open62541/src/server/ua_subscription_manager.c:33:5: error: right shift count >= width of type [-Werror]
/home/sten/Desktop/open62541/src/server/ua_subscription_manager.c:34:5: error: right shift count >= width of type [-Werror]
```
gcc 4.8.2 @ 32 bit linux
it is funny, since I did not have this at my 64bit machine at home

ichrispa, 2015-07-08:
I know the error (saw it before on windows). Interestingly enough, this error does not occur on my 32Bit laptop.

On a 64Bit System, the DateTime returns Picoseconds. Two Datetimes are then combobulated, where one is kept and one is xored.  The purpose of this function is to establish a semirandom but unique GUID for monitoredItems. 

I will fix it first thing tomorrow by only using the higher 32 bit on 64Bit plattforms and the normal 32Bit on x86. Not a serious issue.

ichrispa, 2015-07-09:
Added a bit of dirty-patching to make the random-guid-initializer work on 32 & 64 Bit plattforms.

closed 2015-07-09 by Stasik0