open62541/open62541 #294: Canonical examples instead of "big mess examples"

jpfr, 2015-07-09:
I started to add many "small" server examples, each showing a single feature.

https://github.com/acplt/open62541/blob/master/examples/server_variable.c
https://github.com/acplt/open62541/blob/master/examples/server_method.c
https://github.com/acplt/open62541/blob/master/examples/server_repeated_job.c

Adding another example is just a two-liner here: https://github.com/acplt/open62541/blob/master/examples/CMakeLists.txt

Since the examples are an important part of the documentation, we imho should get rid of the big example servers and split everything into canonical and understandable mini-examples.

HolgerJeromin, 2015-07-09:
I like it!

Stasik0, 2015-07-20:
we need to add examples to the build-chain in order to ensure the consistency of the api

Stasik0, 2015-08-07:
i added server_nodeset.c

Stasik0, 2015-09-04:
refers to #285

closed 2015-09-04 by Stasik0