open62541/open62541 #296: Bugs in subscriptions

Stasik0, 2015-07-09:
Now the bug list starts:

* pull the cpu temperature into subscriptoin - the value is not changed. It is different from the node. I can not understand where the difference is between cpu temperature and cureent time.
* timestamp information on the subscription is set to 1? I.e. it is not updated.

ichrispa, 2015-07-13:
If you register multiple subscriptions in ua expert, do you get different values?

Is the cpu temperature routine called (printf)?

Stasik0, 2015-07-13:
yes, but they do not change

yes, it is, for each subsubscription

ichrispa, 2015-07-14:
The problem is apparently the encoded bytestring of the UA_Double. If you switch the datatype to Int32, you can see the temperature change live (as expected).

Here's what happens: In https://github.com/acplt/open62541/blob/master/src/server/ua_subscription.c#L531 the monitored item copies its newly aquired value into a bytestring and compares it to the last value seen in lastSampledValue (also a bytestring).

For a UA_Double, this bytestring simply doesn't change. The bytestring_compare always returns a perfect match, the monitoreditem thinks the value did not change and hence the subscriptions never gets a new value.

The actual temperature value in readTemperature does change. As far as I can tell, there is something wrong with the encodeBinary function of UA_Double that produces 9 byte byteString, which in itself is somehow wrong...

Stasik0, 2015-07-14:
I also see that the relevant test is grayed out... still encode seems to work, since we *do* get valid numbers for the node's value...

Stasik0, 2015-07-14:
@ichrispa as you see from the commited test, the decoder seems to work okay... the length is 8 not 9

ichrispa, 2015-07-14:
interesting... still, I will look into this in more detail (I'm still swamped). Bytestring_compare does not recognize the changed value, either because I am sampling it wrong when getting it from the datasource or because the variant being encoded does not actually point to the correct value.

I strikes me as odd that this should only happen with float though... I think I'll focus on 64bit/32bit field lengths here. I halfway suspect that the sprintf("%lf") used in readTemperature might be part of this issue...

ichrispa, 2015-07-20:
@Stasik0: I fixed the bug, but it wasn't in subscriptions.

To establish if a value is changed, monitoredItems encode the last sampled value as a bytestring (UA_Variant containing the value). A comparison is then run with UA_Bytestring_compare(lastsampled, newvalue). UA_Bytestring_compare internally used strncmp.

strncmp `compares the only first (at most) n bytes of s1 and s2` (man page), meaning it breaks at a NULL. Hence it does not fully perform a full bytewise compare, which is fatal to bytestrings containing non-string data (as f.e. transport encodings). 

@jpfr, @Stasik0: I replaced strncmp with memcmp in https://github.com/acplt/open62541/blob/master/src/ua_types.c#L158 and everything appears to be working as expected. Please review if this imposes any negative consequences for you.

Stasik0, 2015-07-20:
yeah, seems to be okay

closed 2015-07-20 by ichrispa