open62541/open62541 #297: cygwin compile supporting label:easy hacks

joyhope, 2015-07-13:
The project could compile on the cygwin environment with following changes.

```
# compiler flags
if(CMAKE_COMPILER_IS_GNUCC OR "x${CMAKE_C_COMPILER_ID}" STREQUAL "xClang")
    if(CYGWIN)
      add_definitions(-std=c99 -pipe -Wall -Wextra -Wformat -Wno-unused-parameter
                      -Wno-unused-function -Wno-unused-label -Wpointer-arith -Wreturn-type -Wsign-compare -Wmultichar
                      -Wcast-qual -Wmissing-prototypes -Wstrict-prototypes # -Wshadow -Wconversion
                      -Winit-self -Wuninitialized -Wformat-security -Wformat-nonliteral)
    endif()
    if(NOT CYGWIN)
      add_definitions(-std=c99 -pipe -Wall -Wextra -Werror -Wformat -Wno-unused-parameter
                      -Wno-unused-function -Wno-unused-label -Wpointer-arith -Wreturn-type -Wsign-compare -Wmultichar
                      -Wcast-qual -Wmissing-prototypes -Wstrict-prototypes # -Wshadow -Wconversion
                      -Winit-self -Wuninitialized -Wformat-security -Wformat-nonliteral)
    endif()
    # binary size reduction settings
	add_definitions(-ffunction-sections -fdata-sections -fno-stack-protector -fno-unwind-tables
                    -fno-asynchronous-unwind-tables -fno-math-errno -fmerge-all-constants -fno-ident)
    set(CMAKE_C_LINK_FLAGS "${CMAKE_C_LINK_FLAGS} -Wl,--gc-sections")
    set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS} -Wl,--gc-sections")
    set(CMAKE_SHARED_LIBRARY_LINK_C_FLAGS "") # cmake sets -rdynamic by default
	if(NOT WIN32)
     if(NOT CYGWIN)
	    add_definitions(-fvisibility=hidden -fPIC)
        set(CMAKE_C_LINK_FLAGS "${CMAKE_C_LINK_FLAGS} -Wl,-z,norelro -Wl,--hash-style=gnu -Wl,--build-id=none")
      endif()
	endif()
    if(CMAKE_BUILD_TYPE STREQUAL "MinSizeRel" OR CMAKE_BUILD_TYPE STREQUAL "Release")
        set(CMAKE_C_LINK_FLAGS "${CMAKE_C_LINK_FLAGS} -s")
        set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS} -s")
    endif()
endif()
```

Stasik0, 2015-07-13:
@joyhope is CYGWIN detected automatically? It it already a working patch?

joyhope, 2015-07-13:
It is already working with this small change.  I am not know whether the small change is useful for the project.

Stasik0, 2015-07-13:
@joyhope I'd consider it useful as long as it does not break existing functionality (which seems to be the case). Could you do a fork of our repository, add the change and propose a pull-request? In this case you will own the merged commit... otherwise I could add it by myself, but you will lose the "contribution"

PR #298 (2015-07-14): support cygwin envirnment
It could be compiled in the cygwin environment.
closes #297
closed 2015-07-14 by Stasik0