open62541/open62541 #299: Cannot compile with ENABLE_AMALGAMATION and ENABLE_SUBSCRIPTION

Stasik0, 2015-07-14:
Errors give something like:

```
In file included from /home/sten/Desktop/open62541/open62541.c:25:0:
/home/sten/Desktop/open62541/open62541.h:4183:5: error: expected specifier-qualifier-list before ‘LIST_ENTRY’
     LIST_ENTRY(UA_Client_NotificationsAckNumber_s) listEntry;
     ^
```
etc

ichrispa, 2015-07-14:
I think the order of inclusion of queue.h and subscriptions.h might be messed up; is #define LIST_ENTRY(field) above that line in amalgamation?

Stasik0, 2015-07-15:
okay the problem is that queue.h is not included in the public header files. @jpfr @ichrispa I think it would be better if we would introduce ua_client_internal aswell to hide all the client's gutts like its done with the server. Does it make sense?

jpfr, 2015-07-15:
>  introduce ua_client_internal aswell to hide all the client's gutts

Good idea!

ichrispa, 2015-07-16:
> Does it make sense?

Yep, I'd support that.

commit 5f2f9498 (2015-07-17): fixes #299, adding a non-public ua_client_internal.h to hide internals of client's implementation
closed 2015-07-17 by Stasik0