open62541/open62541 #300: Use PCG Random Number generator

jpfr, 2015-07-15:
The latest cygwin patch worked around the differences in the random number generator across operating systems.

We could simply roll our own PCG-based RNG.
http://www.pcg-random.org/download.html

- literally 5 lines of code
- nicer randomness than the builtin RNG
- removes another friction point for porting

Currently, we store the seed in the server struct. For concurrency, we should declare it as thread local storage. So threads don't interfere / cache-congest.

ichrispa, 2015-07-25:
@jpfr: Would this also replace rand_r from stdlib.h in ua_types.c?

I am trying to compile open62541 for android-17 and the subminimal support of libc is turning out to be a problem...

But I am happy to report that the stack runs great on my Kobo Touch eBook Reader - I don't know what use that will be, but it proves we are in principal still ok on ARM targets :)

Stasik0, 2015-07-26:
@ichrispa afaik, it will

DuhaNN, 2015-08-14:
@ichrispa can you help me with android? Do you have wrapper? Thanks

jpfr, 2015-09-22:
The change was implemented in 85cd217.

closed 2015-09-22 by jpfr