open62541/open62541 #302: Building problems with ENABLE_METHODCALLS and/or ENABLE_SUBSCRIPTIONS

martina-felber, 2015-07-27:
Hey there! Many thanks for your open62541 project!
I tried your example code and could successfully add some variables to the server and read them with the client. 
Unfortunately I wasn't able to get your example code with ENABLE_SUBSCRIPTIONS and/or ENABLE_METHODCALLS working. 
I tried to build the open62541 header and c file and the client.c file from the /examples directory. 
I am using Eclipse with MinGW and get the following error if ENABLE_METHODCALLS is defined: 
gcc -o OPC_Holy_Client.exe "src\\open62541.o" "src\\client.o" "C:\\MinGW\\lib\\libws2_32.a" 
src\open62541.o: In function `processMSG':
../src/open62541.c:7779: undefined reference to `Service_Call'

If ENABLE_SUBSCRIPTIONS is defined I get the following first error: 
..\src\open62541.c:1279:9: error: unknown type name 'UA_SubscriptionManager' UA_SubscriptionManager subscriptionManager;

I added every .c file from your /src and /deps directory but I only got one error after the other, that some variables/functions are not defined or some headers could not be found. 

After that I followed the build instructions and generated some source code and the executables for client and server with cmake under Linux Open Suse.
I tried to build with the additional generated source files, but the project still does not compile and 
if I try to execute the generated executables I get the error: ./server_static: line 1: syntax error: unexpected "(". 
Maybe I have missed something at the build process. 
Many thanks in advance for your help.
Best regards.

jpfr, 2015-07-27:
Hello Martina,

I tried to reproduce the error and could compile the code with the following steps:

- Checkout the current master (not the 0.1 release)
- Run cmake-gui (a windows graphical user interface to cmake)
- Select ENABLE_AMALGAMATION (for the single-file-release), ENABLE_METHODCALLS, ENABLE_SUBSCRIPTIONS and BUILD_EXAMPLECLIENT.
- Generate the makefile for mingw
- Open a terminal (cmd) and cd into the cmake build directory
- Run C:\MinGW\bin\mingw32-make.exe

The steps might be a bit different for Eclipse + MinGW. But the general approach should be the same. The amalgamation process gives you the single .c/.h release that you can then easily drop into third projects.

ichrispa, 2015-07-27:
Dear Martina,

I tried compiling the current master branch under Windows (MSVS2015) and Linux (OpenSUSE 13.1 x86_64). The Linux cmake options where as follows:
 ```cmake -DBUILD_EXAMPLESERVER=On -DBUILD_EXAMPLECLIENT=On -DENABLE_AMALGAMATION=On -DENABLE_METHODCALLS=On -DENABLE_SUBSCRIPTIONS=On -DENABLE_AMALGAMATION=On -DENABLE_GENERATE_NAMESPACE0=On -DENABLE_ADDNODES=On ../```

Please note that using MSVS2015 is manditory due to the C99 requirements and I need to omit amalgamation for the compilation to work. If namespace generation is enabled with MSVS, you will also have to remove the server certificates embedded in tools/schema/namespace0/Opc.Ua.NodeSet2.xml (the compiler will fail with a "line too long" error message otherwise).

In both cases I managed to compile the stack without problems. I suggest to recompile the dll/so in a clean build environment. If the build continues to fail, please post (for each OS separately) which cmake options where enabled and which errors where produced.

Best regards,

ichrispa

martina-felber, 2015-07-28:
Hello!

Many thanks for your help and fast and detailed answers. 

1. First I tried jpfr's suggestions (OS = Windows 7) and I was able to compile the client, but not the server. Probably I took the wrong source files. 

* I downloaded the source code from https://github.com/acplt/open62541, branch master as zip file. 
* I ran cmake-gui, selected the source code and build output directory and pressed the Configure button. 
* I chose MinGW Makefiles and the first option - Use default native compilers. 
* When I pressed the Finish button a message box popped up with the
following error: Error in configuration process, project files may be invalid. 
The error output in the box below was the following:

CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.3/Modules/CMakeMinGWFindMake.cmake:22 (message):
sh.exe was found in your PATH, here:
C:/MinGW/msys/1.0/bin/sh.exe
For MinGW make to work correctly sh.exe must NOT be in your path.
Run cmake from a shell that does not have sh.exe in your PATH.
If you want to use a UNIX shell, then use MSYS Makefiles.
Call Stack (most recent call first):
CMakeLists.txt:4 (project)

CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
Missing variable is:
CMAKE_C_COMPILER_ENV_VAR
CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
Missing variable is:
CMAKE_C_COMPILER
CMake Error: Could not find cmake module file: D:/Projekte/OPC/open62541-master_try/build_client/CMakeFiles/3.3.0/CMakeCCompiler.cmake
CMake Error at CMakeLists.txt:4 (project):
No CMAKE_C_COMPILER could be found.

Tell CMake where to find the compiler by setting the CMake cache entry
CMAKE_C_COMPILER to the full path to the compiler, or to the compiler name
if it is in the PATH.

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
Configuring incomplete, errors occurred!

* I pressed the Configure Button again and got the following output:

The C compiler identification is GNU 4.8.1
Check for working C compiler: C:/MinGW/bin/gcc.exe
Check for working C compiler: C:/MinGW/bin/gcc.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Found PythonInterp: C:/Python27/python.exe (found version "2.7.10") 
Found Git: C:/Program Files (x86)/Git/cmd/git.exe (found version "1.9.5.msysgit.1") 
fatal: Not a git repository (or any of the parent directories): .git

CMake Warning at CMakeLists.txt:22 (message):
  Git failed (not a repo, or no tags).  Build will not contain git revision
  info.

Git version: unknown--git-commit-id-unknown
CMAKE_BUILD_TYPE not given; setting to 'RelWithDebInfo'.
Configuring done

* Then I could choose following options:
BUILD_EXAMPLECLIENT, ENABLE_AMALGAMATION, ENABLE_METHODCALLS, ENABLE_SUBSCRIPTION
and CMAKE_BUILD_TYPE: RelWithDebInfo was already filled in.

* I pressed the Generate button and got following output: 
fatal: Not a git repository (or any of the parent directories): .git
CMake Warning at CMakeLists.txt:22 (message):
Git failed (not a repo, or no tags).  Build will not contain git revision
info.

[... 41 lines cut ...]
mingw32-make[2]: *** [CMakeFiles/server.dir/examples/server.c.obj] Error 1
CMakeFiles\Makefile2:178: recipe for target 'CMakeFiles/server.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/server.dir/all] Error 2
Makefile:82: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

* I got the same errors when I tried to compile the source files in Eclipse. 
* Afterwards I tried the pre release v0.1.0-RC4 from following website https://github.com/acplt/open62541/releases too, but I couldn't configure 
ENABLE_METHODCALLS and ENABLE_SUBSCRIPTION, because the options did not exist. 

3. After that I followed ichrispa's instructions and tried to compile under OpenSUSE 13.1 (virtual machine). 
Unfortunately the building process does not finish. I've been waiting for an hour, but the build stucks at: 
[ 35%] Built target amalgamation
Scanning dependencies of target open62541-object
[ 41%] Building C object CMakeFiles/open62541-object.dir/open62541.c.o

I restarted my computer and tried again, I also tried with only BUILD_EXAMPLECLIENT enabled, without the BUILD_EXAMPLESERVER but the build does not finish. 

But if I leave the option -DENABLE_GENERATE_NAMESPACE0=On the build finishes and it works. Thank you very much!

I hope I can manage to build the sources in Eclipse with an ARM cross compiler afterwards.

I haven't tried to build with MS Visual Studio yet, because I'm afraid that the installation process requires another few hours, but I'm going to try.

Thank you very much for your help.

Stasik0, 2015-07-28:
>    1. Then I tried to build the server. I performed exactly the same
>    1. After that I followed ichrispa's instructions and tried to compile

Hello Martina,

you may try to set GENERATE_NAMESPACE0_FILE="Opc.Ua.NodeSet2.Minimal.xml"
in order to speed up the ns0 generation, normally it should not be needed
to compile

The problems you have with MinGW seem to be an MinGW issue:
http://sourceforge.net/p/mingw/bugs/2256/, quote: "I confirm that replacing
"_cdecl" with "__cdecl" in the file MINGW_PATH/usr/include/unistd.h will
fix the compilation issue." or just get a latest issue.

Best Regards

martina-felber, 2015-07-28:
Hello!
Thank you very much! With the indiciation of the namespace0 file the build process finished within a minute and after editing the unistd.h I could build with MinGW too.
Many thanks, I would have never been able to build it without your help.

ichrispa, 2015-07-28:
For OpenSUSE, you may also try setting the build type from 'RelWithDbgInfo' to 'Debug', which cuts down the compilation time considerably. You can then also use the full Namespace0 (Opc.Ua.NodeSet2.xml). The build otherwise takes about 90 Minutes, given that the compilers tries to optimize the rather huge generated namespace.

For Windows, the problem is definetely located in the setup of the MinGW environment.

martina-felber, 2015-07-28:
Thank you for the information!
Only with the option -DENABLE_GENERATE_NAMESPACE0=On (and GENERATE_NAMESPACE0_FILE="Opc.Ua.NodeSet2.Minimal.xml" to reduce build time) I was able to build with cmake and afterwards use the generated open62541.c and open62541.h file to set up a project in Eclipse for Windows 7 and OpenSuse (with cross compiler for ARM).
I thought I could build the project with the single file distributions of http://open62541.org/ but I had to use the generated files.
I don't understand why -DENABLE_GENERATE_NAMESPACE0=On is necessary or rather for what the namespaces are used. I saw in the simple server example that namespace 0 is used: 
`<UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);>`
And I tried a few other random namespace numbers but I don't understand the concept of the namespaces.
Many thanks and best regards 
Martina

FlorianPalm, 2015-07-28:
Namespaces are used to divide the address space into several mostly indepedent pieces. This concept is especially useful when you load several user defined information models. Each model defines a set of nodes and connections between them. To avoid naming collision (same node Identifier used more than one time within one namespace) every information model should be stored into a seperate namespace. 

Standard nodes which are present on every OPC UA server are stored into the namespace 0. Here you find the Server Object, types, reference hirarchies etc. So if you create your own nodes, you should create them into another namespace but namespace 0.

Eventhough, the actual data is stored into different namespaces you can interconnect the nodes with references to allow the client to browse your modeled data.

We also use the namespace identifier to decide which data storage is used to store the nodes ("nodestore" or user defined data storage)

Stasik0, 2015-07-28:
DENABLE_GENERATE_NAMESPACE0=On should be not necesserly, though... What are the errors you get without it?

ichrispa, 2015-07-28:
@martina-felber: A little Addendum to FlorianPalms great explanation. If you use References, Objects, etc. defined in Namespace 0 they must also be defined in the server application. We currently have 3 different methods of doing that:

'DENABLE_GENERATE_NAMESPACE=Off' will use a minimal subset of Namespace 0 types and objects hardcoded into the Server_new() function. This configuration is targeted at small devices that do not have a lot of memory or applications that can cope with the minimal nodeset.

'DENABLE_GENERATE_NAMESPACE=On'  will use a python based namespace compiler to analyze a XML Definition file and automatically create suitable C-Code for embedding this namespace into the server. Because the specifications namespace 0 is simply huge (about 40000 lines of code being generated), it might overwhelm the compiler to optimize it. Hence there are two XML's to draw from:
  * Opc.Ua.NodeSet2.xml is the complete nodeset for namespace 0 as defined in the OPC UA Spec.
  * Opc.Ua.NodeSet2.Minimal.xml is a highly reduced subset of the first, about equal in functionality to the hardcoded nodeset.

Generating the namespace has the overall advantage that you can generate your own namespace models (namespace 1,2,...) with tools like UAModeler and tie them into the server as well. We are still working on a user friendly method of automating the loading of user defined namespaces.

martina-felber, 2015-07-29:
Hello!
Thank you for your great explanation of namespaces and their use. I have to admit that I don't know much about OPC yet. I have to thoroughly go through your example code. Unfortunately the OPC specifications are not freely available and I struggle a bit to find a good general (not vendor specific) documentation.

@Stasik0 I have to apologise, you were right: ENABLE_GENERATE_NAMESPACE0=On is not necessary. I don't know why, but when I tried yesterday the generated executables worked, but I couldn't build the files in Eclipse. Maybe I have taken the wrong files by mistake. I retried this morning and it worked. I'm very sorry for my misstatement. 

Thank you very much for your great support!

Stasik0, 2015-07-29:
The specs are freely available from the opc foundation (you only need to register, though).

martina-felber, 2015-07-29:
@Stasik0 thank you very much, I thought only company members get access. I'm sorry, thank you!

closed 2015-07-29 by martina-felber