open62541/open62541 #303: Userland UANodeSet code generation

Stasik0, 2015-07-27:
Hey,

currently pyUANamespace uses some non-userland functions which is a pity. Should we adapt the code/export the needed functions to userland? Are there some assumptions about "userland" NodeSets that can be made e.g. 
* no stale nodes or no nodes which parents do not exist -> no need to publish `UA_NodeStore_insert`

Can we map some of the internal functions to userland api, e.g. to use UA_Server_addObjectNode for objectNodes (this should be easy).

ichrispa, 2015-07-27:
The compiler does run a node reachability analysis before printing, so issue must arise from the ignorelist or blackllist/whitelist functionality.

List of functions needed to be published to userland (to be continued):
  * symmetric: 'UA_(Server|Client)_(add|delete)(Object|Variable)Node()' (note: all addNodes should return the actual nodeId inserted into the nodestore)
  * symmetric: 'UA_(Server|Client)_forEachChildOfDo(NodeId ofNode, callback do)'
  * symmetric: 'UA_(Server|Client)_(get|set)ValueOfVariable(NodeId variableNodeId, void *data, Datatype *type)'
  * symmetric: 'UA_(Server|Client)_createInstanceOfObjectType(nodeId objectTypeNode)'

Note: I have these functions in the depths of my applications, all I need is to backport them to master.

jpfr, 2015-07-27:
That's looking good @ichrispa !
But what do you mean by symmetric?

1. Returning the actual nodeid after adding is a good idea. The addnodes services does that already. So we can just reuse its mechanisms and structs.
2. What is a "child"? A component? A property? A subtype?
3. That's the same as the read/write services
4. That's the same as the addnodes service (also see 1)

To me, the question is whether we
- expose "wrappers" for the **existing UA services only** into the userspace (call the service with the admin-session)
- write convenience functions atop / replacing the UA services that are exposed via the network layer.

Maybe we can have a "layered" architecture.
`Really low level < UA Services < Convenience Functions`
Then, each layer only sees what is directly below.

ichrispa, 2015-07-28:
> But what do you mean by symmetric?
I mean that the function should be present in both server and client and have the same argument structure. In addition, if there is an "add" in terms of an orthogonal instruction set there should be a "delete" and a "get" were appropriate.

> What is a "child"? A component? A property? A subtype?
Any forward reference. The reference's nodeId can be passed to the user defined callback; that way the user can decide how to deal with the child; note that in my application I had an own reference type I was looking for.

> 3. That's the same as the read/write services
Indeed, but the server cannot invoke this function. On the serverside, I sometimes needed to update a single value, which always turned out to be messy. But if the server can do this, the client should be able to call something very similar.

> 4. That's the same as the addnodes service (also see 1)
Not quite. Imagine you have defined an ObjectType "Dog" that "hasProperty" VariableType "Name" and "hasComponent" ObjectType House. You might want to create an instance of a dog - which entails creating all three nodes. CreateInstance would create an Object for every objectType, a variable for each variable|dataType. This currently needs to be done by hand, but at least I would welcome a more generic approach. Does that make sense?

ichrispa, 2015-07-28:
Whoopsy... after that short travis scare I'm making a new branch for these API features.

@jpfr: How difficult would it be to relocate/redefine the UA_<type>Node struct's to make them accessible to the userspace? I would suggest we only allow a UA_(Client|Server)_getNodeCopy(nodeId) function and not expose the sensitive nodestore, but still allow the user to actually get a full struct if he wants to?

jpfr, 2015-07-29:
> How difficult would it be to relocate/redefine the UA_Node struct's to make them accessible to the userspace? I would suggest we only allow a UA_(Client|Server)_getNodeCopy(nodeId) function and not expose the sensitive nodestore, but still allow the user to actually get a full struct if he wants to?

When we expose the node, the user needs to `_release` the node all the time. That's already difficult to enforce in the server code. Instead, we can easily give (server side) userspace access to the UA services. Then, the 'normal' services already encapsulate the logic to prevent the user to do dumb things like breaking the information model consistency.

With the admin session, we can write very thin wrappers around the existing services. Like this internal function (used by addvariablenode et al).
```c
UA_AddNodesResult
UA_Server_addNode(UA_Server *server, UA_Node *node, const UA_ExpandedNodeId parentNodeId,
                  const UA_NodeId referenceTypeId) {
    return UA_Server_addNodeWithSession(server, &adminSession, node, parentNodeId, referenceTypeId);
}
```

> CreateInstance would create an Object for every objectType, a variable for each variable|dataType. This currently needs to be done by hand, but at least I would welcome a more generic approach.

The AddNodes service entails precise instantiation rules for variable and object types. But we haven' implemented them so far. Eventually, AddNodes will have to cover all the use cases you described. Note that also variable types may define 'property nodes' as children. For example with a unit description of the variable value. Imho the only use case that is not covered by the standard UA services is adding a new method to an object type (with a pointer to the function).

What we could do for C++-like object orientation is adding user-defined constructor and destructor function pointers to object types. These can then create user-defined data in the background that we point to from the object node (and forward to the method callbacks).

ichrispa, 2015-07-29:
> When we expose the node, the user needs to _release the node all the time.

I worked around that by implementing a UA_Server_getNodeCopy() function. Releasing the node is delegated there. So users can now add|delete nodes from the nodestore and copy|deleteCopy copies of the node. That should allow users to retrieve all information about a node without conflicting with node management, right?

The problem right now is that users have to include src/server/ua_nodes.h and its dependencies to investigate those node copies. That's kind of ugly. So I would suggest maybe putting nodes.h or a copy of the struct definitions into /include?

I have worked on a couple of tutorials and symmetrized the UA_(Server|Client)_add(Object|Variable)Node() functions; symmetry referring to the fact that the call looks the same, takes the same arguments in the same order and return a Statuscode. It's still work in progress though, as I still have to cover the rest of the addNodes in a similar fashion.

ichrispa, 2015-07-29:
Minor progress report on the API high level rehaul. 

@jpfr: here's what I meant with symmetric:

```
UA_StatusCode UA_EXPORT 
UA_Client_addObjectNode( UA_Client *client, UA_NodeId reqId, UA_QualifiedName browseName, 
                         UA_LocalizedText displayName,  UA_LocalizedText description, UA_ExpandedNodeId parentNodeId, 
                         UA_NodeId referenceTypeId, UA_UInt32 userWriteMask, UA_UInt32 writeMask, 
                         UA_ExpandedNodeId typeDefinition, UA_NodeId *createdNodeId)
                                                 
UA_StatusCode UA_EXPORT
UA_Server_addObjectNode(UA_Server *server, UA_NodeId nodeId, const UA_QualifiedName browseName, 
                        UA_LocalizedText displayName, UA_LocalizedText description, const UA_NodeId parentNodeId, 
                        const UA_NodeId referenceTypeId, UA_UInt32 userWriteMask, UA_UInt32 writeMask, 
                        const UA_ExpandedNodeId typeDefinition, UA_NodeId *createdNodeId);
```
Note how the calls are principally the same, you just substitute 'Client' for 'Server' and voila :)

The server now support addNode and deleteNode functions for all node types. The client... not so much (but the once it does are symmetric to the server). Next I will create the deleteNodes service, which is still completely missing.

Restructuring the functions to contain all node attributes is vital. @Stasik0  and of course myself want to reuse the namespace compiler for the userspace - these functions now make it possible once addMonodirectionalReference is also in the userspace. I am still searching for a neat way to bootstrap nodes.

Theoretically (though this is a long shot) a perfectly symmetrical userspace API might also allow users to generate a namespace model that is uploaded to the server from the client. I think that would be cool.

We also have the node iterator, getNodeCopy and deleteNodeCopy running.

ichrispa, 2015-07-30:
@Stasik0: After building around in the HL-Abstraction branch, we can use userspace functions for creating all node types. I need to solve 2 problems to get you the userspace capable code generation:

  * addMonodirectionalReference needs to be exported (not that much of a problem really, since addReference is already there).
  * We need to be able to bootstrap nodes. I believe the simplest way of getting there would be to allow for UA_NULL parentNodeIds in the standard addNode calls somehow. I would suggest using a reserved ID, such as ns=65536,i=4294967296 (ns=-1,i=-1) to indicate a null parent maybe? It would only be handled specially for parentNodeIds in add<XY>Node, not as a new global wildcard.

Stasik0, 2015-07-31:
@ichrispa  looks a bit dirty, but i am fine with it. @jpfr @FlorianPalm what about you, guys?

FlorianPalm, 2015-07-31:
Basicly I like your "symmetric"-approach @ichrispa. But instead of adding this functionality which apparently does not comply with the standard we could add a method and call it from the client-side, couldn't we? By doing this, we are free in chosing the parameters and semantics.

ichrispa, 2015-07-31:
@FlorianPalm: Client side addNodes mandate a parentId (unless we do not comply to the standard and ignore it). We have the same problem as we have on the serverside.

jpfr, 2015-08-01:
Every node (besides root) needs to have at least one hierarchical "upwards" reference.
Requiring a parent nodeid and reference type is a very elegant approach to enforce the constraint.

Since ns0 is bootstrapped in ua_server.c, I see currently no use case for parentless nodes.
Convince me with a good example, and I might change my mind.

Stasik0, 2015-08-01:
Imho this can be the case in the xml file that is not constructed properly, the python script has to "create" the correct insertion order.

jpfr, 2015-08-02:
> the python script has to "create" the correct insertion order

https://en.wikipedia.org/wiki/Topological_sorting

ichrispa, 2015-08-02:
So to conclude: We are not using any bootstrapping magic nodeids that oppose the spec. Instead, I will have the namespace class in the compiler determine a minimal dependency order of the nodes and reorder the printing sequence. Userspace add<XY>Node functions will be used wherever possible (the HL-Abstraction branch has them for all node types), or fall back to the current bootstrapping procedure if necessary.

What remains to be seen is how some "special" references like DataType will be affected by this. "Root" for example is a "Foldertype", a scenario which makes bootstrapping manditory. Given that namespace 0 exists, this conundrum might be avoidable in userspace nodesets.

Please note that solving the correct order will require some extensive matrix-decabubulation... solving a 1500x1500 matrix to be diagonal will make compilations a bit more lengthy.

jpfr, 2015-08-02:
> I am also considering avoiding the matrix and using a broad/width-search algorithm starting from root.

Yes! Here's an article I enjoyed with some python snippets.

http://www.embeddedrelated.com/showarticle/799.php
`There are two simple algorithms for a topological sort, Kahn’s algorithm and a depth-first-search algorithm. Both have execution times that are proportional to a weighted sum of the number of nodes and edges aN+bE.`

Stasik0, 2015-08-03:
@ichrispa +1 for no-bootstrapping in userland .xml files, just being curios, is it possible that UaModeler already enforces the right order so no action is required here?

ichrispa, 2015-08-04:
@Stasik0: No, UAModeler uses the node creation order as far as I can tell.

To my utter amazement I managed to rewrite the namespace compiler to use userspace functions whenever possible (17ec7526dd6b8bf7f663443407d120ff6c1e02fe); which for reasons I cannot explain (propably sunspots) appears to be working.

The compiler attempts to find an already initialized parent node for each node. If none can be found, it will fall back to the default bootstrapping behavior. Otherwise it will use the userland add<Type>Node() (which might explain why I was so opsessed with extending the API for all node types).

There are still several TODOs, including node reordering, not all node attributes being covered by the addNode functions and duplicate DataType/SubTypeOf References being created.

closed 2015-08-24 by Stasik0