open62541/open62541 #304: Namespace generator calls addNode multiple times

Stasik0, 2015-07-27:
for example

```
// Node: opcua_node_object_t(i=85), Objects
...
// Referencing node found and declared as parent: i=84/Root using i=35/Organizes
UA_Server_addNode(server, (UA_Node*) opcua_node_object_t_i_85, UA_EXPANDEDNODEID_NUMERIC(0, 84), UA_NODEID_NUMERIC(0, 35));
// Referencing node found and declared as parent: i=84/Root using i=35/Organizes
UA_Server_addNode(server, (UA_Node*) opcua_node_object_t_i_85, UA_EXPANDEDNODEID_NUMERIC(0, 84), UA_NODEID_NUMERIC(0, 35));
```

ichrispa, 2015-08-02:
The bug has been noted, I will fix this along with #303.

ichrispa, 2015-08-05:
Q: Do you want the fix backported to master or wait until the HL-Branch has taken a somewhat solid shape and gets merged back?

Stasik0, 2015-08-05:
Imho it is not that critical atm to be backportes.

Stasik0, 2015-08-07:
just to add another related bug -> the current version of the generator creates two HasTypeDefinitions for an ObjectTypeNode, one is created via the function UA_Server_addObjectTypeNode the second one explicitely via UA_Server_AddMonodirectionalReference directly afterwards

ichrispa, 2015-08-12:
The bug should be fixed in the latest commit to the hl-api branch.

Stasik0, 2015-08-12:
@ichrispa i confim, still there are double-relations presented e.g. for i=85 there are two HasTypeDefinitons to i=61... the first one seems to be created via addObjectNode even if i=61 does not exist at the moment. The second one is created by the bootstraping. Is addObjectNode not checking the existence of referenced nodes?
```
//  Creating this node has resolved the following open references:
...
UA_Server_AddMonodirectionalReference(server, UA_NODEID_NUMERIC(0, 85), UA_EXPANDEDNODEID_NUMERIC(0, 61), UA_NODEID_NUMERIC(0, 40), UA_TRUE);
```

ichrispa, 2015-09-01:
fixed in 4493355fb978d768ccb7d024e56bec3d55c71716

closed 2015-09-04 by Stasik0