open62541/open62541 #305: Segmentation fault when reading UA_ExtensionObject

billran, 2015-07-29:
Reading the (1, 51043) Demo node in the example server results in a segmentation fault.  I am using the MinGW tool chain in Eclipse Mars CDT on Windows 7 Pro and Open62541 release 0.1
[image]
.  See attached screen capture.  The calling code is taken from the example:

  UA_ReadRequest rReq;
  UA_ReadRequest_init(&rReq);
  rReq.nodesToRead = UA_ReadValueId_new();
  rReq.nodesToReadSize = 1;
  rReq.nodesToRead[0].nodeId = UA_NODEID_NUMERIC(1, 51043);
  rReq.nodesToRead[0].attributeId = UA_ATTRIBUTEID_VALUE;
  UA_ReadResponse rResp = UA_Client_read(client, &rReq);
  UA_ReadRequest_deleteMembers(&rReq);
  UA_ReadResponse_deleteMembers(&rResp);

ichrispa, 2015-07-29:
Try the current master instead of 0.1; if that is working, we can backport the fix.

I take it that the segfault is caused here?
'''
UA_ReadResponse rResp = UA_Client_read(client, &rReq);
'''

billran, 2015-07-29:
The seg fault is still in Git-Revision: 8d09a93 dated July 28th.  The UA_NodeId pointer in the function UA_NodeId_init() is null at the end of the read calling stack.  P.S.  I am building a 32 bit dll.

ichrispa, 2015-07-29:
Hi billran,

when trying to reproduce this error, NodeId(1,51043) is a Bytestring (Demo/Scalar/14). The node does not appear to have a value, which is exactly what is returned by the request (no segfaults, no contained DataValue). At least on x86_64 bit Linux (gcc/clang compiling /w and /wo -m32) this error is not reproducible. I'll give MSVS a try tomorrow.

Could you please provide a stack trace for the given exception?

Best regards,

ichrispa

billran, 2015-07-29:
I am testing against the demo server.exe in your 0.1 release.  That server has (1, 51043) as an ExtensionObject value for node 21.  Here's a screen shot from UaExpert:
[image]

ichrispa, 2015-07-31:
Hi @billran,

I cannot confirm the bug in the current master, meaning someone has already fixed this part. Also, the UA_Variant_decodeBinary() function that causes this bug appears to have undergone somewhat of a rewrite since 0.1.

As for fixing this error in 0.1, I will need to request some help from the recursive decoding binary gurus who build this part of the stack... 

@jpfr or @Stasik0: In https://github.com/acplt/open62541/blob/0.1/src/ua_types_encoding_binary.c#L803 decodeBinary is called to decode the dataType. However, nothing actually allocates the dst->data. dataType is ExtensionObject, but dst->data remains NULL. 

Should there be a  `dst->data = UA_new(dataType);` right above that line or is the dataType allocated by some other magical means?

billran, 2015-07-31:
Ok, and keep up the great work you all have been doing.

Stasik0, 2015-07-31:
@ichrispa just giving a quick look, doesnt the _init allocate the memore https://github.com/acplt/open62541/blob/0.1/src/ua_types_encoding_binary.c#L753?

Still no idea where it might break

@billran do you need the backport to 0.1, or could you imagine to switch to the newer version asap?

ichrispa, 2015-07-31:
@Stasik0:
>  doesnt the _init allocate the memore
No, _init(*ptr) just writes initial values to *ptr. _new() would allocate and then call _init().

Stasik0, 2015-07-31:
oh, you are right!

ichrispa, 2015-07-31:
I pushed 215fee64dcfc2701d576a8859ae1b4e539fcbebf, which appears to fix the issue. I would be more at ease if @Stasik0 or @jpfr could have a look at some point to verify that I did not make matters worse.

FlorianPalm, 2015-07-31:
For single extension objects it does solve the problem. As @Stasik0 stated UA_malloc is a bit faster. 
But I do not understand this line:
``C
 dst->data = UA_malloc(dataType->memSize & ((!isArray && arraySize==-1) ? 1: arraySize));
``
Shouldn't be the "bitwise and" a "multiplication sign"?

ichrispa, 2015-08-01:
Yep. That should be a "*".

jpfr, 2015-08-02:
Fixed a memleak: e1c8991546cbfe45a6c5d04f268d82edaac6395e.

Release-master @Stasik0, should we tag a 0.1.1 release?
Several important fixes have landed in the recent weeks.

Stasik0, 2015-08-02:
Yea, just tag it, everything else should work automatically

Stasik0, 2015-08-03:
@billran could you please test the newest 0.1.1 release? https://github.com/acplt/open62541/releases/tag/v0.1.1

billran, 2015-08-03:
Ok, I should be able to do that tomorrow .  I am traveling today.

billran, 2015-08-04:
The segmentation problem is fixed.  Thanks.

closed 2015-08-04 by Stasik0