open62541/open62541 #306: request for client and server API examples - cmake with BUILD_EXAMPLES configuration doesn't build

martina-felber, 2015-07-30:
Hello!

I am looking for more examples showing the usage of your client and server API. Are there more examples available than the provided ones in the /examples directory?

I thought that cmake would generate some examples with the option BUILD_EXAMPLES enabled. 
I am using Windows 7, configured the MinGW makefile option and enabled following configurations: 
BUILD_EXAMPLECLIENT, BUILD_EXAMPLESERVER, BUILD_EXAMPLES, ENABLE_ADDNODES, ENABLE_AMALGAMATION, ENABLE_METHODCALLS, ENABLE_SUBSCRIPTIONS
and got following errors:

[ 80%] Building C object examples/CMakeFiles/server_method.dir/server_method.c.o
bj
D:\Projekte\OPC\open62541-master_examples\open62541-master\examples\server_metho
d.c: In function 'main':
D:\Projekte\OPC\open62541-master_examples\open62541-master\examples\server_metho
d.c:70:29: error: too few arguments to function 'UA_Server_addMethodNode'
                             &helloWorldMethod, 1, &inputArguments, 1, &outputAr
guments);
                             ^
In file included from D:\Projekte\OPC\open62541-master_examples\open62541-master
\examples\server_method.c:15:0:
D:/Projekte/OPC/open62541-master_examples/open62541-master/build/open62541.h:389
5:1: note: declared here
 UA_Server_addMethodNode(UA_Server *server, const UA_QualifiedName browseName, U
A_NodeId nodeId,
 ^
examples\CMakeFiles\server_method.dir\build.make:62: recipe for target 'examples
/CMakeFiles/server_method.dir/server_method.c.obj' failed
mingw32-make[2]: *** [examples/CMakeFiles/server_method.dir/server_method.c.obj]
 Error 1
CMakeFiles\Makefile2:349: recipe for target 'examples/CMakeFiles/server_method.d
ir/all' failed
mingw32-make[1]: *** [examples/CMakeFiles/server_method.dir/all] Error 2
Makefile:82: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

I saw your example to add an int32 value (the answer) and arrays, scalars and matrices with every built-in datatype. 
On the server side I would like to know how to add more complex variables, than the built-in types and how to define the VariableTypes/Objecttypes/Foldertypes. I thought of a simple test set up like this:

Foldertype
Server
Machine 1
      Parameter
            Temperature
            Humidity
            Speed
            Something Else
            ...
      Miscellaneous
            ...

Machine 2
      Parameter
            Temperature
            Humidity
            Speed
            Something Else
            ...
      Miscellaneous
            ...

Machine 3
      Different Params
      SensorValues
            Sensor 1
                  Value XY
            Sensor 2

How to define a common Objecttype for Machine 1 and 2's equal variables?

On the client side I could list the nodes Foldertype, Server, Machine 1, Machine 2 and Machine 3. Thanks for your "Browsing nodes in objects folder" example using function UA_Client_browse. In order to discover i.e. the nodes of machine 1, I had to start another browse request with the nodeID of machine 1 and got the "Parameter" object. Is there any easier way to get every node of a specific branch? I tried to use BrowseNext and TranslateBrowsePathsToNodeIds, but couldn't get it working without an example. 

Many thanks for your help
Best regards
Martina

ichrispa, 2015-07-30:
You could have a look at https://github.com/acplt/open62541/blob/api_hl_abstractions/doc/firstSteps.rst and https://github.com/acplt/open62541/blob/api_hl_abstractions/doc/tutorial_nodes.rst; but be warned that both are still very much work in progress.

The errors you reported by BUILD_EXAMPLES are my bad. I will fix those ASAP.

ichrispa, 2015-07-30:
The examples should build fine now.

> How to define a common Objecttype for Machine 1 and 2's equal variables?
On the serverside, you can use UA_Server_addObjectTypeNode(); I think it currently only exists in the high level abstraction branch. The function is also provided for the client.

> On the client side I could list the nodes Foldertype, Server, Machine 1, Machine 2 and Machine 3.
The High-Level abstraction branch also holds node iterators called UA_(Server|Client)_forEachChildOfCall. Have a look if those are not a bit more fitting for executing your node search. Otherwise you could just reuse their code in your application.

martina-felber, 2015-07-30:
@ichrispa  many thanks to you!
the BUILD_EXAMPLES configuration works, and thanks for your tutorials
unfortunately I wasn't able to find any function "UA_(Server|Client)_forEachChildOfCall"

ichrispa, 2015-07-30:
Hi @martina-felber,

you need to checkout the current api_hl_abstractions branch (https://github.com/acplt/open62541/tree/api_hl_abstractions) in which I am working on making our API a bit more user friendly by abstracting the OPC UA technical details as much as possible. I am also working on the given tutorial to accompany those API functions.

UA_Server_forEachChildNodeCall() is defined in UA_Server.h, line 304
UA_Client_forEachChildNodeCall() is defined in UA_Client.h, line 108
UA_Server_addObjectTypeNode() is defined in UA_Server.h, line 216
UA_Client_addObjectTypeNode() is defined in UA_Client.h, line 83

jpfr, 2015-07-30:
Hey @ichrispa,
great to see the passion and the optimism in your work.

But please do not recommend alpha alpha feature branches to end users. Recommending the master branch instead of stable releases is already borderline. Things change daily in the development branches and this will be very frustrating for someone whose code breaks.

ichrispa, 2015-07-30:
Hi @jpfr,

thanks and you are of course right about the everchanging functionality of features in development and potential bugs. You are very welcome to suggest to @martina-felber how she can add <Object|Variable>Type nodes or instances thereof using the 0.1 release (or master), as I cannot find a solution given their current APIs.

jpfr, 2015-07-30:
> You are very welcome to suggest to @martina-felber how she can add Type nodes or instances thereof using the 0.1 release (or master), as I cannot find a solution given their current APIs.

Touché :-)
Currently, the only way to add an objecttype in master is to call "internal" APIs or the AddNodes service over the wire.
https://github.com/acplt/open62541/blob/master/src/server/ua_services_nodemanagement.c#L196
However, having the objecttype in itself doesn't help since we don't properly instantiate.

Your work is definitely on the right track. Just take care that we don't end up with duplicated functionality in several places.

>  Is there any easier way to get every node of a specific branch?

@martina-felber That's exactly what the browse service is for. You select a reference type (and its subtypes) and get all children of a node that match the type. If UA Expert or some other client GUI show the correct structure, they are using the browse service to discover the children as well.
How many references are returned for your request?

martina-felber, 2015-07-31:
@ichrispa thanks for your link to the high level abstraction branch, sorry I didn't get it at your post before. 
I built the high level branch code with ENABLE_EXAMPLESERVER, ENABLE_EXAMPLECLIENT, ENABLE_AMALGAMATION, ENABLE_SUBSCRIPTIONS, ENABLE_ADDNODES, ENABLE_METHODCALLS
at the first try it works fine, I added the generated code to eclipse and added some nodes (for Machine 1) to the server, which UAExpert displays correctly (thanks for the hint to use this tool) 

afterwards I tried to debug the client code, but every time I try I get a segmentation fault during debugging (below is a picture of my eclipse window):
[image]

@jpfr > How many references are returned for your request? 
Sorry I don't understand your question. I get the same results for UA_Client_browse and UA_Client_forEachChildNodeCall, except with UA_Client_forEachChildNodeCall there's one more node in the Objects Folder with namespace 0 and ID 84. After the request for the Objects Folder, I have to start a new request for Machine 1, and afterwards for Params, is that what you mean?

Another issue is the method call function: no matter which example code I tried (master, high level) the call is always unsuccessful, with following client output:
Method call was unsuccessfull, and 80750000 returned values available. 

I'm sorry that I take so much of your time.
Many thanks and best regards
Martina

ichrispa, 2015-07-31:
Hi @martina-felber ,

> After the request for the Objects Folder, I have to start a new request for Machine 1, and 
> afterwards for Params, is that what you mean?

Yes, you will have to start a new request there. There is no way of knowing which references or nodes you (i.e. the user) is interested in examining. If you want to see the instance tree, you should create new iteration requests for hasComponent, organizes and hasProperty type references.

> Another issue is the method call function: no matter which example code I tried (master, high level) 
> the call is always unsuccessful, with following client output:
> Method call was unsuccessfull, and 80750000 returned values available.

You can return the number of values returned, as that looks like the error code UA_STATUSCODE_BADMETHODINVALID. Quite propably the method node the example calls does not exist on the server.

> @jpfr >How many references are returned for your request? 
> Sorry I don't understand your question.

A browseRequest returns a number of results in the browseResponse, along with a possible contiuation point. I would suggest you look either look it up in the specification of OPC UA Part 4 or you monitor the communications with wireshark (that has an OPC UA decoder) to see what the server returns.

Best regards,

ichrispa

martina-felber, 2015-08-03:
Hello!

I checked out the current master code on Friday and today and tried several times, but I couldn't get the methodcalls working. The response of the client is always: 
Method call was unsuccessfull, and 80750000 returned values available.
the return value is UA_STATUSCODE_BADMETHODINVALID as @ichrispa already said.
and the server doesn't write "getMonitoredItems was called".

Stasik0, 2015-08-03:
Is it working with UaExpert?

martina-felber, 2015-08-03:
No, I get the error message: Call failed with status 'BadMethodInvalid'

Stasik0, 2015-08-03:
@martina-felber it seems to be fixed now, can you confirm?

martina-felber, 2015-08-03:
yes, thank you very much, the method call works now =)

closed 2015-08-03 by martina-felber