open62541/open62541 #309: update variable (for example "the.answer") periodically or at any specific time at the server?

martina-felber, 2015-08-03:
Hello!

I have a few questions:

1. Is it possible, that the server updates a variable node, for example "the.answer" periodically? Or at any specific time?
I tried to use UA_Server_addRepeatedJob() as in your example to call every 2 seconds a callback function. Then I don't know what to do next. The Client provides an UA_Client_write function, but I couldn't find an Ua_Server_ function to update a variable node's value. 
Do I have to delete the node and create it with a new value?

2. The subscription of "the.answer" works fine, but does the client always have to poll with UA_Client_doPublish? Is there another mechanism like an "interrupt", so the client gets notified and jumps to a callback function?

Many thanks in advance.
Best regards
Martina

ichrispa, 2015-08-03:
Hi @martina-felber,

> Is it possible, that the server updates a variable node, for example "the.answer" periodically? Or at any specific time?

If you want to update the variable periodically I would suggest you use a dataSource Variable node (see for example 'current time' or 'cpu temperature'). I am aware of the shortcoming of our API and I am working on a `UA_(Server|Client)_(get|set)AttributeValue()` function for the hl-abstraction branch - which, again, I cannot recommend using yet.

If you do feel adventurous you can try the UA_(Server|Client)_delete<Type>Node() functions in a timed job - wouldn't recommend it though.

> The subscription of "the.answer" works fine, but does the client always have to poll with UA_Client_doPublish? Is there another mechanism like an "interrupt", so the client gets notified and jumps to a callback function?

The current subscription model is kind of interrupt driven in that doPublish() of the client automatically triggers/calls the function you registered when a change is detected. But since there is no main loop for Clients yet (suggested in #279), you do still have to call doPublish() manually at regular intervalls.

Best regards,

ichrispa

martina-felber, 2015-08-04:
Hi ichrispa!

Many thanks for your answer. I didn't see the refresh button of UaExpert at the Attributes window and therefore didn't realize that at every read the current time value gets updated.

Reading with the open62541 Client works fine, but I can't write a value. I copied the client code from writing a value to the.answer and changed the datatype to UA_DateTime. I tried to set the time back to 0, the status code is good, but the current time is still the actual UTC time. Is this, because there's no write function defined at the DataSource? Unfortunately the status code good is returned. 

I tried to define a new DataSource node, with a read and write function. At each read a variable counts up, from 0 beginning. The write function sets the variable to the given data value. The strange thing is, that UaExpert calls the function always three times when I click the refresh button, but at your open62541 client reading and writing works fine. Do you know if this is a bug from UAExpert? Or have I done something wrong?

May I ask another question? I didn't completely understand your example function „readTimeData“, in which you look at the beginnning if any range is defined.
if(range) {
        value->hasStatus = UA_TRUE;
        value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
        return UA_STATUSCODE_GOOD;
    }

You don't expect a defined range so the status of value is BADINDEXRANGEINVALID but you return statuscode good. Should this function always return statuscode good, and any error has to be written into value->status? If so, how to return an error code at the write function?

Many thanks and best regards
Martina

ichrispa, 2015-08-05:
Hi @martina-felber,

dataSources have a .read and .write function; writing currentTime has no effect, since afaik the function is not defined.

The function themselves return a StatusCode to indicate whether they succeeded. If a function reports that it has failed to execute, it can not be expected that the  DataValue returned by it is valid. If an error is to be conveyed to the client, but the function technically did perform its job, the statuscode in the DataValue is returned to the client (which may be something other than GOOD).

As far as reading values with UAExpert is concerned, I highly suggest you monitor the communications with wireshark or a similar tool (wireshark has an OPC UA protocol decoder). UAExpert propably uses Browse or Read multiple times in an inefficient way that causes the value to be read.

ReadTimeData does not return a range, but a scalar. Any request for a range indicates the reading of an array/matrix, which is not in accordance with the nodes valueRank. Hence requesting a range is an error in itself (note that the DataValue is initialized by readTimeData, so the function returns GOOD).

As far as your last question is concerned I suggest you set a breakpoint in you read function and take a look at the the range contents using a debugger (I would suggest gdb). Compare that to the information on the wire sent by wireshark. I have never used ranges in dataSources before, so I can't give you more then generic advice on that subject.

Regards,

ichrispa

martina-felber, 2015-08-06:
Many thanks for your answer and advice!
I don't know why, but it seems as UaExpert calls a read request multiple times for each data source node.
best regards Martina

closed 2015-08-06 by martina-felber