open62541/open62541 #310: planned releases and their features in the future (within 1-2 years)

martina-felber, 2015-08-03:
Hey!

I would like to know, which releases and features you approximately plan within a year or two. Or if you plan to expand the stack for another profile.
Many thanks and best regards
Martina

ichrispa, 2015-08-05:
Hi @martina-felber,

I cannot answer for everyone involved in this project, but I will reluctantly admit that the stack is being developed on a best-effort basis and not on a fixed feature roadmap. My current take is to implement whatever I can when it becomes necessary for me or my colleagues to use them (and of course if time permits).

I would ultimately like to have a full featured OPC UA stack with compile-time configurable profiles in open62541, but there is still some way to go.

Regards,

ichrispa

martina-felber, 2015-08-06:
Many thanks for your honest answer!
Best regards
Martina

closed 2015-08-06 by martina-felber