open62541/open62541 #311: Write service only supports value (why?)

ichrispa, 2015-08-03:
Hi all,

I am just finishing up ```UA_(Server|Client)_(get|set)AttributeValue(..., UA_AttributeId attributeId, void **value)``` for the hl-abstraction branch. The server can write any value but nodeId and nodeClass (node is being copied, modified then replaced by my implementation).

Given the current implementation of ```UA_StatusCode writeValue(UA_Server *server, UA_WriteValue *wvalue)```, the client can only set the value attribute. Is there a part of the spec that I missed? Why is the write service so restrictive?

Best regards,

ichrispa

jpfr, 2015-08-03:
`WriteValue` contains an `AttributeId` field. So in principle you can write to any attribute of the nodes.
Feel free to add missing functionality. :-)

ichrispa, 2015-08-03:
Thanks for @HolgerJeromin for adding the label and @jpfr for the feedback :)

I have ```UA_Server_setAttributeValue(UA_Server *server, UA_NodeId nodeId, UA_AttributeId attributeId, void *value)```. Apart from parsing the value range field from the request, the function should serve to write just about any value - again excluding NodeId and NodeClass. If @jpfr and @Stasik0  as the original implementers don't object, I would try to slightly rewrite the writeValue service to redirect the actual node modification to that (userspace) function and avoid code duplicity?

Stasik0, 2015-08-03:
No objections here

jpfr, 2015-08-03:
> I would try to slightly rewrite the writeValue service to redirect the actual node modification to that (userspace) function and avoid code duplicity?

The service will get a lot bigger in the future. For example, if we add node-level user access control, information model consistency rules, and so on.

To keep things consistent, we can either generally
- Beef up the userspace functions to take session information, etc. (which user is trying to do stuff) as an input and make the services lightweight wrappers around the userspace function.
- Have the functionality in the service and make the userspace function a wrapper around the service (with special 'admin credentials').

ichrispa, 2015-08-03:
Clientside abstractions use services as a rule, so the hl-capabilities are just macros for services.

As far as the serverside is concerned, I'd generally prefer the second approach, i.e.:
> Have the functionality in the service and make the userspace function a wrapper around the service (with special 'admin credentials').

Unfortunately, that means that my current approach is kind of wrong... so instead of redirecting the service to my function I guess I should add the functionality to the service, right? The hl-api should then only construct a UA_WriteValue struct and pass it to the service.

How do we solve the issue of all service functions being ```static``` in the respective service.c file? Should ua_server.h include all service headers (which don't exist yet)?

ichrispa, 2015-08-04:
My question has been adequately answered. I'll consider how to re-implented the serverside hl-abstractions using the existing services once the namespace compiler is up and running.

closed 2015-08-04 by ichrispa