open62541/open62541 #312: Problem with deleting a node in the node hirachy (UA_STATUSCODE_BADENCODINGERROR)

Hexsc, 2015-08-04:
Hey guys,

First of all, thank you for your excellent source code,
I had some great (and sometimes not so great) hours with it.

I've tried several versions of open62541, ran a couple of examples and mostly all are working now.
Now my question, I added some nodes to the node hirachy in my example and can delete them without problems with UA_Server_deleteNode(UA Server* server, UA_NodeId nodeId); (high level branch) by the server.
So I tried to delete a node with the client, using:

UA_DeleteNodesRequest delReq;
UA_DeleteNodesRequest_init(&delReq);
delReq.nodesToDelete->deleteTargetReferences = 1;
delReq.nodesToDelete->nodeId = UA_NODEID_STRING_ALLOC(1, "the.answer");
delReq.nodesToDeleteSize = 1;
UA_DeleteNodesResponse delResp = UA_Client_deleteNodes(client, &delResp);
if(delResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
{
	printf("Node has been deleted!\n\n");
}

I can compile the code with eclipse luna using mingw compiler.
If I debug it, sometimes it crashes at delReq.nodesToDelete->deleteTargetReferences = 1; and sometimes that works, but the statuscode I get back is: UA_STATUSCODE_BADENCODINGERROR.
After some time I ignored that and tried to start the client exe, but than a window opened which told me that there has been a problem with the client and it has to be closed.
The server is working fine (tried it with UaExpert) and without this few code lines, the client works also, so the problems origin has to be there.

Does somebody know, what I've done wrong? (I am unfortunately bloody new with OPC_UA.)

Thanks in advance.

With best regards,
Hexsc

ichrispa, 2015-08-04:
hi @Hexsc 

the highlevel branch is mayor experimental magic, using it is not recommended. Other than that I would suggest you give UA_Client_delete<Type>Node() a try, which should simplify things. 

```
UA_DeleteNodesResponse delResp = UA_Client_deleteNodes(client, &delResp);
```

Shouldn't your client be sending a &delReq at this point?

Regards, 

ichrispa

Stasik0, 2015-08-04:
@Hexsc but back to your original issue, the problem is that you are dereferencing a NULL pointer ```delReq.nodesToDelete```. Before assigning a value, you need to allocate memory e.g. ```delReq.nodesToDelete = UA_DeleteNodesItem_new();```

Hexsc, 2015-08-04:
Thanks for your fast answers!

@ichrispa UA_Client_deleteNode() is unhappily not working, the node has not been deleted and all his parameters are like before. 

Server:
UA_Variant *myIntegerVariant = UA_Variant_new();
    UA_Int32 myInteger = 23;
    UA_Variant_setScalarCopy(myIntegerVariant, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    const UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "the answer");
    const UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "the.answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myIntegerNodeId, myIntegerName, UA_LOCALIZEDTEXT("en-US", "int32_lt"),
    					UA_LOCALIZEDTEXT("en-US", "int32_lt"), parentNodeId, parentReferenceNodeId, 0, 0, myIntegerVariant, NULL);
(You should know this example, after all it's your own)

Client:
UA_Client_deleteNode(client, UA_NODEID_STRING(1, "the.answer"));

&delReq instead of &delResp is of course right, I don't know what got into me. 
But it is still not working, even not with the suggestion of @Stasik0. I've tried that before, but it does not cause any changes.
I think, my solution is unfortunately logically not working at all.

Once more, thank you for trying to help me.

With best regards,
Hexsc

closed 2015-08-04 by Hexsc