open62541/open62541 #313: Authenticate Clients with username/password

Hexsc, 2015-08-04:
Hi!

I have a little question to you.
Is it possible to authenticate clients with username/password at a server with the actual master branch?
At open62541.c I found:

...........
.Login_enableAnonymous = UA_TRUE,				// in my case UA_FALSE
.Login_enableUsernamePassword = UA_TRUE,
.Login_usernames = (char *[]){"user1","user2"},
.Login_passwords = (char *[]){"password","password1"},
...........

so it schould be possible to manage that, right?

But I found no place at the client, neither at the darkest corners of the client definition, where a username/password can be added.

Thanks for your answers in advance.
With best regards,
Hexsc

Stasik0, 2015-08-04:
Hi,

it is only possible with external clients e.g. with UaExpert, not with our client, yet. Should not be that hard to add, though.

Hexsc, 2015-08-05:
Ok, thank you very much!

HolgerJeromin, 2015-08-05:
I would like to let this issue open till it is possible with our client.

jpfr, 2015-12-04:
So we would have `UA_Client_connectAnonymous`, `UA_Client_connectUsername`, ...?

HolgerJeromin, 2015-12-04:
Makes sense.

Stasik0, 2015-12-14:
hey guys, what about to add some "authentication=UA_CLIENT_NONE" field to the client structure? In this case we can keep one function and not break existing code...

also..

currently the connect function selects the *first* endpoint with anonymous access.... this is probably not good aswell, what can we do, i propose to support "opc.tcp://host:port/endpoint" syntax to enforce the endpoint and add ```UA_Client_getEndpoints(UA_String, UA_Client, UA_String**)``` method returning an array of enpoint strings

jpfr, 2016-03-05:
Can we tag this either 0.2 or 0.3?
Getting the endpoints is ready. Now we need to distinguish between them in the connect function.

Stasik0, 2016-03-05:
+1 for 0.2, i suggest putting authentication into client structure? Can we elaborate an api?

Stasik0, 2016-03-11:
Ping, so, what about the api? Should we leave the method signature unchanged and add the connection parameters to the client config struct?

jpfr, 2016-03-15:
Implemented by @Stasik0 in #558.
So this is done.

closed 2016-03-15 by jpfr