open62541/open62541 #314: memory leak at UA_Server_addVariableNode - if parentNodeId is UA_NODEID_STRING

martina-felber, 2015-08-06:
Hello!

I have a problem with memory leaks using UA_Server_addVariableNode, UA_Server_addObjectNode or any other UA_Server_add method. The leak occurs when the parent nodeId is allocated at the call with UA_NODEID_STRING.

example code:
```C
	// following code creates a memory leak:
	/* add folder /Root/Objects/Params */
	UA_Server_addObjectNode(server, UA_QUALIFIEDNAME(1, "Params"), UA_NODEID_STRING(1, "Params"),
		UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
		UA_NODEID_NUMERIC(0, UA_NS0ID_FOLDERTYPE));

	/* add boolean variable "BoolDummy" to /Root/Objects/Params/ */
	UA_Variant *BoolVariant = UA_Variant_new();
	UA_Boolean BoolDummy = UA_TRUE;
	UA_Variant_setScalarCopy(BoolVariant, &BoolDummy, &UA_TYPES[UA_TYPES_BOOLEAN]);
	// M E M O R Y   L E A K !!!!!!!!!!!!!!!!!
	UA_Server_addVariableNode(server, BoolVariant, UA_QUALIFIEDNAME(1, "Boolean Param"),
		UA_NODEID_STRING(1, "Bool"), UA_NODEID_STRING(1, "Params"), // <- here are 6 bytes leak
		UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES));

	 // parent NodeId - UA_NODEID_STRING(1, "Params") is not deallocated

	//////////////////////////////////////////////////////////////////////////////////////////
	// following code does not create a memory leak, because parent nodeId is numeric
	UA_Server_addObjectNode(server, UA_QUALIFIEDNAME(1, "Params1"), UA_NODEID_NUMERIC(1, 2),
		UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
		UA_NODEID_NUMERIC(0, UA_NS0ID_FOLDERTYPE));

	/* add boolean variable "BoolDummy1" to /Root/Objects/Params1/ */
	UA_Variant *BoolVariant1 = UA_Variant_new();
	UA_Boolean BoolDummy1 = UA_TRUE;
	UA_Variant_setScalarCopy(BoolVariant1, &BoolDummy1, &UA_TYPES[UA_TYPES_BOOLEAN]);
	UA_Server_addVariableNode(server, BoolVariant1, UA_QUALIFIEDNAME(1, "Boolean Param"),
		UA_NODEID_STRING(1, "Bool1"), UA_NODEID_NUMERIC(1, 2), // <- no leak with numeric
		UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES));

	//////////////////////////////////////////////////////////////////////////////////////////
	// following code creates a memory leak
	UA_NodeId Params1NodeId = UA_NODEID_STRING(1, "Params1");

	UA_Server_addObjectNode(server, UA_QUALIFIEDNAME(1, "Params1"), Params1NodeId,
		UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
		UA_NODEID_NUMERIC(0, UA_NS0ID_FOLDERTYPE));

	/* add boolean variable "BoolDummy2" to /Root/Objects/Params1/ */
	UA_Variant *BoolVariant2 = UA_Variant_new();
	UA_Boolean BoolDummy2 = UA_TRUE;
	UA_Variant_setScalarCopy(BoolVariant2, &BoolDummy2, &UA_TYPES[UA_TYPES_BOOLEAN]);
	UA_Server_addVariableNode(server, BoolVariant2, UA_QUALIFIEDNAME(1, "Boolean Param"),
		UA_NODEID_STRING(1, "Bool2"), Params1NodeId,
		UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES));

	// UA_Server_addVariableNode -> Ua_NodeId_copy() -> memory leak!!!!!!!!!!!!!!!

```

Is this an intended behaviour? How can I add variable nodes or other nodes to folders with a string nodeId without having a memory leak?

Many thanks and best regards
Martina

FlorianPalm, 2015-08-06:
Hi Martina, you are right. A UA_ExpandedNodeId_deleteMembers() call within UA_Server_addVariableNode is missing. Thank you for your report :+1:

martina-felber, 2015-08-07:
you are welcome!
I am using your release v0.1.1, are you going to update it within two weeks at the latest?

closed 2015-08-07 by FlorianPalm