open62541/open62541 #315: UA_DataValue and UA_DiagnosticValue Use of Bit Fields

billran, 2015-08-07:
UA_DataValue and UA_DiagnosticValue structs use bit fields for boolean flags.  Both java and C# do not have native language support for bit fields.

In my case (I am attempting to write a java client), this does not allow passing a UA_ReadResponse object back to java for example.  Instead, I would have to define a parallel set of structs, then copy all the data out of the UA_ReadResponse into the new structs, then pass that object back to java.  This would have to be done for every case where these structs are members.

Instead, would it be possible to change the implementation to use individual bytes, or if size is a concern to mask off a single byte?

If you do not think this is a good idea, could you give me some guidance on how to write and populate these new structs?

Stasik0, 2015-08-07:
hi @billran, you are right: using bitfields is a matter of space-efficiency since one of our goal is to target embedded devices and to minimize the footprint of the stack

what about doing a trick like that?

```
typedef struct {
    UA_Boolean    hasValue : 1;
    UA_Boolean    hasStatus : 1;
    UA_Boolean    hasSourceTimestamp : 1;
    UA_Boolean    hasServerTimestamp : 1;
    UA_Boolean    hasSourcePicoseconds : 1;
    UA_Boolean    hasServerPicoseconds : 1;
    UA_Variant    value;
    UA_StatusCode status;
    UA_DateTime   sourceTimestamp;
    UA_Int16      sourcePicoseconds;
    UA_DateTime   serverTimestamp;
    UA_Int16      serverPicoseconds;
} UA_DataValue;

union UA_DataValue_union {
    UA_Int32 i;
    UA_DataValue dataValue;
};
```
and than carefully reading the bits out of the integer i? Probably it would either not work in Java (no union types,  eh?).

Cannot you just cast the pointer to UA_DataValue to (UA_Int32*) and then pass it back to java? Or generate a set of C mutators that are working as an interface to the java-world?

An the last suggestion, if anything else does not work, I would not mind if you introduce a precompiler-switch e.g. NO_BITFILEDS and reimplement the respective structs depending on this switch to handle every bit explicitely in the C code.

billran, 2015-08-16:
To get a working java client that supports read, write and browse in release 0.1.1, I used the BridJ open source project (https://code.google.com/p/bridj/) which supports bit fields, but does not support passing structs by value.  Structs must be passed by reference (i.e. a pointer argument or return pointer).

This restriction necessitated creating wrapper functions around the Open62541 functions.  The wrapper when necessary allocated heap memory, copied the struct data and then returned a pointer to it.  Since the caller is now responsible for freeing that memory, a corresponding "delete" function was written.

For example, here is the function to read a node and one to free the allocated memory:

UA_ReadResponse *synchRead(UA_Client *client, UA_NodeId *nodeId) {
	UA_ReadRequest rReq;
	UA_ReadRequest_init(&rReq);
	rReq.nodesToRead = UA_ReadValueId_new();
	rReq.nodesToReadSize = 1;
	rReq.nodesToRead[0].nodeId = *nodeId;
	rReq.nodesToRead[0].attributeId = UA_ATTRIBUTEID_VALUE;

	UA_ReadResponse rResp = UA_Client_read(client, &rReq);

	UA_ReadResponse *reply = UA_ReadResponse_new();
	UA_ReadResponse_copy(&rResp, reply);

	UA_ReadRequest_deleteMembers(&rReq);
	UA_ReadResponse_deleteMembers(&rResp);

	return reply;
}

void deleteReadResponse(UA_ReadResponse *readResponse) {
	UA_ReadResponse_deleteMembers(readResponse);
}

The java caller looks like this:
	private boolean readNode(Pointer<UA_Client> client,
			Pointer<UA_NodeId> nodeToRead) {
			
		// read the node
		Pointer<UA_ReadResponse> response = Open62541ExtLibrary.synchRead(client, nodeToRead);
				
		// print out the data read
		boolean rv = dumpNodeRead(response);

		// free the response
		Open62541ExtLibrary.deleteReadResponse(response);

		return rv;
	}

I tested a recursive browse down the Objects folder and  reading and writing the most common Demo folder scalar values.  
That all worked fine, so this is a viable design for a java client.

closed 2015-08-16 by billran