open62541/open62541 #316: Problem build in Visual Studio 2013

Kal-elpatch, 2015-08-07:
Hi guys!
I  build the open62541.sln with CMAKE gui, but occurs this error when compile

error MSB6006: "cmd.exe" exited with code 9009.	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\Microsoft.CppCommon.targets	170	5	open62541-object

Someone can help me please?

Stasik0, 2015-08-07:
Which branch are you using? Which msvs version are you using?

Kal-elpatch, 2015-08-07:
I use Microsoft Visual Studio Ultimate  2013  Version 12.0.31101.00 Update 4
I download the source code from
http://open62541.org/ .zip Button
and from
https://github.com/acplt/open62541.git

Kal-elpatch, 2015-08-07:
Master branch sorry

Stasik0, 2015-08-07:
Afaik open62541 builds only with msvs 15 due to the lack of c99 support in the older versions :( Master is compiled with vs15 by our ci server.

closed 2015-08-07 by Stasik0