open62541/open62541 #317: Bug in open62541.c

tpiat, 2015-08-07:
If write 
UA_ClientNew and next UA_ClientDelete,
clientDelete try to free memory in clinet-token, this memory was not allocated, no any truobles, but it is  possble error.
I can remove this effect by addintion string 

UA_UserTokenPolicy_deleteMembers(&client->token);

in UA_Client_delete;

FlorianPalm, 2015-08-07:
It is a free on a null pointer (the memory for the client is allocated with calloc) . Afaik this should not be a problem.

tpiat, 2015-08-07:
I have receive not-null pointer.

FlorianPalm, 2015-08-07:
You are right. The bug was fixed in the master but not in the current release. Thanks!

closed 2015-08-07 by FlorianPalm

tpiat, 2015-08-19:
I'm sorry... but I have SIGSEGV at line 10284
function UA_Client_delete,
UA_UserTokenPolicy_deleteMembers(&client->token); 

 in new version. In version 1.0  this signal was happend without this string.
Probably, in this version (1.1.0) this function call no longer needed.