open62541/open62541 #319: Build on OSX

schumajs, 2015-08-07:
Hi guys,

I managed to build the project and run the examples on OSX with some minor adjustments. I can submit a patch file but the latest revisions seem to break the examples. Correct me if I'm wrong and if I'm just doing something wrong. Let me know if you're interested in having this merged. If so I would merge the newest revisions into my feature branch first.

Best,

Jens

Stasik0, 2015-08-07:
hey @schumajs, yea sure. Unfortunately afaik no-one of the core team has a mac running so we can not test directly. OSX support is clearly desired!

schumajs, 2015-08-07:
Ok, I have trouble to make it work at the moment. The last working pull was around 1 pm today (before a whole bunch of new revisions were pushes). Did you rebase the repo by any chance?

Stasik0, 2015-08-07:
Master is protected from history-rewrites, so the changes are only incremental. Can you spot the exact problematic commit?

schumajs, 2015-08-07:
The last working revision for me was 886053f61954d7954263094c9afb1f004cebc77a. After that i got lots of requests to merge manually:

Auto-merging tools/generate_datatypes.py
CONFLICT (content): Merge conflict in tools/generate_datatypes.py
Auto-merging tools/amalgamate.py
CONFLICT (content): Merge conflict in tools/amalgamate.py
Auto-merging src/ua_util.h
CONFLICT (content): Merge conflict in src/ua_util.h
Auto-merging src/ua_types_encoding_binary.c
CONFLICT (content): Merge conflict in src/ua_types_encoding_binary.c
Auto-merging src/ua_types.c
Auto-merging src/server/ua_services_session.c
Auto-merging src/server/ua_server_worker.c
CONFLICT (content): Merge conflict in src/server/ua_server_worker.c
Auto-merging src/server/ua_server_internal.h
CONFLICT (content): Merge conflict in src/server/ua_server_internal.h
Auto-merging src/server/ua_server_binary.c
CONFLICT (content): Merge conflict in src/server/ua_server_binary.c
Auto-merging src/server/ua_server.c
CONFLICT (content): Merge conflict in src/server/ua_server.c
Auto-merging src/server/ua_securechannel_manager.c
Auto-merging src/client/ua_client.c
CONFLICT (content): Merge conflict in src/client/ua_client.c
Auto-merging include/ua_server.h
CONFLICT (content): Merge conflict in include/ua_server.h
Auto-merging include/ua_client.h
Removing examples/server_simple.c
Auto-merging examples/networklayer_tcp.c
CONFLICT (content): Merge conflict in examples/networklayer_tcp.c
Auto-merging appveyor.yml
CONFLICT (add/add): Merge conflict in appveyor.yml
Auto-merging README.md
CONFLICT (content): Merge conflict in README.md
Auto-merging CMakeLists.txt
CONFLICT (content): Merge conflict in CMakeLists.txt
Auto-merging .travis.yml
CONFLICT (content): Merge conflict in .travis.yml
Automatic merge failed; fix conflicts and then commit the result.

I'm not sure where it comes from but we were able to reproduce the problem on all machines that were on commit 886053f61954d7954263094c9afb1f004cebc77a before.

schumajs, 2015-08-07:
Note: We didn't change any of the files.

Stasik0, 2015-08-08:
The commit was in 0.1 branch, wasn't it? And now you probably want to merge 0.1 and master which is a huge difference.

Probably it would be easier to reapply    the changes on the master from the scratch. How big is the diff?

jpfr, 2015-08-10:
> We didn't change any of the files besides CMakeLists.txt.

@schumajs
Great! Then we can probably integrate the changes in no time.
Can you show the result of a simple `git diff CMakeLists.txt 0.1` to compare with the 0.1 branch?

HolgerJeromin, 2015-08-10:
We can use Travis to build on Mac OS X as well: 
http://docs.travis-ci.com/user/workers/os-x-infrastructure/

schumajs, 2015-08-14:
Hey guys,

I just wanted to let you know that I'm still on it. I applied my changes to the master branch but ended up with weird include errors that could be fixed by changing the include order. I need to dig a bit deeper to find the cause of those errors. Anyway the lib is building. I couldn't make the examples build, though. If you want have a first peek you can checkout branch 'osx' in my fork: https://github.com/schumajs/open62541/tree/osx

Stasik0, 2015-08-17:
@schumajs okay, looks good, the changes see to be marginal. What are the errors you getting from the example builds?

ngoossens, 2015-08-28:
Hi Guys,

Following on from the work @schumajs did I have submitted pull request https://github.com/acplt/open62541/pull/353. This get all targets building on OS X and all unit tests pass too.

If this gets merged I am happy to set up travis to do an OS X compile. 

Regarding the [build instructions](https://github.com/acplt/open62541/wiki/Building%20open62541), on OSX Yosemite with Xcode and [Homebrew](http://brew.sh/) installed, the following additional packages are required to get everything up and running
``` 
brew install cmake
brew install check
brew install libxml2
brew install userspace-rcu
brew install graphviz
brew install doxygen
brew install swig

pip install lxml
pip install sphinx
```

Please let me know if you have any questions.

Stasik0, 2015-08-28:
@ngoossens nice one! Great job! I left you some comments in PR. Do you have some experience in getting it running on Travis CI OSX stack?

ngoossens, 2015-08-28:
Thanks @Stasik0, happy to help out. No experience with the OS X stack in Travis CI but I am always happy to learn new things!

Stasik0, 2015-09-01:
fixed in #357
@ngoossens, feel free to edit the wiki build instructions page

HolgerJeromin, 2015-09-01:
Thanks for your contribution!

closed 2015-09-01 by HolgerJeromin