open62541/open62541 #320: DataSource Variable works on Windows but not on Linux - invalid free

martina-felber, 2015-08-11:
Hello!

I've tried to define a datasource variable node, which increments a value on each read and offers a write function to set the value of the variable to the given param. I copied your example code which adds the "current time" data source variable. Your example works finde on Windows and Linux. But my defined data source variable works only on Windows and crashes on Linux (openSUSE 13.1). Valgrind states that the server frees an invalid pointer.

Server Code main():
```C
	// add data source variable "Inc_And_Set"
	// at each read the value is incremented
	// with a write call the value can be set to the input param
	UA_Int32 Inc_And_Set_Value = -1;
	UA_DataSource Inc_And_Set_Source = (UA_DataSource) {
					.handle = &Inc_And_Set_Value,
					.read = IncValue,
					.write = SetValue};
	UA_Server_addDataSourceVariableNode(server, Inc_And_Set_Source,
			UA_QUALIFIEDNAME(1, "Inc_And_Set"), UA_NODEID_STRING(1, "Inc_And_Set"),
			UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), NULL);
```
Server Code IncValue():
```C
/* UA_DataSource .read function
 * increments value of handle at each call
 */
UA_StatusCode IncValue(void *handle, UA_Boolean includeSourceTimeStamp,
		const UA_NumericRange *range, UA_DataValue *value) {
	printf("IncValue call() \n");
	if(range) {
		value->hasStatus = UA_TRUE;
		value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
		return UA_STATUSCODE_GOOD;
	}

	if(handle != 0 && value != 0) {
		UA_Int32* pValue = (UA_Int32*) handle;

		value->value.type = &UA_TYPES[UA_TYPES_INT32];
		value->value.arrayLength = -1;
		value->value.data = pValue;
		value->hasValue = UA_TRUE;

		if(includeSourceTimeStamp) {
			value->hasSourceTimestamp = UA_TRUE;
			value->sourceTimestamp = UA_DateTime_now();
		}
		(*pValue)++;
		printf("return good \n");
		return UA_STATUSCODE_GOOD;
	}
	printf("return error \n");
	return UA_STATUSCODE_BADARGUMENTSMISSING; // don't know which error code to use
}
```

Client Code main() - ReadRequest:
```C
    ////////////////////////////////////////////////////////////////////////////////////
    // read datasource variable "Inc_And_Set"
    printf("\nRead Inc_And_Set value: ");
    UA_Int32 * pValue = 0;
    UA_ReadRequest_init(&rReq);
    rReq.nodesToRead = UA_ReadValueId_new();
    rReq.nodesToReadSize = 1;
    rReq.nodesToRead[0].nodeId = UA_NODEID_STRING_ALLOC(1, "Inc_And_Set");
    rReq.nodesToRead[0].attributeId = UA_ATTRIBUTEID_VALUE;

    rResp = UA_Client_read(client, &rReq);
    if(rResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD &&
       rResp.resultsSize > 0 && rResp.results[0].hasValue &&
       UA_Variant_isScalar(&rResp.results[0].value) &&
       rResp.results[0].value.type == &UA_TYPES[UA_TYPES_INT32]) {
[... 113 lines cut ...]
==5588== Copyright (C) 2002-2012, and GNU GPL'd, by Julian Seward et al.
==5588== Using Valgrind-3.8.1 and LibVEX; rerun with -h for copyright info
==5588== Command: ./opc_linux_datasource_client
==5588== 

Read current time: 08/11/2015 08:30:17.470.943.000

Read Inc_And_Set value: 0

Client finished

==5588== 
==5588== HEAP SUMMARY:
==5588==     in use at exit: 0 bytes in 0 blocks
==5588==   total heap usage: 79 allocs, 79 frees, 1,446,625 bytes allocated
==5588== 
==5588== All heap blocks were freed -- no leaks are possible
==5588== 
==5588== For counts of detected and suppressed errors, rerun with: -v
==5588== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
user@linux:~/AM335x-052-eglibc/eclipse/workspace/opc_linux_datasource_client/Debug> 

Many thanks for your help.
Best regards
Martina

Stasik0, 2015-08-11:
As far as I understand it you will need to allocate memory for the data value and copy it into it.
The server tries to free this pointer after the response has been sent and crashes.
So instead of passing a pointer to handle to value.data as in ```value->value.data = pValue;``` try to allocate ```UA_Int32* pValue = UA_Int32_new();``` and copy the value. Since the last implementation, the user has to do it manually.

martina-felber, 2015-08-11:
Many thanks for your hint!

closed 2015-08-11 by martina-felber