open62541/open62541 #321: Client memory Leak if connection failed - Master branch

martina-felber, 2015-08-11:
Hello!

I'm using your master branch code and get a memory leak if the connection to the server failed.
If I start the client without a running server I get the following output from valgrind:

user@linux:~/AM335x-052-eglibc/eclipse/workspace/opc_linux_datasource_client/Debug> valgrind --leak-check=yes ./opc_linux_datasource_client 
==5701== Memcheck, a memory error detector
==5701== Copyright (C) 2002-2012, and GNU GPL'd, by Julian Seward et al.
==5701== Using Valgrind-3.8.1 and LibVEX; rerun with -h for copyright info
==5701== Command: ./opc_linux_datasource_client
==5701== 
[08/11/2015 08:40:30.387.163] warning/communication     Connection failed
==5701== 
==5701== HEAP SUMMARY:
==5701==     in use at exit: 65,544 bytes in 2 blocks
==5701==   total heap usage: 15 allocs, 13 frees, 68,920 bytes allocated
==5701== 
==5701== 65,544 (8 direct, 65,536 indirect) bytes in 1 blocks are definitely lost in loss record 2 of 2
==5701==    at 0x4029EAD: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==5701==    by 0x80657A9: ClientNetworkLayerTCP_connect (open62541.c:12752)
==5701==    by 0x8062522: UA_Client_connect (open62541.c:11617)
==5701==    by 0x8048D7D: main (client.c:20)
==5701== 
==5701== LEAK SUMMARY:
==5701==    definitely lost: 8 bytes in 1 blocks
==5701==    indirectly lost: 65,536 bytes in 1 blocks
==5701==      possibly lost: 0 bytes in 0 blocks
==5701==    still reachable: 0 bytes in 0 blocks
==5701==         suppressed: 0 bytes in 0 blocks
==5701== 
==5701== For counts of detected and suppressed errors, rerun with: -v
==5701== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
user@linux:~/AM335x-052-eglibc/eclipse/workspace/opc_linux_datasource_client/Debug> 

Is this an intended behaviour? Or did I something wrong?
Best regards
Martina

martina-felber, 2015-08-11:
I don't know what you did, but it does not work.
I downloaded the current master branch half an hour ago, the example code crashes and if I use the new open62541.c and h file I get following output from valgrind when starting the server:

user@linux:~/AM335x-052-eglibc/eclipse/workspace/opc_linux_browse_server/Debug> valgrind --leak-check=yes ./opc_linux_browse_server 
==7472== Memcheck, a memory error detector
==7472== Copyright (C) 2002-2012, and GNU GPL'd, by Julian Seward et al.
==7472== Using Valgrind-3.8.1 and LibVEX; rerun with -h for copyright info
==7472== Command: ./opc_linux_browse_server
==7472== 
[08/11/2015 11:09:44.477.793] warning/communication     Error during socket binding
==7472== Invalid read of size 4
==7472==    at 0x805BB13: processJobs (open62541.c:8436)
==7472==    by 0x805C301: UA_Server_run_mainloop (open62541.c:8981)
==7472==    by 0x805C37B: UA_Server_run (open62541.c:9018)
==7472==    by 0x806B473: main (server.c:290)
==7472==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==7472== 
==7472== 
==7472== Process terminating with default action of signal 11 (SIGSEGV)
==7472==  Access not within mapped region at address 0x0
==7472==    at 0x805BB13: processJobs (open62541.c:8436)
==7472==    by 0x805C301: UA_Server_run_mainloop (open62541.c:8981)
==7472==    by 0x805C37B: UA_Server_run (open62541.c:9018)
==7472==    by 0x806B473: main (server.c:290)
==7472==  If you believe this happened as a result of a stack
==7472==  overflow in your program's main thread (unlikely but
==7472==  possible), you can try to increase the size of the
==7472==  main thread stack using the --main-stacksize= flag.
==7472==  The main thread stack size used in this run was 8388608.
==7472== 
==7472== HEAP SUMMARY:
==7472==     in use at exit: 117,051 bytes in 713 blocks
==7472==   total heap usage: 1,127 allocs, 414 frees, 194,679 bytes allocated
==7472== 
==7472== LEAK SUMMARY:
==7472==    definitely lost: 0 bytes in 0 blocks
==7472==    indirectly lost: 0 bytes in 0 blocks
==7472==      possibly lost: 0 bytes in 0 blocks
==7472==    still reachable: 117,051 bytes in 713 blocks
==7472==         suppressed: 0 bytes in 0 blocks
==7472== Reachable blocks (those to which a pointer was found) are not shown.
==7472== To see them, rerun with: --leak-check=full --show-reachable=yes
==7472== 
==7472== For counts of detected and suppressed errors, rerun with: -v
==7472== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Speicherzugriffsfehler
user@linux:~/AM335x-052-eglibc/eclipse/workspace/opc_linux_browse_server/Debug>

Stasik0, 2015-08-11:
But client has no leaks :) note the error "https://github.com/acplt/open62541/blob/master/examples/networklayer_tcp.c#L283" -> it seems that the server cannot bind the socket -> the server is probably running twice

However there was a problem with error handling in this case. I pushed a fix

Stasik0, 2015-08-11:
btw: @martina-felber would you write me a mail? http://www.plt.rwth-aachen.de/cms/PLT/Der-Lehrstuhl/Team/~eljy/Gruener-Sten/ We may have an interesting event for you.

martina-felber, 2015-08-12:
thanks for the fix!
I don't know why but I still can't get it working, I enabled ENABLE_METHODCALLS, ENABLE_SUBSCRIPTIONS, ENABLE_AMALGAMATION, BUILD_EXAMPLECLIENT, BUILD_EXAMPLESERVER, ENABLE_ADDNODES with cmake.

I used the open62541 files and server.c and client.c from the examples directory. At first I thought it's a problem with methodcalls, because both applications crashed, when the method call was invoked. I uncommented it and both crashed at the subscriptions call. I uncommented the subscriptions part in client and server and now the server crashes when writing "the.answer". I have no clue.

But your master branch from 10. August works fine.

Stasik0, 2015-08-12:
@martina-felber I cannot reproduce it, master works good here on linux... try a clean checkout of the master in another folder...

martina-felber, 2015-08-12:
okay, probably I got the wrong files. I tried on Linux openSuse 13.1:
git clone https://github.com/acplt/open62541.git
ENABLE_METHODCALLS, ENABLE_SUBSCRIPTIONS, ENABLE_AMALGAMATION, BUILD_EXAMPLECLIENT, BUILD_EXAMPLESERVER, ENABLE_ADDNODES

I'm not sure if "git clone https://github.com/acplt/open62541.git" is the current master branch, I just copied the build instructions. I usually download the zip files in Windows.

I executed the built examples - the server crashes - server output:
user@linux:~/AM335x-052-eglibc/opc_next_try/open62541/build> ./server
[08/12/2015 09:24:05.784.336] info/communication        Listening on opc.tcp://linux.site:16664
getMonitoredItems was called
Speicherzugriffsfehler
user@linux:~/AM335x-052-eglibc/opc_next_try/open62541/build> 

client output:
user@linux:~/AM335x-052-eglibc/opc_next_try/open62541/build> ./client
Browsing nodes in objects folder:
NAMESPACE NODEID           BROWSE NAME      DISPLAY NAME    
0         61               FolderType       FolderType      
0         2253             Server           Server          
1         96               current time     current time    
1         the.answer       the answer       the answer      
1         50000            Demo             Demo            
1         62541            ping             ping            
Create subscription succeeded, id 2807376561
Monitoring 'the.answer', id 2807376561
The Answer has changed!

Reading the value of node (1, "the.answer"):
the value is: 42

Writing a value of node (1, "the.answer"):
the new value is: 43
The Answer has changed!
Subscription removed
Method call was successfull, and 1 returned values available.
Created 'NewReference' with numeric NodeID 12133
Created 'NewObjectType' with numeric NodeID 12134
Created 'NewObject' with numeric NodeID 176
Created 'NewVariable' with numeric NodeID 177
user@linux:~/AM335x-052-eglibc/opc_next_try/open62541/build> 

I ran the examples with valgrind too, it outputs a lot of invalid free, read and write.
Thanks.

Stasik0, 2015-08-12:
but the original problem is fixed, isn't it?

could you test if it is a regression or not i.e. compare it to the version of 10. August. I am pretty sure i did not introduce it

Stasik0, 2015-08-12:
okay, hopfully it solved it.. i have seen some memleaks in the server, but it is not caused by me :)

martina-felber, 2015-08-13:
thank you very much, your code works!

commit d8862f17 (2015-08-11): fixes #321, thanks @martina-felber
closed 2015-08-13 by martina-felber