open62541/open62541 #322: memory leak at UA_Server_addMethodNode if UA_NodeId is a string

martina-felber, 2015-08-14:
Hello!

Thanks for your example code for method calls in the master branch (I downloaded the current master branch this morning). 
Unfortunately I get a memory leak if I use UA_Server_addMethodNode and the NodeId is a string. 

```C
	// -> this causes a memory leak (12 bytes for "Hello Server")
	UA_Server_addMethodNode(server, UA_QUALIFIEDNAME(1,"Hello Server"), UA_NODEID_STRING(1, "Hello Server"),
			UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
			&getMessage, 1, &inputArguments, 1, &outputArguments, NULL);
```

```C
	// -> a numeric ID does not cause a memory leak
	UA_Server_addMethodNode(server, UA_QUALIFIEDNAME(1,"Hello Server"), UA_NODEID_NUMERIC(1, 12),
			UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
			&getMessage, 1, &inputArguments, 1, &outputArguments, NULL);
```

valgrind output:

==2195== 
==2195== HEAP SUMMARY:
==2195==     in use at exit: 12 bytes in 1 blocks
==2195==   total heap usage: 879 allocs, 878 frees, 173,702 bytes allocated
==2195== 
==2195== 12 bytes in 1 blocks are definitely lost in loss record 1 of 1
==2195==    at 0x4029EAD: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2195==    by 0x8049CFE: UA_NodeId_copy (open62541.c:2410)
==2195==    by 0x8058E3D: UA_Server_addNodeWithSession (open62541.c:7530)
==2195==    by 0x8058C6A: UA_Server_addNode (open62541.c:7478)
==2195==    by 0x80590A4: UA_Server_addMethodNode (open62541.c:7578)
==2195==    by 0x806923F: main (server.c:89)

Or did I call the method with wrong params?
Many thanks in advance for your help!
Best regards
Martina

ichrispa, 2015-08-14:
Hi Martina,

does that node ever get deleted after you create it? Is there any reason to assume that the memory allocated by addNodeWithSession should have been deallocated?

Best regards,

Chris

martina-felber, 2015-08-14:
Hi Chris!

I'm not very familiar with your internal API. I stepped through your code and like valgrind stated at line 7530 is memory allocated but never freed. This happens intern in function UA_Server_addNodeWithSession(). In general the NodeId gets copied several times and freed afterwards but I think not at these lines of code:

the param NodeId gets copied to result
```C
        if(UA_NodeId_copy(&node->nodeId, &result.addedNodeId) != UA_STATUSCODE_GOOD) {
            result.statusCode = UA_STATUSCODE_BADOUTOFMEMORY;
            goto ret2;
        }
```

result is returned to function _addNodes line 7578:
```C
 UA_AddNodesResult addRes = UA_Server_addNode(server, (UA_Node*)newMethod, parentNodeId, referenceTypeId);
```

and then in line 7601 addRes gets a new value, but the string nodeId is not freed before
```C
addRes = UA_Server_addNode(server, (UA_Node*)inputArgumentsVariableNode, methodExpandedNodeId,
                               UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY));
```

As it's all internal I thought that the memory should have been deallocated, because the user has no chance to do so. I hope I didn't make a mistake.

Many thanks and best regards
Martina

ichrispa, 2015-08-14:
Hi @martina-felber, 

how are you stopping the server? Do you for example intercept any signals and perform a cleanup before exiting?

Beste regards,

Chris

martina-felber, 2015-08-14:
Hi!

I just started the server, no client connection and then I pressed Ctrl + C to terminate the server. My server only adds this one method node and if the nodeId is a string -> a memory leak occurs. 

user@linux:~/AM335x-052-eglibc/eclipse/workspace/opc_linux_method_server/Debug> valgrind --leak-check=yes ./opc_linux_method_server 
==3451== Memcheck, a memory error detector
==3451== Copyright (C) 2002-2012, and GNU GPL'd, by Julian Seward et al.
==3451== Using Valgrind-3.8.1 and LibVEX; rerun with -h for copyright info
==3451== Command: ./opc_linux_method_server
==3451== 
[08/14/2015 11:43:06.282.826] info/communication        Listening on opc.tcp://linux.site:16664
^C[08/14/2015 11:43:07.632.876] info/server     Received Ctrl-C

==3451== 
==3451== HEAP SUMMARY:
==3451==     in use at exit: 12 bytes in 1 blocks
==3451==   total heap usage: 847 allocs, 846 frees, 171,229 bytes allocated
==3451== 
==3451== 12 bytes in 1 blocks are definitely lost in loss record 1 of 1
==3451==    at 0x4029EAD: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==3451==    by 0x8049CFE: UA_NodeId_copy (open62541.c:2410)
==3451==    by 0x8058E3D: UA_Server_addNodeWithSession (open62541.c:7530)
==3451==    by 0x8058C6A: UA_Server_addNode (open62541.c:7478)
==3451==    by 0x80590A4: UA_Server_addMethodNode (open62541.c:7578)
==3451==    by 0x80691DC: main (server.c:89)
==3451== 
==3451== LEAK SUMMARY:
==3451==    definitely lost: 12 bytes in 1 blocks
==3451==    indirectly lost: 0 bytes in 0 blocks
==3451==      possibly lost: 0 bytes in 0 blocks
==3451==    still reachable: 0 bytes in 0 blocks
==3451==         suppressed: 0 bytes in 0 blocks
==3451== 
==3451== For counts of detected and suppressed errors, rerun with: -v
==3451== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
user@linux:~/AM335x-052-eglibc/eclipse/workspace/opc_linux_method_server/Debug> 

the same thing happens if I use your example server.c code and use a string NodeId instead of UA_NODEID_NUMERIC(1,62541)

StalderT, 2015-08-16:
For me I have two leak with UA_NODEID_STRING(1, "Hello Server") instead of UA_NODEID_NUMERIC(1,62541)

==2648== HEAP SUMMARY:
==2648==     in use at exit: 24 bytes in 2 blocks
==2648==   total heap usage: 1,622 allocs, 1,620 frees, 323,469 bytes allocated
==2648== 
==2648== 12 bytes in 1 blocks are definitely lost in loss record 1 of 2
==2648==    at 0x4029FEC: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2648==    by 0x403E9F2: UA_String_copy (open62541.c:2199)
==2648==    by 0x403F461: UA_NodeId_copy (open62541.c:2416)
==2648==    by 0x404EFBB: UA_Server_addNodeWithSession (open62541.c:7530)
==2648==    by 0x404FA1A: UA_Server_addNode (open62541.c:7478)
==2648==    by 0x404FA1A: UA_Server_addMethodNode (open62541.c:7578)
==2648==    by 0x804A6DF: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==2648== 
==2648== 12 bytes in 1 blocks are definitely lost in loss record 2 of 2
==2648==    at 0x4029FEC: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2648==    by 0x403E9F2: UA_String_copy (open62541.c:2199)
==2648==    by 0x403F461: UA_NodeId_copy (open62541.c:2416)
==2648==    by 0x404FA76: UA_Server_addMethodNode (open62541.c:7589)
==2648==    by 0x804A6DF: main (in /data/linux/open62541/src/open62541-master/examples/a.out)

This patch correct one of the two leak : https://github.com/StalderT/open62541/commit/fb844c3b10b2aa295c72caf29dd9007a1a2ae721

With the patch : 

==2770== 
==2770== HEAP SUMMARY:
==2770==     in use at exit: 12 bytes in 1 blocks
==2770==   total heap usage: 1,999 allocs, 1,998 frees, 3,019,190 bytes allocated
==2770== 
==2770== 12 bytes in 1 blocks are definitely lost in loss record 1 of 1
==2770==    at 0x4029FEC: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2770==    by 0x403E9F2: UA_String_copy (open62541.c:2199)
==2770==    by 0x403F461: UA_NodeId_copy (open62541.c:2416)
==2770==    by 0x404FA7A: UA_Server_addMethodNode (open62541.c:7589)
==2770==    by 0x804A6E2: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==2770== 
==2770== LEAK SUMMARY:
==2770==    definitely lost: 12 bytes in 1 blocks
==2770==    indirectly lost: 0 bytes in 0 blocks
==2770==      possibly lost: 0 bytes in 0 blocks
==2770==    still reachable: 0 bytes in 0 blocks
==2770==         suppressed: 0 bytes in 0 blocks

there is still this:

    UA_AddNodesResult addRes = UA_Server_addNode(server, (UA_Node*)newMethod, parentNodeId, referenceTypeId);
    if(addRes.statusCode != UA_STATUSCODE_GOOD) {
        UA_MethodNode_delete(newMethod);
        return addRes.statusCode;
    }
    
    UA_ExpandedNodeId methodExpandedNodeId;
    UA_ExpandedNodeId_init(&methodExpandedNodeId);
    UA_NodeId_copy(&addRes.addedNodeId, &methodExpandedNodeId.nodeId);
    
    if (createdNodeId != UA_NULL)
      UA_NodeId_copy(&addRes.addedNodeId, createdNodeId); // THIS FUNCTION ALLOCATE MEMORY BUT ARE NOT FREED !
    
    /* create InputArguments */

Regards,

Thomas

StalderT, 2015-08-16:
For the second memory leak : UA_Server_addMethodNode make 2 allocations for createdNodeId. Remove the second.

You can find the patch here : https://github.com/StalderT/open62541/commit/41583938296c638fd9131ee3db904ad5f2a1d622

martina-felber, 2015-08-17:
many thanks for your patches!

closed 2015-08-17 by martina-felber