open62541/open62541 #323: Subscriptions in winxp are not working

Hexsc, 2015-08-14:
Hi!

I want to mention that subscriptions are not working work in winxp.
I tested my code in win7 and openSUSE-13.1 and it worked fine, but in winxp only the first UA_Client_doPublish command directly after creating the subscription works without problems.
The others are working randomly and sometimes the exe crashes at UA_Client_removeSubscription.

Regards,
Hexsc

Stasik0, 2015-08-17:
@Hexsc it seems that there is a memory problem, can you start your server in valgrind and repeat the actions that cause the error? Then show us the valgrind trace. Probably there is some uninitialised read or write

HolgerJeromin, 2015-08-17:
@Stasik0: valgrind and windows (xp)? :-)

FlorianPalm, 2015-08-17:
server -> linux
client -> winxp

Stasik0, 2015-08-17:
@HolgerJeromin I did not see that the client has problems. Still one may try to run client in valgrind to detect problems under openSUSE.

Stasik0, 2015-08-17:
In the client, apart from memleaks, I see the following error
```
==32297== Invalid read of size 4
==32297==    at 0x40579B1: UA_Client_removeSubscription (open62541.c:11824)
==32297==    by 0x8049515: main (client.c:131)
==32297==  Address 0x42fa1c8 is 56 bytes inside a block of size 64 free'd
==32297==    at 0x402C3D8: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==32297==    by 0x4057F04: UA_Client_unMonitorItemChanges (open62541.c:11951)
==32297==    by 0x40579AA: UA_Client_removeSubscription (open62541.c:11825)
==32297==    by 0x8049515: main (client.c:131)
==32297== 
Subscription removed
```
@Hexsc you a probably running some other client code? Do you?

Hexsc, 2015-08-17:
@Stasik0  I took your example code (master branch 13.8) and changed several things to test subscriptions (e.g. with datasource variables). In valgrind, I have the same invalid read as you, additional to a the bunch of memory leaks.

Stasik0, 2015-08-17:
But no further invalid access-indications?

Hexsc, 2015-08-17:
No.

StalderT, 2015-08-17:
A patch to remove Invalid read of size 4

https://github.com/acplt/open62541/compare/master...StalderT:patch-4

@Hexsc can you try client on xp with patch?

Memory leak persists : 

==2952== 
==2952== HEAP SUMMARY:
==2952==     in use at exit: 170 bytes in 7 blocks
==2952==   total heap usage: 155 allocs, 148 frees, 1,121,862 bytes allocated
==2952== 
==2952== 10 bytes in 1 blocks are definitely lost in loss record 3 of 7
==2952==    at 0x4029FEC: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2952==    by 0x403EA32: UA_String_copy (open62541.c:2199)
==2952==    by 0x403F4A1: UA_NodeId_copy (open62541.c:2416)
==2952==    by 0x40499F2: UA_Client_monitorItemChanges (open62541.c:11897)
==2952==    by 0x8048EAF: main (in /data/linux/open62541/src/open62541-master/examples/b.out)
==2952== 
==2952== 32 (16 direct, 16 indirect) bytes in 1 blocks are definitely lost in loss record 5 of 7
==2952==    at 0x4029FEC: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2952==    by 0x404A090: UA_Client_processPublishRx (open62541.c:12040)
==2952==    by 0x404A1E8: UA_Client_doPublish (open62541.c:12076)
==2952==    by 0x8048EF9: main (in /data/linux/open62541/src/open62541-master/examples/b.out)
==2952== 
==2952== 64 (60 direct, 4 indirect) bytes in 1 blocks are definitely lost in loss record 6 of 7
==2952==    at 0x402ADC8: calloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2952==    by 0x4043E94: UA_Array_decodeBinary (open62541.c:3418)
==2952==    by 0x4043B5D: UA_decodeBinary (open62541.c:4403)
==2952==    by 0x4049F62: UA_Client_processPublishRx (open62541.c:12009)
==2952==    by 0x404A1E8: UA_Client_doPublish (open62541.c:12076)
==2952==    by 0x8048EEB: main (in /data/linux/open62541/src/open62541-master/examples/b.out)
==2952== 
==2952== 64 (60 direct, 4 indirect) bytes in 1 blocks are definitely lost in loss record 7 of 7
==2952==    at 0x402ADC8: calloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2952==    by 0x4043E94: UA_Array_decodeBinary (open62541.c:3418)
==2952==    by 0x4043B5D: UA_decodeBinary (open62541.c:4403)
==2952==    by 0x4049F62: UA_Client_processPublishRx (open62541.c:12009)
==2952==    by 0x404A1E8: UA_Client_doPublish (open62541.c:12076)
==2952==    by 0x8049214: main (in /data/linux/open62541/src/open62541-master/examples/b.out)
==2952== 
==2952== LEAK SUMMARY:
==2952==    definitely lost: 146 bytes in 4 blocks
==2952==    indirectly lost: 24 bytes in 3 blocks
==2952==      possibly lost: 0 bytes in 0 blocks
==2952==    still reachable: 0 bytes in 0 blocks
==2952==         suppressed: 0 bytes in 0 blocks

Stasik0, 2015-08-17:
@StalderT you are a second quicker than I :( :+1:

StalderT, 2015-08-17:
First memory leak : https://github.com/acplt/open62541/compare/master...StalderT:patch-5

Hexsc, 2015-08-17:
@StalderT It works now, many thanks to you!
But why did it not work before, that invalid free cannot be that obstructive, right?

Stasik0, 2015-08-17:
double free is an undefined behaviour... we were just lucky it did not crash on windows7 and opensuse

Stasik0, 2015-08-17:
@StalderT i allowed myself to open a pull-request of the fix

StalderT, 2015-08-17:
fix for second memory leak https://github.com/StalderT/open62541/commit/e7f8f0a26263f2915cfa06add351a227be523080

But not sure that it don't affect other things.

Stasik0, 2015-08-17:
> fix for second memory leak StalderT@e7f8f0a
>
> But not sure that it don't affect other things.

Let us pull the invalid read first and ask @ichrispa for memleak review

Hexsc, 2015-08-17:
With the newest master i have again invalid reads :(

==7417== Invalid read of size 4
==7417==    at 0x806366F: UA_Client_removeSubscription (open62541.c:11831)
==7417==    by 0x8049F2D: main (client.c:318)
==7417==  Address 0x42b3e20 is 56 bytes inside a block of size 64 free'd
==7417==    at 0x402B1DD: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==7417==    by 0x8063668: UA_Client_removeSubscription (open62541.c:11832)
==7417==    by 0x8049F2D: main (client.c:318)
==7417== 
Int32 subscription removed
==7417== Invalid read of size 4
==7417==    at 0x806366F: UA_Client_removeSubscription (open62541.c:11831)
==7417==    by 0x8049F4F: main (client.c:322)
==7417==  Address 0x42e4358 is 56 bytes inside a block of size 64 free'd
==7417==    at 0x402B1DD: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==7417==    by 0x8063668: UA_Client_removeSubscription (open62541.c:11832)
==7417==    by 0x8049F4F: main (client.c:322)
==7417== 
Float subscription removed
==7417== Invalid read of size 4
==7417==    at 0x806366F: UA_Client_removeSubscription (open62541.c:11831)
==7417==    by 0x8049F71: main (client.c:326)
==7417==  Address 0x4283920 is 56 bytes inside a block of size 64 free'd
==7417==    at 0x402B1DD: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==7417==    by 0x8063668: UA_Client_removeSubscription (open62541.c:11832)
==7417==    by 0x8049F71: main (client.c:326)
==7417== 
Datasource subscription removed

Stasik0, 2015-08-17:
@Hexsc I cannot reproduce it, do you use this one? https://github.com/acplt/open62541/commit/c53521a86cee2cb15fa20b5112a56346f1b4ce53

StalderT, 2015-08-17:
It also works for me : 

==2401== Memcheck, a memory error detector
==2401== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==2401== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==2401== Command: b.out
==2401== 
Browsing nodes in objects folder:
NAMESPACE NODEID           BROWSE NAME      DISPLAY NAME    
0         61               FolderType       FolderType      
0         2253             Server           Server          
1         96               current time     current time    
1         the.answer       the answer       the answer      
1         50000            Demo             Demo            
1         62541            ping             ping            
Create subscription succeeded, id 3840531737
Monitoring 'the.answer', id 3840531737
The Answer has changed!

Reading the value of node (1, "the.answer"):
the value is: 43

Writing a value of node (1, "the.answer"):
the new value is: 44
The Answer has changed!
Subscription removed
Method call was successfull, and 1 returned values available.
==2401== 
==2401== HEAP SUMMARY:
==2401==     in use at exit: 170 bytes in 7 blocks
==2401==   total heap usage: 162 allocs, 155 frees, 1,187,540 bytes allocated
==2401== 
==2401== 10 bytes in 1 blocks are definitely lost in loss record 3 of 7
==2401==    at 0x4029FEC: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2401==    by 0x403E9F2: UA_String_copy (open62541.c:2199)
==2401==    by 0x403F461: UA_NodeId_copy (open62541.c:2416)
==2401==    by 0x40499B2: UA_Client_monitorItemChanges (open62541.c:11896)
==2401==    by 0x804905F: main (in /data/linux/open62541/src/open62541-master/examples/b.out)
==2401== 
==2401== 32 (16 direct, 16 indirect) bytes in 1 blocks are definitely lost in loss record 5 of 7
==2401==    at 0x4029FEC: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2401==    by 0x404A050: UA_Client_processPublishRx (open62541.c:12037)
==2401==    by 0x404A1A8: UA_Client_doPublish (open62541.c:12073)
==2401==    by 0x80490A9: main (in /data/linux/open62541/src/open62541-master/examples/b.out)
==2401== 
==2401== 64 (60 direct, 4 indirect) bytes in 1 blocks are definitely lost in loss record 6 of 7
==2401==    at 0x402ADC8: calloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2401==    by 0x4043E54: UA_Array_decodeBinary (open62541.c:3418)
==2401==    by 0x4043B1D: UA_decodeBinary (open62541.c:4403)
==2401==    by 0x4049F22: UA_Client_processPublishRx (open62541.c:12006)
==2401==    by 0x404A1A8: UA_Client_doPublish (open62541.c:12073)
==2401==    by 0x804909B: main (in /data/linux/open62541/src/open62541-master/examples/b.out)
==2401== 
==2401== 64 (60 direct, 4 indirect) bytes in 1 blocks are definitely lost in loss record 7 of 7
==2401==    at 0x402ADC8: calloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2401==    by 0x4043E54: UA_Array_decodeBinary (open62541.c:3418)
==2401==    by 0x4043B1D: UA_decodeBinary (open62541.c:4403)
==2401==    by 0x4049F22: UA_Client_processPublishRx (open62541.c:12006)
==2401==    by 0x404A1A8: UA_Client_doPublish (open62541.c:12073)
==2401==    by 0x80493C4: main (in /data/linux/open62541/src/open62541-master/examples/b.out)
==2401== 
==2401== LEAK SUMMARY:
==2401==    definitely lost: 146 bytes in 4 blocks
==2401==    indirectly lost: 24 bytes in 3 blocks
==2401==      possibly lost: 0 bytes in 0 blocks
==2401==    still reachable: 0 bytes in 0 blocks
==2401==         suppressed: 0 bytes in 0 blocks
==2401== 
==2401== For counts of detected and suppressed errors, rerun with: -v
==2401== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)

Hexsc, 2015-08-21:
@Stasik0  @StalderT 
I always used the latest release of the master branch and not that one you suggested,
assuming that it would work too.
The [c53521a] (https://github.com/acplt/open62541/commit/c53521a86cee2cb15fa20b5112a56346f1b4ce53) version works fine, thanks and sorry for getting on your nerves . :+1:

Stasik0, 2015-08-21:
@Hexsc but the one is the master :) could you test master too... (sorry for nerving you now)

Hexsc, 2015-08-21:
@Stasik0 I don't really know, what you mean, [c53521a] (https://github.com/acplt/open62541/commit/c53521a86cee2cb15fa20b5112a56346f1b4ce53) is a previous version of the master branch, right?
And it's the only version on which I get no invalid reads.
All later versions are letting me despair, cause they don't work (respectively they work, but with invalid reads). 
Some memory leaks are always present, but that is already known by you.

Stasik0, 2015-08-21:
@Hexsc well this is something that i do not understand... c53521a was the commit that introduced the fix, but the respective lines are contained in all other versions (and in the current HEAD) e.g. https://github.com/acplt/open62541/blob/master/src/client/ua_client.c#L697 and https://github.com/acplt/open62541/blob/master/src/client/ua_client.c#L797

could you please state how exactly you produce the invalid read
I am doing this
```
git checkout master
git pull 
make all
./server &
valgrind ./client
```

I get some memleaks but no errors
```
==7705== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

Hexsc, 2015-08-21:
I use:
```javascript
./server &
valgrind --leak-check=yes ./client
```
and get:

==2692== Invalid read of size 4
==2692==    at 0x806366F: UA_Client_removeSubscription (open62541.c:11825)
==2692==    by 0x8049F2D: main (client.c:318)
==2692==  Address 0x42b3e20 is 56 bytes inside a block of size 64 free'd
==2692==    at 0x402B1DD: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2692==    by 0x8063668: UA_Client_removeSubscription (open62541.c:11826)
==2692==    by 0x8049F2D: main (client.c:318)
==2692== 
Int32 subscription removed
==2692== Invalid read of size 4
==2692==    at 0x806366F: UA_Client_removeSubscription (open62541.c:11825)
==2692==    by 0x8049F4F: main (client.c:322)
==2692==  Address 0x42e4358 is 56 bytes inside a block of size 64 free'd
==2692==    at 0x402B1DD: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2692==    by 0x8063668: UA_Client_removeSubscription (open62541.c:11826)
==2692==    by 0x8049F4F: main (client.c:322)
==2692== 
Float subscription removed
==2692== Invalid read of size 4
==2692==    at 0x806366F: UA_Client_removeSubscription (open62541.c:11825)
==2692==    by 0x8049F71: main (client.c:326)
==2692==  Address 0x4283920 is 56 bytes inside a block of size 64 free'd
==2692==    at 0x402B1DD: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==2692==    by 0x8063668: UA_Client_removeSubscription (open62541.c:11826)
==2692==    by 0x8049F71: main (client.c:326)
==2692== 
Datasource subscription removed

Stasik0, 2015-08-21:
can you please post the 3rd line of your open62541.c? mine is 
```
 * Git-Revision: v0.1.0-RC4-408-gdb4a39c
```
which is the current head db4a39cbab

Hexsc, 2015-08-21:
It's:
```javascript
* Git-Revision: unknown--git-commit-id-unknown
```
I think you found the Problem. But what to do?

Stasik0, 2015-08-21:
How did you get the files? Obviously you do not use git, didnt you? Try to get the latest master build from open62541.org/releases/

Hexsc, 2015-08-21:
I downloaded the zip from https://github.com/acplt/open62541/tree/master, 
created the makefiles with cmake and built the source files with:
```javascript
mingw32-make
```
(mingw is in the systempath)

Stasik0, 2015-08-21:
Did you enable amalgamation in cmake? Why not using git clone? Git seems to be installed?

Hexsc, 2015-08-21:
Yes, I enabled amalgamation.
I'm not familiar with git, so I didn't use it.
Thought it would also work without it.
But from your statement I gather that I should? :)

Stasik0, 2015-08-21:
well i am not sure, please try both:
* use these files: http://open62541.org/releases/db4a39cbab.zip
* try to checkout using git:
'''
git clone https://github.com/acplt/open62541.git
'''
then verify you have " * Git-Revision: v0.1.0-RC4-408-gdb4a39c". And try to get the error.

commit c53521a8 (2015-08-17): Merge pull request #331 from StalderT/patch-4

Update ua_client.c, fixes #323
closed 2015-09-04 by Stasik0