open62541/open62541 #324: Subscription timestamp problem

StalderT, 2015-08-14:
Hello,

I use the server in git examples directory.

If I browser the "current time" variable with UAExpert :

1) With subscription I have no timestamp.

2) If I read the variable I have timestamp.

You can see the result :

[image]

Is it normal ?

Stasik0, 2015-08-17:
requiring @ichrispa

ichrispa, 2015-08-18:
Will look into it.

ichrispa, 2015-08-20:
In theory, this is an error of uaexpert; Data Change notifications are never sent with a timestamp (so showing them is an error in itself). I can easily enable server timestamps; source timestamps are currently meaningless, as without an event system there is no possible way a sample in a queue can have any other timestamp then the servers timestamp.

StalderT, 2015-08-20:
Hello,

I have test with the patch https://github.com/acplt/open62541/commit/b7c997a9f923a1eebd36ed48e93ba205e78ebc16

The result are : 

[image]

The second variable is on a another OPC UA server (not open62541).

You can see that sourceTimestamp are updated to.

Stasik0, 2015-08-20:
@ichrispa the pragmatic way would be to assume that source time=server time as its probably done in the other server

ichrispa, 2015-08-21:
fixed in hl branch

commit b7c997a9 (2015-08-20): Forcing dataChangeNotifications to return all timestamp fields (even though sourceTimestamps are meaningless here). Pertains to / fixes #324.
closed 2015-08-21 by ichrispa