open62541/open62541 #327: Memory leak in UA_Subscription_deleteMembers

StalderT, 2015-08-15:
UA_Subscription_deleteMembers want to Delete unpublished Notifications.

UA_Subscription_deleteMembers call LIST_REMOVE and after Subscription_deleteUnpublishedNotification. But Subscription_deleteUnpublishedNotification use the list to Delete unpublished Notifications and "not" are never freed.

You can find code to correct the problem here : https://github.com/StalderT/open62541/commit/097f693ab70cd7d0111b42ec28848dd09eaf5161

Best regards,

Thomas

Stasik0, 2015-08-15:
> You can find code to correct the problem here : StalderT@097f693

Cool! This was the missing memleak. Could you make a PR to save your authorship?

StalderT, 2015-08-15:
May be same problem in Subscription_updateNotifications

    if(Subscription_queuedNotifications(subscription) >= 10) {
        // Remove last entry
        LIST_FOREACH(msg, &subscription->unpublishedNotifications, listEntry)
            LIST_REMOVE(msg, listEntry);
        UA_free(msg);
    }

msg can be removed by LIST_REMOVE but only the first msg is freed.

msg->notification->notificationData->body.data
msg->notification->notificationData
msg->notification

may possibly not be freed.

Stasik0, 2015-08-15:
I am pretty sure that you van check the first one by running server in valgrind and executing the client, not sure about the second one...

StalderT, 2015-08-15:
Yes I have test and validate the first with valgrind. For the second I have no idea how to test it.

Stasik0, 2015-08-15:
Maybe just by changing 10 to 1? Could you look into the debugger? You can also try to issue subscriptions in uaexpert...

StalderT, 2015-08-15:
I will try It monday.

The comment indicate : remove last entry

But loop seems to want to delete all items?

Have you more information?

jpfr, 2015-08-17:
If you want to free the memory in a linked list, you should use `LIST_FOREACH_SAFE`.
Otherwise, you would access the recently freed memory to jump to the next element.

Something like this:
```c
   LIST_FOREACH_SAFE(msg, &subscription->unpublishedNotifications, listEntry) {
        LIST_REMOVE(msg, listEntry);
        UA_free(listEntry);
    }
    UA_free(msg);
```

StalderT, 2015-08-17:
Not easy to reproduce but if I change : 

    if(Subscription_queuedNotifications(subscription) >= 10) {

to

    if(Subscription_queuedNotifications(subscription) >= 1) {
        printf("I AM HERE\n");

We have a memory leak : 
I AM HERE
^C[08/17/2015 08:46:38.297.172] info/server	Received Ctrl-C

==1663== 
==1663== HEAP SUMMARY:
==1663==     in use at exit: 312 bytes in 4 blocks
==1663==   total heap usage: 39,746 allocs, 39,742 frees, 108,764,589 bytes allocated
==1663== 
==1663== 312 (12 direct, 300 indirect) bytes in 1 blocks are definitely lost in loss record 4 of 4
==1663==    at 0x4029FEC: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==1663==    by 0x404BD52: Subscription_updateNotifications (open62541.c:13272)
==1663==    by 0x404DD4A: processJobs (open62541.c:8450)
==1663==    by 0x404DEDD: processRepeatedJobs (open62541.c:8714)
==1663==    by 0x404DEDD: UA_Server_run_mainloop (open62541.c:8956)
==1663==    by 0x404DFE9: UA_Server_run (open62541.c:9024)
==1663==    by 0x804A783: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==1663== 
==1663== LEAK SUMMARY:
==1663==    definitely lost: 12 bytes in 1 blocks
==1663==    indirectly lost: 300 bytes in 3 blocks
==1663==      possibly lost: 0 bytes in 0 blocks
==1663==    still reachable: 0 bytes in 0 blocks
==1663==         suppressed: 0 bytes in 0 blocks
==1663== 
==1663== For counts of detected and suppressed errors, rerun with: -v
==1663== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)

allocated by : 
    msg = (UA_unpublishedNotification *) UA_malloc(sizeof(UA_unpublishedNotification));

StalderT, 2015-08-17:
With the patch : https://github.com/acplt/open62541/compare/master...StalderT:patch-3

I have the result : 

I AM HERE
^C[08/17/2015 09:09:08.692.339] info/server	Received Ctrl-C

==2001== 
==2001== HEAP SUMMARY:
==2001==     in use at exit: 0 bytes in 0 blocks
==2001==   total heap usage: 17,825 allocs, 17,825 frees, 56,379,512 bytes allocated
==2001== 
==2001== All heap blocks were freed -- no leaks are possible
==2001== 
==2001== For counts of detected and suppressed errors, rerun with: -v
==2001== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

Stasik0, 2015-08-17:
this is the last open PR, right?

StalderT, 2015-08-17:
yes https://github.com/acplt/open62541/pull/333

StalderT, 2015-08-18:
Close : https://github.com/acplt/open62541/commit/47ca120f5d6b4ce4d2812e4e0294b64b3745d976

closed 2015-08-18 by StalderT