open62541/open62541 #330: Make server.c more robust

StalderT, 2015-08-17:
Hello,

UA_Server_addMethodNode can allocate memory if createdNodeId is not NULL.
After we must call UA_NodeId_deleteMembers to free the memory.

Patch here : https://github.com/acplt/open62541/pull/328

In case of error in UA_Server_addMethodNode the return value is different from UA_STATUSCODE_GOOD and server.c does not release memory.

Patch here : https://github.com/acplt/open62541/pull/329

Regards,

Thomas

HolgerJeromin, 2015-08-17:
Thanks for your code!
I don't think these description issues are needed, the master guys for this code will find the PRs :-)

StalderT, 2015-08-17:
Sorry I'm new with Github and I still do not know all the mechanisms. I learn and I close this issue.

closed 2015-08-17 by StalderT